ramses-rf/ramses_rf #10: Running as per README on windows fails

Jongmassey, 2020-10-21:
Windows 10, Python 3.8, NanoCUL with latest evofw3

running the install as per README.md
```
git clone https://github.com/zxdavb/evohome_rf
cd evohome_rf
pip install -r requirements.txt
```
and then the client 
```
python client.py monitor COM4
```

produces the follow error output
```
config {'schema': {}, 'allowlist': {}, 'blocklist': {}, 'evofw_flag': None, 'disable_discovery': True, 'use_schema': True, 'enforce_allowlist': False, 'disable_sending': False, 'max_zones': 12, 'packet_log': 'packet.log', 'enforce_blocklist': True, 'reduce_processing': 0, 'debug_mode': 0, 'serial_port': 'COM4', 'execute_cmd': None, 'poll_devices': [], 'probe_devices': []}
Starting evohome_rf...
23:48:50.371 Starting evohome_rf, **config = {'schema': {}, 'allowlist': {}, 'blocklist': {}, 'evofw_flag': None, 'disable_discovery': True, 'use_schema': True, 'enforce_allowlist': False, 'disable_sending': False, 'max_zones': 12, 'packet_log': 'packet.log', 'enforce_blocklist': True, 'reduce_processing': 0, 'debug_mode': 0, 'execute_cmd': None, 'poll_devices': [], 'probe_devices': []}
Traceback (most recent call last):
  File "client.py", line 173, in <module>
    cli()
 File "[...shortened python 3.8 path...]\LocalCache\local-packages\Python38\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "[...shortened python 3.8 path...]\LocalCache\local-packages\Python38\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "[...shortened python 3.8 path...]\LocalCache\local-packages\Python38\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "[...shortened python 3.8 path...]\LocalCache\local-packages\Python38\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "[...shortened python 3.8 path...]\LocalCache\local-packages\Python38\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "[...shortened python 3.8 path...]\LocalCache\local-packages\Python38\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "client.py", line 85, in monitor
    debug_wrapper(**obj, **kwargs)
  File "client.py", line 121, in debug_wrapper
    asyncio.run(main(serial_port, **config))
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.1776.0_x64__qbz5n2kfra8p0\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.1776.0_x64__qbz5n2kfra8p0\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "client.py", line 136, in main
    gwy = Gateway(serial_port, loop=loop, **config)
  File "C:\Users\Jonathan\source\repos\evohome_rf\evohome_rf\__init__.py", line 131, in __init__
    load_schema(self, self._schema) if self.config["use_schema"] else {}
  File "C:\Users\Jonathan\source\repos\evohome_rf\evohome_rf\schema.py", line 232, in load_schema
    schema = SYSTEM_SCHEMA(schema)
  File "[...shortened python 3.8 path...]\LocalCache\local-packages\Python38\site-packages\voluptuous\schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "[...shortened python 3.8 path...]\LocalCache\local-packages\Python38\site-packages\voluptuous\schema_builder.py", line 594, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "[...shortened python 3.8 path...]\LocalCache\local-packages\Python38\site-packages\voluptuous\schema_builder.py", line 432, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: required key not provided @ data['controller']```

Jongmassey, 2020-10-21:
Using this older commit seems to work though
```
 git show
commit 24a96561ab192b26e710df487cb3e45e10af88c1 (HEAD -> master, origin/master, origin/HEAD)
Author: David Bonnes <zxdavb@bonnes.me>
Date:   Tue Apr 7 09:57:05 2020 +0100
```

zxdavb, 2020-10-22:
Yeah, sorry - the thing is under active development - everyone else is using it with Home Assistant, so wouldn't have see this. I have already fixed this, but not pushed it yet. You can do this workaround:
```
python client.py -c config.json ...
```

Where the **config.json** file is like:
```
{
    "config": {
        "use_schema": true,
        "enforce_allowlist": false,
        "enforce_blocklist": true,

        "serial_port": "/dev/ttyUSB0",
        "disable_discovery": true,
        "disable_sending": false,

        "packet_log": null
    },

    "schema": {
        "controller": "01:145038"
    },

    "allowlist": {
        "01:145038": {"name": "CTL:Controller"},
        "04:056053": {"name": "TRV:FrontRoom1"},
    },

    "blocklist": {
        "01:032820": {},
        "12:207082": {"name": null             },
        "30:082155": {"name": "VMS:PosInputVent"},
    }
}
```

zxdavb, 2020-10-22:
The minimum you need is:
```
{
    "config": {
        "use_schema": true,
        "disable_discovery": false,
    },
    "schema": {
        "controller": "01:145038"
    }
}
```
and  you should read [this](https://github.com/zxdavb/evohome_cc/wiki/3.-Installation-&-Configuration#example-configuration)

Jongmassey, 2020-10-22:
Many thanks, I plan on using it with HA eventually I was just doing some bench testing first. 

One thing I'm slightly unclear on is:

> The controller is optional, but is recommended

I don't have a honeywell controller, I plan on using HA with the nanocul dongle as a controller. From the error message it seems that controller is a required configuration value - is that correct? If so, do I need to obtain the ID from the nanocul to populate this value?

I plan on getting some TRVs but as it stands I only have the older relay unit and wireless thermostat. I see their friendly names in the logs - how do I map these to IDs?
```
2020-10-22T10:30:33.920 MSG: || 043 |  I |     |            |            | THM:247796 | mixvalve_config  | 016 | 01C80... || {'zone_idx': '01', 'max_flow_temp': 55, 'pump_rum_time': 15, 'actuator_run_time': 150, 'min_flow_temp': 15, 'unknown_0': 1}
2020-10-22T10:30:34.917 MSG: || 043 |  I |     |            |            | THM:247796 | mixvalve_config  | 016 | 01C80... || {'zone_idx': '01', 'max_flow_temp': 55, 'pump_rum_time': 15, 'actuator_run_time': 150, 'min_flow_temp': 15, 'unknown_0': 1}
2020-10-22T10:30:35.916 MSG: || 043 |  I |     |            |            | THM:247796 | mixvalve_config  | 016 | 01C80... || {'zone_idx': '01', 'max_flow_temp': 55, 'pump_rum_time': 15, 'actuator_run_time': 150, 'min_flow_temp': 15, 'unknown_0': 1}
2020-10-22T10:30:36.906 MSG: || 043 |  I |     |            |            | THM:247796 | datetime         | 009 | 00380... || {'datetime': '2020-10-22 10:30:01'}
2020-10-22T10:33:55.903 MSG: || 043 |  I |     |            |            | THM:247796 | relay_demand     | 002 | 000C     || {'zone_idx': '00', 'relay_demand': 0.06}
2020-10-22T10:33:56.892 MSG: || 043 |  I |     |            |            | THM:247796 | relay_demand     | 002 | 000C     || {'zone_idx': '00', 'relay_demand': 0.06}
2020-10-22T10:33:57.899 MSG: || 043 |  I |     |            |            | THM:247796 | relay_demand     | 002 | 000C     || {'zone_idx': '00', 'relay_demand': 0.06}
2020-10-22T10:36:16.874 MSG: || 082 |  I |     |            |            | THM:234839 | relay_demand     | 002 | 0000     || {'zone_idx': '00', 'relay_demand': 0.0}
2020-10-22T10:36:17.864 MSG: || 082 |  I |     |            |            | THM:234839 | relay_demand     | 002 | 0000     || {'zone_idx': '00', 'relay_demand': 0.0}
2020-10-22T10:36:18.872 MSG: || 081 |  I |     |            |            | THM:234839 | relay_demand     | 002 | 0000     || {'zone_idx': '00', 'relay_demand': 0.0}
2020-10-22T10:36:34.000 MSG: || 077 |  I |     | BDR:247252 |            | >broadcast | actuator_enabled | 003 | 0000FF   || {'actuator_enabled': False}
2020-10-22T10:40:33.920 MSG: || 043 |  I |     |            |            | THM:247796 | mixvalve_config  | 016 | 01C80... || {'zone_idx': '01', 'max_flow_temp': 55, 'pump_rum_time': 15, 'actuator_run_time': 150, 'min_flow_temp': 15, 'unknown_0': 1}
2020-10-22T10:40:34.910 MSG: || 043 |  I |     |            |            | THM:247796 | mixvalve_config  | 016 | 01C80... || {'zone_idx': '01', 'max_flow_temp': 55, 'pump_rum_time': 15, 'actuator_run_time': 150, 'min_flow_temp': 15, 'unknown_0': 1}
2020-10-22T10:40:35.912 MSG: || 043 |  I |     |            |            | THM:247796 | mixvalve_config  | 016 | 01C80... || {'zone_idx': '01', 'max_flow_temp': 55, 'pump_rum_time': 15, 'actuator_run_time': 150, 'min_flow_temp': 15, 'unknown_0': 1}
2020-10-22T10:40:36.904 MSG: || 043 |  I |     |            |            | THM:247796 | datetime         | 009 | 00380... || {'datetime': '2020-10-22 10:40:01'}
```

zxdavb, 2020-10-27:
Your logs show that you are using very old code - please pull the latest version of the master branch - you will notice some of the columns disappear.

The requirement for a controller is an anachronism - the system can easily work without it, I just need to remove the relevant bug. Your controller will be found by using: 
```
python client.py -rrr monitor ...
```

Unfortunately, the controller-free option is not yet available (i.e. running the heating system without a controller, as opposed to running my code without a controller) - you will need to continue using the CMT927 for now. Using a non-evohome controller is also a WIP - give me a week or so.  Go to the HA forum for ongoing info.

Be aware: in some ways, the older HC60NG is superior to a BDR91A.