ramses-rf/ramses_rf #23: A few things noticed ....

martynwendon, 2021-03-16:
Apologies if this is the wrong "format" for reporting issues, but I figured as I'd spotted a few random things it best to create one post rather than many,

So I'm running evohome_rf 0.6.6 - mainly in preparation for playing with https://github.com/smar000/evohome-Listener/blob/gw_evohome_rf/ 

My system consists of the evo controller ATC928, evo gateway RFG100, 10 HR92 TRVs, 9 DTS92 room thermostats and a BDR91.  Combi boiler, 10 zones each consisting of HR92 and DTS92 except for the hall zone, where the ATC928 is sited and used as the thermostat for that zone.  I also have an HGI80 which is currently working under Domoticz, using a really old integration that IIRC is entirely passive in nature.

All been in and working for many years, to the best of my knowledge all bound correctly i.e. each room thermostat (or the ATC928) can be adjusted and the corresponding TRV will actuate.

I left evohome_rf running on a spare HGI80 for an hour or so, the output is attached as is the packet.log

1) The config file wiki entry mentions the schema as having "heating_control" but the output from client.py contains "heating_relay".  Perhaps they are synonymous, but thought it worth mentioning.

> Schema[01:073322 (evohome)] = {
>     "controller": "01:073322",
>     "system": {
>         "heating_relay": "13:002386"
>     },

2) I ran "python client.py -c config.json monitor" with a very simple first config.json :

> {
>     "config": {
>         "disable_sending": null,
>         "disable_discovery": false,
>         "enforce_allowlist": null,
>         "enforce_blocklist": null,
>         "evofw_flag": null,
>         "max_zones": 12,
>         "packet_log": "packet.log",
>         "reduce_processing": 0,
>         "serial_port": "/dev/ttyUSB0",
>         "use_names": null,
>         "use_schema": true
>     },
> "schema": {
>         "controller": "01:073322",
>         "system": {
>             "heating_control": "13:002386"
>         }
>     }
> }

This resulted in:

> Usage: client.py listen [OPTIONS] SERIAL_PORT
> Try 'client.py listen -h' for help.
> 
> Error: Missing argument 'SERIAL_PORT'.

No drama, but presume the serial_port option should be read here.

Similarly, I had "packet_log" in the config.json set, but no packet log was created during the next run, so I ran it again a little later with "-o packet.log" appended.

3) I seem to get many errors like so:

> 14:44:21.740 || CTL:073322 | HGI:002179 | RP | zone_mode        | 08070... || {'zone_idx': '08', 'mode': 'follow_schedule', 'setpoint': 18.0}
> 14:44:22.724 || HGI:002179 | CTL:073322 | RQ | zone_mode        | 09       || {'zone_idx': '09'}
> 14:44:22.740 || CTL:073322 | HGI:002179 | RP | zone_mode        | 09070... || {'zone_idx': '09', 'mode': 'follow_schedule', 'setpoint': 18.0}
> 14:44:23.725 || HGI:002179 | CTL:073322 | RQ | zone_mode        | 0A       || {'zone_idx': '0A'}
> 14:44:23.740 || CTL:073322 | HGI:002179 | RP | zone_mode        | 0A7FF... || {'zone_idx': '0A', 'mode': 'follow_schedule', 'setpoint': None}
> 14:44:24.740 || CTL:073322 | HGI:002179 | RP | zone_mode        | 0B7FF... || {'zone_idx': '0B', 'mode': 'follow_schedule', 'setpoint': None}
> 2021-03-16T14:44:25.725524 045 RQ --- 18:002179 01:073322 --:------ 2349 001 0C < Validation error 
> Traceback (most recent call last):
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/message.py", line 324, in is_valid
>     self._payload = payload_parser(self.raw_payload, self)
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/parsers.py", line 325, in wrapper
>     result = _idx(payload[:2], msg)
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/parsers.py", line 163, in _idx
>     assert (
> AssertionError: '0C' exceeds max zone index
> 2021-03-16T14:44:25.742829 045 RP --- 01:073322 18:002179 --:------ 2349 007 0C7FFF02FFFFFF < Validation error 
> Traceback (most recent call last):
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/message.py", line 324, in is_valid
>     self._payload = payload_parser(self.raw_payload, self)
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/parsers.py", line 292, in wrapper
>     result = func(*args, **kwargs)
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/parsers.py", line 1405, in parser_2349
>     **_idx(payload[:2], msg),
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/parsers.py", line 163, in _idx
>     assert (
> AssertionError: '0C' exceeds max zone index
> 2021-03-16T14:44:27.725678 045 RQ --- 18:002179 01:073322 --:------ 2349 001 0E < Validation error 
> Traceback (most recent call last):
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/message.py", line 324, in is_valid
>     self._payload = payload_parser(self.raw_payload, self)
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/parsers.py", line 325, in wrapper
>     result = _idx(payload[:2], msg)
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/parsers.py", line 163, in _idx
>     assert (
> AssertionError: '0E' exceeds max zone index
> 2021-03-16T14:44:27.741794 045 RP --- 01:073322 18:002179 --:------ 2349 007 0E7FFF02FFFFFF < Validation error 
> Traceback (most recent call last):
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/message.py", line 324, in is_valid
>     self._payload = payload_parser(self.raw_payload, self)
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/parsers.py", line 292, in wrapper
>     result = func(*args, **kwargs)
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/parsers.py", line 1405, in parser_2349
>     **_idx(payload[:2], msg),
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/parsers.py", line 163, in _idx
>     assert (
> AssertionError: '0E' exceeds max zone index
> 2021-03-16T14:44:28.725868 045 RQ --- 18:002179 01:073322 --:------ 2349 001 0F < Validation error 
> Traceback (most recent call last):
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/message.py", line 324, in is_valid
>     self._payload = payload_parser(self.raw_payload, self)
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/parsers.py", line 325, in wrapper
>     result = _idx(payload[:2], msg)
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/parsers.py", line 163, in _idx
>     assert (
> AssertionError: '0F' exceeds max zone index
> 2021-03-16T14:44:28.741749 045 RP --- 01:073322 18:002179 --:------ 2349 007 0F7FFF02FFFFFF < Validation error 
> Traceback (most recent call last):
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/message.py", line 324, in is_valid
>     self._payload = payload_parser(self.raw_payload, self)
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/parsers.py", line 292, in wrapper
>     result = func(*args, **kwargs)
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/parsers.py", line 1405, in parser_2349
>     **_idx(payload[:2], msg),
>   File "/usr/src/evohome_rf/evohome_rf/evohome_rf/parsers.py", line 163, in _idx
>     assert (
> AssertionError: '0F' exceeds max zone index

Not sure if that's a bad / invalid packet, or is it simply because max_zones in the config.json is set at 12 .... although I only have 10 zones?

4) There seems to be quite a few orphans on the "gateway".  Assuming that this is the RFG100, I'm not sure what the devices are as I don't believe the RFG100 is something you can impact setup wise?  Also I see there's a device type "63" which I see in const.py is a "special" NULL device?

5) The bedroom 2 zone is missing a TRV, it's listed as orphan "04:139374".  It's definitely there :-) It actuates correctly when adjusting that rooms thermostat so I assume that it's bound correctly!

6) Any idea what device type "30" is?  I couldn't see it in the const,py but it's convinced that I have one!

7) The hall zone is missing the sensor, this should be the ATC928 evohome controller itself.  I believe that it's correctly bound as it does cause the hall TRV to actuate when the setpoint is adjusted,  I *think* I read somewhere that it's not possible to discover the sensor for a zone where the evohome controller itself is the sensor, but that reference eludes me now ..... there's a lot of info across github and the HA forum!

Should the discovery be accurate?  I can obviously create a schema that suits the system how I believe it is is configured.  But equally so, I did set this system up nearly 6 years ago, so it could be wrong ...  although all evidence points that it is working correctly.

[packet.log](https://github.com/zxdavb/evohome_rf/files/6150854/packet.log)
[output.txt](https://github.com/zxdavb/evohome_rf/files/6150855/output.txt)

zxdavb, 2021-03-17:
Please edit your posts appropriately, so they can be read with ease.

The wiki is out of date - I am short of time.

1. Use this for **config.json** for now.
```
{
    "config": {
        "max_zones": 10,
    },
    "schema": {
        "controller": "01:073322",
    }
}
```
Look at adding in an `allow_list` when you can, and a `block_list` otherwise.

2. Use this command:
```
python client.py -c config.json monitor /dev/ttyUSB0 -o packet.log
```
... later on you can:
```
cat packet.log | python client.py -c config.json parse
```
3. `AssertionError: '0C' exceeds max zone index` -- this is Zone #13, so it has even exceeded 12 - I don't know why this happened - will have a think about it...

4. Neighbours - use a `block_list` to exclude them. The 'Gateway' is the HGI80 (or other USB device). The 63 device is usually a special NULL device.

5. When I run , I get:
```
Schema[01:073322 (evohome)] = {
    "controller": "01:073322",
    "system": {
        "heating_relay": "13:002386"
    },
    "orphans": [
        "04:243797",
        "30:123643"
    ],
    "stored_hotwater": {},
    "underfloor_heating": {},
    "zones": {
...
        "01": {
            "heating_type": "radiator_valve",
            "sensor": "22:030424",
            "devices": [
                "04:139374",
                "22:030424"
            ]
        },
```
... but this may be because I'm running different code to you.

6. The `30:` (GWY) is the RFG100.

7. Correct. Not possible to discover the sensor when it is the controller.

Discovery should be accurate. In almost all cases, where people have suspected discovery is wrong, and have exuded trival causes (e.g. lost packets) it hasn't been the case.

zxdavb, 2021-03-17:
It looks like you have to re-bind `TRV:243797` to zone `07`.

martynwendon, 2021-03-17:
Thanks very much for your time and help!

I rebound the TRV and that seems to have sorted that particular orphan.  Strange though as I said the TRV / thermostat appeared to be working correctly.

Will look at sorting the allowlist next and moving on with getting the gw_evohome_rf set up.

Let me know if there's anything you want me to check / log on that AssertionError, I still seem to get those a lot, they appear to cycle between 0A - 0F

Thanks again!

martynwendon, 2021-04-22:
I'm going to close this issue now as I think all the questions have been answered.

> 1. `AssertionError: '0C' exceeds max zone index` -- this is Zone #13, so it has even exceeded 12 - I don't know why this happened - will have a think about it...

> Let me know if there's anything you want me to check / log on that AssertionError, I still seem to get those a lot, they appear to cycle between 0A - 0F

I think I worked this out .... by accident, nothing heroic!

I have two HGI80 with my original one running under a really old evohome implementation with Domoticz.

The second HGI80 I was testing with evohome_rf and later with gw_evohome_rf.

The last few days I turned off the Domoticz integration and moved over to gw_evohome_rf as it's been working well ..... I noticed those spurious errors stopped.

I subsequently saw that gw_evohome_rf isn't yet handling sending to evohome, so I restarted the Domoticz integration in the meantime and those errors started again.

So it looks like something Domoticz is doing causes them, since the address cycles I guess it might be some sort of polling loop that just blanket polls X zones.

Thanks again for all your help and advice!

closed 2021-04-22 by martynwendon