ramses-rf/ramses_rf #28: Validation error, payload[6:8] 0x03

amorsen, 2021-05-14:
2021-05-14T13:48:21.306620 087 RP --- 30:253184 34:010943 --:------ 2349 013 00064000FFFFFF00110E0507E5 < Validation error: 00 (0x03) 

Traceback (most recent call last):
  File "/home/amorsen/src/ramses_rf/ramses_rf/message.py", line 400, in is_valid
    self._payload = payload_parser(self.raw_payload, self)
  File "/home/amorsen/src/ramses_rf/ramses_rf/parsers.py", line 264, in wrapper
    result = func(*args, **kwargs)
  File "/home/amorsen/src/ramses_rf/ramses_rf/parsers.py", line 1411, in parser_2349
    assert payload[6:8] not in ("00", "02"), f"{payload[6:8]} (0x03)"
AssertionError: 00 (0x03)

I believe that the problem is not due to a receiver problem, as the same exact message is received every 30 minutes.

amorsen, 2021-05-14:
The device also, once in a while, sends this packet:

2021-05-14T14:07:32.811474 085  W --- 30:253184 34:010943 --:------ 313F 009 006000070E0E0507E5 < Corrupt packet: Invalid code for 34:010943 to Rx: 313F (will be ignored)

zxdavb, 2021-05-14:
I am guessing you've a Round Thermostat bound directly to a RFG100?

amorsen, 2021-05-15:
I have identified 2 x Honeywell DT92E bound to 2 x BDR91A1000.

This is what has been discovered so far:

'22:182330',
'13:136987',
'13:043686',
'22:069734',
'13:033689',
'13:260368',
'30:253184',
'12:175517',
'12:238482'

The DT92E's are the 22: devices, they seem to talk to two of the 13: devices, presumably the two BDR91's I found. For the rest I am still figuring out whether those are my devices or belong to my neighbour.

The 30: device is the only one sending messages that ramses_rf can't parse. None of its messages pass validation. I do not know which brand or model device it is.

amorsen, 2021-05-17:
It works correctly now, with 0.9.4:

2021-05-17T11:57:28.983573 092  I --- 30:253184 --:------ 30:253184 2E04 008 00FFFFFFFFFFFF00 
11:57:28.983 || GWY:253184 |            |  I | system_mode      | 00FFF... || {'system_mode': 'auto', 'until': None}

closed 2021-05-20 by zxdavb