ramses-rf/ramses_rf #32: Eavesdropping causes CorruptStateError

Vrobenmat, 2021-06-23:
Eavesdropping enabled to have HR80's present correctly in Schema (HR80 ID's not published by controller during discovery)

It appears the HR80's (once in the schema) then contribute to zone heat demands, based on which various automations are triggered: circulation pumps on/off for specific zones, electric blinds/shutters etc.

`055 RP --- 13:259021 18:070162 --:------ 3EF1 007 00012C012C00FF < Inconsistent state: 13:259021 (BDR) changed parent: 01:201047 (evohome) to 01:201047_00 (radiator_valve), (try restarting the client library)

Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.8/site-packages/ramses_rf/message.py", line 711, in process_msg
    update_entities(msg, msg._gwy._prev_msg)  # update the state database
  File "/srv/homeassistant/lib/python3.8/site-packages/ramses_rf/message.py", line 652, in update_entities
    this._gwy.device_by_id[this.src.id]._handle_msg(this)
  File "/srv/homeassistant/lib/python3.8/site-packages/ramses_rf/devices.py", line 1207, in _handle_msg
    super()._handle_msg(msg)
  File "/srv/homeassistant/lib/python3.8/site-packages/ramses_rf/devices.py", line 328, in _handle_msg
    super()._handle_msg(msg)
  File "/srv/homeassistant/lib/python3.8/site-packages/ramses_rf/devices.py", line 486, in _handle_msg
    self._set_parent(self._ctl._evo._get_zone(msg.payload["parent_idx"]))
  File "/srv/homeassistant/lib/python3.8/site-packages/ramses_rf/devices.py", line 505, in _set_parent
    raise CorruptStateError(
ramses_rf.exceptions.CorruptStateError: Inconsistent state: 13:259021 (BDR) changed parent: 01:201047 (evohome) to 01:201047_00 (radiator_valve),  (try restarting the client library)`

zxdavb, 2021-06-23:
> Eavesdropping enabled to have HR80's present correctly in Schema (HR80 ID's not published by controller during discovery)

This is odd behaviour by the controller, related to HR80s - it is reproducible: when a controller is asked for the TRVs in a zone, it fails to mention the HR80s.

> It appears the HR80's (once in the schema) then contribute to zone heat demands, 

This is correct - the evohome controller does not announce the heat demand of zones (nor does it `RP` to an `RQ` for this data) - the value is constructed by **ramses_rf** from the heat demand of the zone's member devices.

The heat demand of these devices is used only if they're identified as members of that zone by the working schema.

Eavesdropping has always been unsatisfactory, and it relies upon devices' telling the 'truth' as the controller understands it, and upon an undocumented protocol.
 - so, the issue is - how to best get that information into the schema, without falling back to eavesdropping?
 - the alternative is to address this specific bug with eavesdropping...
 
 Maybe do both? There should also be a warning logged that eavesdropping is best avoided.

Vrobenmat, 2021-06-23:
Not sure if eavesdropping has an effect on all device categories (as the error seems related to non-radiator valve device categories,) but we know its useful for old HR80 '04' type radiator valves.

Also knowing that these valves are no longer produced and getting somewhat older, their failure rate is increasing and would be gradually replaced by HR91/HR92 or whatever is the current valve, not sure that investing in a full fix just for HR80's is efficient. However, if usecase is only to get HR80's in the schema, then there may be a simpler way to just do eavesdropping for that.

Warning along the lines that there may be unexpected behaviour with eavesdropping is useful.

zxdavb, 2021-06-23:
>Not sure if eavesdropping has an effect on all device categories

It does.

zxdavb, 2021-06-23:
Can you please put PM/send me a copy your HA **configuration.yaml** file (the `evohome_cc:` / `logger:` sections), and the contents of **.storage/evohome_cc**?

Vrobenmat, 2021-06-23:
[evohome_cc.txt](https://github.com/zxdavb/ramses_rf/files/6702897/evohome_cc.txt)
[configuration.yaml.txt](https://github.com/zxdavb/ramses_rf/files/6702903/configuration.yaml.txt)

zxdavb, 2021-06-24:
So sorry - I think we got crossed wires. @TheMystery?

If so, I was wanting to confirm the sensor faking first, before addressing the other issue - can you do that?

zxdavb, 2021-06-24:
Hmmm... Maybe you're not using sensor faking - sorry, I got confused. 

For your HR80s try:
```text
  config:
    enable_eavesdrop: false
    restore_state: false
  schema:
    controller: 01:201047
    zones:
      "00": {"devices": ["04:053851", "04:039319", "04:053859"]}
```

I suggest you _only_ do this for the HR80s, and not the HR92s (i.e. those that orphan).

Vrobenmat, 2021-06-24:
Explicitly put the HR80's in the zones config as you suggested, but they still appear as orphans in the schema.

zxdavb, 2021-06-25:
@Vrobenmat Please try again with release **0.10.5** of **evohome_cc**.

Vrobenmat, 2021-06-25:
looking good! :-) Schema is now reflecting the HR80's explicitly defined in config. Have disable eavesdropping. Maybe worth adapting the warning or the wiki to indicate the eavesdropping is useful as a one-off to find undiscoverable devices that then need to be added to the zone config.

` "zones": {
                        "00": {
                            "heating_type": "radiator_valve",
                            "sensor": null,
                            "devices": [
                                "04:177708",
                                "04:177712",
                                "04:179991",
                                "04:179993",
                                "04:179995",
                                "04:177718",
                                "04:177709",
                                "04:176841",
                                "04:053851",
                                "04:039319",
                                "04:053859",
                                "04:039325"
                            ]
                        },
                        "01": {
                            "heating_type": "radiator_valve",
                            "sensor": "22:090227",
                            "devices": [
                                "04:173125",
                                "04:173131",
                                "04:179981",
                                "22:090227"
                            ]
                        },
                        "02": {
                            "heating_type": "radiator_valve",
                            "sensor": "22:090231",
                            "devices": [
                                "04:173127",
                                "04:177713",
                                "22:090231"
                            ]
                        },
                        "03": {
                            "heating_type": "radiator_valve",
                            "sensor": "22:028749",
                            "devices": [
                                "04:177711",
                                "04:173146",
                                "04:179866",
                                "22:028749"
                            ]
                        },
`

zxdavb, 2021-06-25:
>Maybe worth adapting the warning or the wiki to indicate the eavesdropping is useful as a one-off to find undiscoverable devices that then need to be added to the zone config.

Could you please add something to: https://github.com/zxdavb/ramses_rf ?

Also: leave this issue open - I need to solve the underlying issue.

zxdavb, 2023-05-12:
I believe all these points, above, have been addressed - so am closing this issue.

Please comment below if I am wrong.

closed 2023-05-12 by zxdavb