ramses-rf/ramses_rf #33: Unavailable state

iMiMx, 2021-06-23:
I noticed my climate.* entities became unknown over night:

https://www.dropbox.com/s/t5rh2ee7dcni1bj/Screenshot%202021-06-23%20at%2007.48.00.png?dl=0

If you trigger a change, it seems to re-learn/update the correct state - so hopefully it's just a timeout or similar issue:

https://www.dropbox.com/s/5wdghka935rn0o3/Screenshot%202021-06-23%20at%2007.52.23.png?dl=0

The rest of the various sensors seem to be ok.  Will capture some logs based on your request:

```
logger:
  default: warn
  logs:
    homeassistant.core: debug
    custom_components.evohome_cc: warn
    ramses_rf.transport: info
When a climate entity becomes unavailable, PM me the log:

Copy to clipboard
cat homeassistant.log | grep -E '(climate|evohome|ramses)'
```

iMiMx, 2021-06-23:
Around 15 minutes after a restart (with restore_state: false) I have 'unknown' state - my assumption was that this should have been relearnt/overwritten within 5 minutes?

https://www.dropbox.com/s/x0iq0b3htelovhm/Screenshot%202021-06-23%20at%2011.00.19.png?dl=0

[15_minute_after_restart.log](https://github.com/zxdavb/ramses_rf/files/6700910/15_minute_after_restart.log)

iMiMx, 2021-06-23:
If I 'nudge' the climate entity up or down, temperature wise, it correctly shows the 'Heat' rather than 'unknown'.

Will wait to see how long, or if, it takes to go back to 'unknown' and will send another log.

iMiMx, 2021-06-24:
Somewhat annoying, after I set restore_state to false and restarted, the main heating zones haven't gone back to unknown yet - the hot water has.

zxdavb, 2021-08-17:
Closing as I beleive is no longer an issue - please let me know if that is not the case.

closed 2021-08-17 by zxdavb