ramses-rf/ramses_rf #37: Running on MacOS

jnewc, 2021-07-25:
Just wanted to let you know that I got this working on MacOS, however I had to make a few changes.

1. Installed `click` library manually using `pip` - not sure if this can be added to the requirements.txt (python is not my strength)
2. Received the following error: `NotImplementedError: Low latency not supported on this platform` - worked around by commenting out the `ser_instance.set_low_latency_mode(True)  # only for FTDI?` block.

Thanks for the great project!

zxdavb, 2021-07-26:
Thanks...

The **click** library is used by the CLI tool, **`client.py`**, which is not part of **ramses_rf** (but which uses it), so is in **`requirements-dev.txt`** rather than **`requirements.txt`**.

I think I will change the whole code block like so:
```python
    try:  # FTDI on Posix/Linux would be a common environment for this library...
        ser_instance.set_low_latency_mode(True)
    except (AttributeError, NotImplementedError, ValueError):
        # AttributeError: not supported on this OS (i.e. not Posix)
        # NotImplemented: not supported on this Platform (e.g. Darwin)
        # ValueError:     Failed to update ASYNC_LOW_LATENCY (i.e. not FTDI)
        pass
```

... is there any chance you could test that & report back?

zxdavb, 2021-08-17:
I believe this has been addressed - please re-open if it is not fixed in the next release (>0.11.x).

closed 2021-08-17 by zxdavb