ramses-rf/ramses_rf #38: Setting Schedules

smar000, 2021-10-12:
Hi David

I was trying out the settings functionality on the current master, using the `client.py` cli. Getting schedules worked fine, but setting schedules has been giving me a few errors as below:

`python ./client.py execute --set-schedule "01:139901" /tmp/sched.jsn /dev/evoqinheng

client.py: Starting ramses_rf...
08:34:02.182 Allowing packets with device_id: 18:134436 (is active gateway?), configure the known_list/block_list as required
08:34:02.183 Won't create a non-allowed device_id: 18:134436 (if required, add it to the known_list)
08:34:02.181 || HGI:134436 | NUL:------ |  I | puzzle_packet    |      || {'ramses_rf': 'v0.14.24', '_parser': 'v0.14.24'}
08:34:02.216 || HGI:134436 | CTL:139901 |  W | zone_schedule    | 0041 || {'zone_idx': '00', 'frag_index': 65, 'frag_total': 1, 'frag_length': 0, 'fragment': '04'}
08:34:02.552 || HGI:134436 | NUL:------ |  I | puzzle_packet    |      || {'ramses_rf': 'v0.14.24', '_parser': 'v0.14.24'}
08:34:03.192 || HGI:134436 | CTL:139901 |  W | zone_schedule    | 0041 || {'zone_idx': '00', 'frag_index': 65, 'frag_total': 1, 'frag_length': 0, 'fragment': '04'}
08:34:04.426 || HGI:134436 | CTL:139901 |  W | zone_schedule    | 0041 || {'zone_idx': '00', 'frag_index': 65, 'frag_total': 1, 'frag_length': 0, 'fragment': '04'}
08:34:06.859 MsgTransport._pkt_receiver(0404| I|01:139901|0041): Expired callback
08:34:06.859 Exception in callback to message layer
Traceback (most recent call last):
  File "/opt/scripts/evohome/ramses_rf.master/ramses_rf/protocol/transport.py", line 414, in _pkt_received
    self._callback(pkt)  # only wanted PKTs to the MSG transport's handler
  File "/opt/scripts/evohome/ramses_rf.master/ramses_rf/protocol/protocol.py", line 165, in _pkt_receiver
    callback[FUNC](False, *callback.get(ARGS, tuple()))
  File "/opt/scripts/evohome/ramses_rf.master/ramses_rf/protocol/schedule.py", line 336, in tx_callback
    msg.payload.get(FRAG_INDEX),
AttributeError: 'bool' object has no attribute 'payload'
08:34:09.304 MsgTransport._pkt_receiver(0404| I|01:139901|0041): Expired callback
08:34:09.304 Exception in callback to message layer
Traceback (most recent call last):
  File "/opt/scripts/evohome/ramses_rf.master/ramses_rf/protocol/transport.py", line 414, in _pkt_received
    self._callback(pkt)  # only wanted PKTs to the MSG transport's handler
  File "/opt/scripts/evohome/ramses_rf.master/ramses_rf/protocol/protocol.py", line 165, in _pkt_receiver
    callback[FUNC](False, *callback.get(ARGS, tuple()))
  File "/opt/scripts/evohome/ramses_rf.master/ramses_rf/protocol/schedule.py", line 336, in tx_callback
    msg.payload.get(FRAG_INDEX),
AttributeError: 'bool' object has no attribute 'payload'
08:34:11.719 PktProtocolQos.send_data(0404| W|01:139901|0041): boff=3, want=0404| I|01:139901|0041, tout=2021-10-12T08:34:11.704: EXPIRED (5/5)
08:34:11.719 PktProtocolQos.send_data(0404| W|01:139901|0041): Expired callback
08:34:11.719 PktProtocolQos.send_data(7FFF| I|63:262142): boff=3, want=0404| I|01:139901|0041, tout=2021-10-12T08:34:11.704: EXPIRED (5/5)
08:34:11.720 PktProtocolQos.send_data(0404| W|01:139901|0041): Expired callback
08:34:11.720 Task exception was never retrieved
future: <Task finished name='Task-14' coro=<PacketProtocolQos.send_data() done, defined at /opt/scripts/evohome/ramses_rf.master/ramses_rf/protocol/transport.py:722> exception=AttributeError("'bool' object has no attribute 'payload'")>
Traceback (most recent call last):
  File "/opt/scripts/evohome/ramses_rf.master/ramses_rf/protocol/transport.py", line 787, in send_data
    _expired_cmd(self._qos_cmd)
  File "/opt/scripts/evohome/ramses_rf.master/ramses_rf/protocol/transport.py", line 741, in _expired_cmd
    callback[FUNC](False, *callback.get(ARGS, tuple()))
  File "/opt/scripts/evohome/ramses_rf.master/ramses_rf/protocol/schedule.py", line 336, in tx_callback
    msg.payload.get(FRAG_INDEX),
AttributeError: 'bool' object has no attribute 'payload'
08:34:20.574 Blocking packets with device_id: 10:142568 (is not whitelisted), if required, add it to the known_list
08:34:41.194 Blocking packets with device_id: 34:112193 (is not whitelisted), if required, add it to the known_list
08:34:50.817 RQ --- 18:136712 10:142568 --:------ 3EF1 001 00 < There appears to be more than one HGI80-compatible device (active gateway: 18:134436), this is unsupported
08:34:50.817 Blocking packets with device_id: 18:136712 (is not whitelisted), if required, add it to the known_list
08:35:01.810 set_schedule(): Function timed out: The callback has expired: failed to set schedule (no hint)

client.py: Finished ramses_rf, results:

client.py: Finished ramses_rf.
 - ended without error (e.g. EOF)

08:35:01.810 Task exception was never retrieved
future: <Task finished name='Task-2' coro=<Gateway.start() done, defined at /opt/scripts/evohome/ramses_rf.master/ramses_rf/__init__.py:155> exception=AttributeError("'bool' object has no attribute 'payload'")>
Traceback (most recent call last):
  File "/opt/scripts/evohome/ramses_rf.master/ramses_rf/__init__.py", line 188, in start
    await asyncio.gather(*self._tasks)
  File "/opt/scripts/evohome/ramses_rf.master/ramses_rf/protocol/protocol.py", line 126, in pkt_dispatcher
    await call_send_data(cmd)
  File "/opt/scripts/evohome/ramses_rf.master/ramses_rf/protocol/protocol.py", line 108, in call_send_data
    await self._dispatcher(cmd)  # send_data, *once* callback registered
  File "/opt/scripts/evohome/ramses_rf.master/ramses_rf/protocol/transport.py", line 787, in send_data
    _expired_cmd(self._qos_cmd)
  File "/opt/scripts/evohome/ramses_rf.master/ramses_rf/protocol/transport.py", line 741, in _expired_cmd
    callback[FUNC](False, *callback.get(ARGS, tuple()))
  File "/opt/scripts/evohome/ramses_rf.master/ramses_rf/protocol/schedule.py", line 336, in tx_callback
    msg.payload.get(FRAG_INDEX),
AttributeError: 'bool' object has no attribute 'payload'`

The test `sched.jsn` file contains:
`{"schedule":[{"day_of_week":0,"switchpoints":[{"heat_setpoint":21.5,"time_of_day":"07:30"},{"heat_setpoint":22.5,"time_of_day":"10:10"},{"heat_setpoint":22.5,"time_of_day":"16:30"},{"heat_setpoint":23.0,"time_of_day":"18:00"},{"heat_setpoint":5.0,"time_of_day":"23:00"}]},{"day_of_week":1,"switchpoints":[{"heat_setpoint":21.5,"time_of_day":"07:30"},{"heat_setpoint":22.5,"time_of_day":"10:10"},{"heat_setpoint":22.5,"time_of_day":"16:30"},{"heat_setpoint":23.0,"time_of_day":"18:00"},{"heat_setpoint":5.0,"time_of_day":"23:00"}]},{"day_of_week":2,"switchpoints":[{"heat_setpoint":21.5,"time_of_day":"07:30"},{"heat_setpoint":22.5,"time_of_day":"10:10"},{"heat_setpoint":22.5,"time_of_day":"16:30"},{"heat_setpoint":23.0,"time_of_day":"18:00"},{"heat_setpoint":5.0,"time_of_day":"23:00"}]},{"day_of_week":3,"switchpoints":[{"heat_setpoint":21.5,"time_of_day":"07:30"},{"heat_setpoint":22.5,"time_of_day":"10:10"},{"heat_setpoint":22.5,"time_of_day":"16:30"},{"heat_setpoint":23.0,"time_of_day":"18:00"},{"heat_setpoint":5.0,"time_of_day":"23:00"}]},{"day_of_week":4,"switchpoints":[{"heat_setpoint":21.5,"time_of_day":"07:30"},{"heat_setpoint":22.5,"time_of_day":"10:10"},{"heat_setpoint":22.5,"time_of_day":"16:30"},{"heat_setpoint":23.0,"time_of_day":"18:00"},{"heat_setpoint":5.0,"time_of_day":"23:00"}]},{"day_of_week":5,"switchpoints":[{"heat_setpoint":20.0,"time_of_day":"07:30"},{"heat_setpoint":20.0,"time_of_day":"09:00"},{"heat_setpoint":22.5,"time_of_day":"11:10"},{"heat_setpoint":23.5,"time_of_day":"18:00"},{"heat_setpoint":5.0,"time_of_day":"22:30"}]},{"day_of_week":6,"switchpoints":[{"heat_setpoint":20.0,"time_of_day":"07:30"},{"heat_setpoint":20.0,"time_of_day":"09:00"},{"heat_setpoint":22.5,"time_of_day":"11:10"},{"heat_setpoint":23.0,"time_of_day":"18:00"},{"heat_setpoint":5.0,"time_of_day":"21:50"}]}],"zone_idx":"00"}`

Is it something I'm doing, or is this still a work in progress?

Many thanks!

zxdavb, 2021-10-12:
@smar000 You don't mention which version of the library you are using - 'current master' is a good hint, but things can/do change very quickly... There have been a lot of changes recently...

I should start by admitting I have no automated test suite...

In short, this used to work fine. However, it was never robust, and was more proof-of-concept. It is broken now, only because of changes made elsewhere in the library.

I will look at it, get it working again and let you know.

zxdavb, 2021-10-12:
OK, look now - note the out put of:
```
python client.py execute /dev/ttyACM0 --get-schedule 01:145038 01
```
... has changed recently. That is, check the structure of your **schedule.json** file.

Then see:
```
python client.py execute /dev/ttyACM0 --set-schedule 01:145038 schedule.json

client.py: Starting ramses_rf...
22:52:51.784 Allowing packets with device_id: 18:006402 (is active gateway?), configure the known_list/block_list as required
22:52:51.785 Won't create a non-allowed device_id: 18:006402 (if required, add it to the known_list)
22:52:51.783 || HGI:006402 | NUL:------ |  I | puzzle_packet    |      || {'ramses_rf': 'v0.14.24', '_parser': 'v0.14.24'}
22:52:51.867 || HGI:006402 | CTL:145038 |  W | zone_schedule    | 0101 || {'zone_idx': '01', 'frag_index': 1, 'frag_total': 3, 'frag_length': 41, 'fragment': '68DE6DCCB10D80300C0551DB710C03515052310303300B73320A440111E0527D9D9C
22:52:51.876 || CTL:145038 | HGI:006402 |  I | zone_schedule    | 0101 || {'zone_idx': '01', 'frag_index': 1, 'frag_total': 3, 'frag_length': 41, 'fragment': ''}
22:52:51.938 || HGI:006402 | CTL:145038 |  W | zone_schedule    | 0101 || {'zone_idx': '01', 'frag_index': 1, 'frag_total': 3, 'frag_length': 41, 'fragment': '68DE6DCCB10D80300C0551DB710C03515052310303300B73320A440111E0527D9D9C
22:52:51.967 || HGI:006402 | CTL:145038 |  W | zone_schedule    | 0102 || {'zone_idx': '01', 'frag_index': 2, 'frag_total': 3, 'frag_length': 41, 'fragment': 'A1AFFB6EFB39FAEEDD46FFDFCD59648AA729780A9E82A7E01978069E8167E025F012
22:52:52.343 || HGI:006402 | CTL:145038 |  W | zone_schedule    | 0102 || {'zone_idx': '01', 'frag_index': 2, 'frag_total': 3, 'frag_length': 41, 'fragment': 'A1AFFB6EFB39FAEEDD46FFDFCD59648AA729780A9E82A7E01978069E8167E025F012
22:52:52.351 || CTL:145038 | HGI:006402 |  I | zone_schedule    | 0102 || {'zone_idx': '01', 'frag_index': 2, 'frag_total': 3, 'frag_length': 41, 'fragment': ''}
22:52:52.412 || HGI:006402 | CTL:145038 |  W | zone_schedule    | 0103 || {'zone_idx': '01', 'frag_index': 3, 'frag_total': 3, 'frag_length': 31, 'fragment': 'E039780E5EBEFEB677A52DF66F5FAFB4F5E30578015E80178D77006F8713D1'}
22:52:52.431 || CTL:145038 | HGI:006402 |  I | zone_schedule    | 0103 || {'zone_idx': '01', 'frag_index': 3, 'frag_total': 0, 'frag_length': 31, 'fragment': ''}
```

It's a bit messy, but is working again.

smar000, 2021-10-13:
Thanks and understood!

zxdavb, 2021-10-17:
@smar000 Unrelated, but your system is one of the few that has these `11`s in the `3EF0` payload:
```
RP --- 10:142568 01:139901 --:------ 3EF0 006 0025110500FF
```
Now that it's Winter, a 24h log from you, including a discovery, could be really inciteful.