ramses-rf/ramses_rf #42: cannot start client.py in monitor mode: getting a TypeError

Marnix-vb, 2021-12-20:
I can run:
pi@raspberrypi:~/ramses_rf $ python client.py listen /dev/ttyUSB0 -o packet.log

but if i run:
pi@raspberrypi:~/ramses_rf $ python client.py monitor /dev/ttyUSB0 -o packet.log

it fails,

pi@raspberrypi:~/ramses_rf $ python -V
Python 3.9.2
on Raspbian GNU/Linux 11 (bullseye)

Traceback (most recent call last):
File "/home/pi/ramses_rf/client.py", line 473, in
cli()
File "/usr/lib/python3/dist-packages/click/core.py", line 829, in call
return self.main(*args, **kwargs)
File "/usr/lib/python3/dist-packages/click/core.py", line 782, in main
rv = self.invoke(ctx)
File "/usr/lib/python3/dist-packages/click/core.py", line 1257, in invoke
sub_ctx = cmd.make_context(cmd_name, args, parent=ctx)
File "/usr/lib/python3/dist-packages/click/core.py", line 700, in make_context
self.parse_args(ctx, args)
File "/usr/lib/python3/dist-packages/click/core.py", line 1048, in parse_args
value, args = param.handle_parse_result(ctx, opts, args)
File "/usr/lib/python3/dist-packages/click/core.py", line 1623, in handle_parse_result
value = self.full_process_value(ctx, value)
File "/usr/lib/python3/dist-packages/click/core.py", line 1965, in full_process_value
return Parameter.full_process_value(self, ctx, value)
File "/usr/lib/python3/dist-packages/click/core.py", line 1592, in full_process_value
value = self.get_default(ctx)
File "/usr/lib/python3/dist-packages/click/core.py", line 1917, in get_default
return Parameter.get_default(self, ctx)
File "/usr/lib/python3/dist-packages/click/core.py", line 1534, in get_default
return self.type_cast_value(ctx, rv)
File "/usr/lib/python3/dist-packages/click/core.py", line 1561, in type_cast_value
return self.type(value or (), self, ctx)
File "/usr/lib/python3/dist-packages/click/types.py", line 46, in call
return self.convert(value, param, ctx)
File "/usr/lib/python3/dist-packages/click/types.py", line 681, in convert
raise TypeError(
TypeError: It would appear that nargs is set to conflict with the composite type arity.

Thanks for looking into this!

Marnix

zxdavb, 2021-12-20:
I cannot reprodue this error:
```
(venv) dbonnes@vm-builder ~/c/ramses_rf (master)> python client.py -ns -rr  monitor /dev/ttyUSB0 -x "RQ 01:145038 1F09 00"

client.py: Starting ramses_rf...
14:16:14.980 It is strongly recommended to provide a known_list, and use it as a whitelist (device_id filter), configure: enforce_known_list = True
14:16:15.056 Not using any device filter: using a known_list (as a whitelist) is strongly recommended)
14:16:15.198 || HGI:013393 | NUL:262142 |  I | puzzle_packet    |      || {'datetime': '2021-12-20T14:16:15.056', 'engine': 'v0.16.26', 'parser': 'v0.16.26'}
14:16:15.227 || HGI:013393 | CTL:145038 | RQ | system_sync      |      || {}
14:16:15.240 || CTL:145038 | HGI:013393 | RP | system_sync      |      || {'remaining_seconds': 137.0, '_next_sync': '14:18:32'}
^C
client.py: Finished ramses_rf, results:

client.py: Finished ramses_rf.
 - ended via: CancelledError (e.g. SIGINT)

Aborted!
```

Have a look at **requirements.txt**:
```
(venv) dbonnes@vm-builder ~/c/ramses_rf (master)> pip list | grep -E '(serial|click)'
click                             8.0.3
pyserial                          3.5
pyserial-asyncio                  0.6
```

zxdavb, 2021-12-20:
```
(venv) dbonnes@vm-builder ~/c/ramses_rf (master)> python client.py monitor /dev/ttyUSB0 -o packet.log

client.py: Starting ramses_rf...
14:17:32.423 It is strongly recommended to provide a known_list, and use it as a whitelist (device_id filter), configure: enforce_known_list = True
14:17:32.499 Not using any device filter: using a known_list (as a whitelist) is strongly recommended)
14:17:32.543 Creating a Device: 18:013393 (<class 'ramses_rf.devices.HgiGateway'>)
14:17:32.542 || HGI:013393 | NUL:262142 |  I | puzzle_packet    |      || {'datetime': '2021-12-20T14:17:32.500', 'engine': 'v0.16.26', 'parser': 'v0.16.26'}
14:17:37.021 RQ --- 18:006402 10:048122 --:------ 3EF0 001 00 < There appears to be more than one HGI80-compatible device (active gateway: 18:013393), this is unsupported
14:17:37.021 Blocking packets with device_id: 18:006402 (is foreign gateway), configure the known_list/block_list as required
14:17:49.549 Creating a Device: 32:168090 (<class 'ramses_rf.devices.HvacHumidity'>)
14:17:49.548 || VMS:168090 | RFG:082155 |  I | ext_ventilation  |      || {'active': False, '_unknown_0': '0000', 'unknown_2': '00'}
14:17:49.594 || HGI:013393 | VMS:168090 | RQ | device_info      |      || {}
14:17:49.694 || HGI:013393 | VMS:168090 | RQ | device_info      |      || {}
14:17:50.099 || HGI:013393 | VMS:168090 | RQ | device_info      |      || {}
14:17:50.141 Creating a Device: 10:048122 (<class 'ramses_rf.devices.OtbGateway'>)
14:17:50.140 || OTB:048122 |            |  I | opentherm_sync   |      || {'ticker': 73}
14:17:50.762 Creating a Device: 12:207082 (<class 'ramses_rf.devices.Thermostat'>)
14:17:50.762  I --- --:------ --:------ 12:207082 1F09 003 010B77 # IS_CONTROLLER (00): is TRUE
14:17:50.762  I --- --:------ --:------ 12:207082 1F09 003 010B77 < AttributeError('Thermostat' object has no attribute '_evo')
14:17:50.761 ||            | THm:207082 |  I | system_sync      | (01) || {'remaining_seconds': 293.5, '_next_sync': '14:22:44'}
14:17:50.774 ||            | THm:207082 |  I | setpoint         |  01  || {'zone_idx': '01', 'setpoint': False}
14:17:50.785 ||            | THm:207082 |  I | temperature      |  01  || {'zone_idx': '01', 'temperature': 17.09}
^C
client.py: Finished ramses_rf, results:
```

Marnix-vb, 2021-12-21:
Thanks for taking the time, and you fixed it for me!

click was still on version 7

pi@raspberrypi:~ $ pip list | grep -E '(serial|click)'
click             7.1.2
pyserial          3.5b0
pyserial-asyncio  0.6

pi@raspberrypi:~/ramses_rf $ sudo pip install click --upgrade
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: click in /usr/lib/python3/dist-packages (7.1.2)
Collecting click
  Downloading https://www.piwheels.org/simple/click/click-8.0.3-py3-none-any.whl (97 kB)
     |████████████████████████████████| 97 kB 324 kB/s
Installing collected packages: click
  Attempting uninstall: click
    Found existing installation: click 7.1.2
    Not uninstalling click at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'click'. No files were found to uninstall.
Successfully installed click-8.0.3

pi@raspberrypi:~/ramses_rf $ pip list | grep -E '(serial|click)'
click             8.0.3
pyserial          3.5b0
pyserial-asyncio  0.6

pi@raspberrypi:~/ramses_rf $ python client.py -ns -rr  monitor /dev/ttyUSB0 -x "RQ 01:145038 1F09 00"

client.py: Starting ramses_rf...
09:21:44.184 It is strongly recommended to provide a known_list, and use it as a whitelist (device_id filter), configure: enforce_known_list = True
09:21:44.237 Not using any device filter: using a known_list (as a whitelist) is strongly recommended)
09:21:44.830 || HGI:137235 | NUL:262142 |  I | puzzle_packet    |      || {'datetime': '2021-12-21T09:21:44.244', 'engine': 'v0.16.26', 'parser': 'v0.16.26'}
09:21:44.890 || HGI:137235 | CTL:145038 | RQ | system_sync      |      || {}
09:21:44.992 || HGI:137235 | CTL:145038 | RQ | system_sync      |      || {}
09:21:45.132 || STA:194767 |            |  I | relay_demand     |      || {'relay_demand': 0.0}
09:21:45.395 || HGI:137235 | CTL:145038 | RQ | system_sync      |      || {}
09:21:47.000 || HGI:137235 | CTL:145038 | RQ | system_sync      |      || {}
^C
client.py: Finished ramses_rf, results:

client.py: Finished ramses_rf.
 - ended via: CancelledError (e.g. SIGINT)

Aborted!

Marnix

closed 2021-12-21 by Marnix-vb