ramses-rf/ramses_rf #45: Orcon ventilation: 31D9 payload not recognized, wrong fan modes

maleadt, 2022-05-09:
I'm looking into using this (awesome!) software with my Orcon HRC-350 (sold to me under the brand Ventiline) through an Arduino running evofw3, and am encountering a couple of issues.

The first is a payload that isn't recognized. When I set the ventilation fan speed to respectively away, 1, 2,  3, or auto, encoded as fan modes 0 to 4, my unit responds with a 31D9 message where the payload is simply either 00000000, 00000100, 00000200,  00000300 or 00000400. That doesn't match the regex in `ramses.py`, which seems to expect much more data:

```
23:13:48.670 || HVC:150155 |  32:134044 |  I | switch_speed     |      || {'rate': 0.0, '_step_idx': 0, '_step_max': 4, 'fan_mode': 'standby'}
23:13:48.689  I --- 32:134044 --:------ 32:134044 31D9 004 00000000 < Corrupt payload: Payload doesn't match '^(00|01|21)[0-9A-F]{4}((00|FE)(00|20){12}(00|08))?$': 00000000

23:04:28.570 || HVC:150155 | HVC:134044 |  I | switch_speed     |      || {'rate': 0.25, '_step_idx': 1, '_step_max': 4, 'fan_mode': 'auto'}
23:04:28.598  I --- 32:134044 --:------ 32:134044 31D9 004 00000100 < Corrupt payload: Payload doesn't match '^(00|01|21)[0-9A-F]{4}((00|FE)(00|20){12}(00|08))?$': 00000100

23:04:34.219 || HVC:150155 | HVC:134044 |  I | switch_speed     |      || {'rate': 0.5, '_step_idx': 2, '_step_max': 4, 'fan_mode': 'low'}
23:04:34.251  I --- 32:134044 --:------ 32:134044 31D9 004 00000200 < Corrupt payload: Payload doesn't match '^(00|01|21)[0-9A-F]{4}((00|FE)(00|20){12}(00|08))?$': 00000200

23:04:38.197 || HVC:150155 | HVC:134044 |  I | switch_speed     |      || {'rate': 0.75, '_step_idx': 3, '_step_max': 4, 'fan_mode': 'medium'}
23:04:38.223  I --- 32:134044 --:------ 32:134044 31D9 004 00000300 < Corrupt payload: Payload doesn't match '^(00|01|21)[0-9A-F]{4}((00|FE)(00|20){12}(00|08))?$': 00000300

23:13:51.928 || HVC:150155 |  32:134044 |  I | switch_speed     |      || {'rate': 1.0, '_step_idx': 4, '_step_max': 4, 'fan_mode': 'high'}
23:13:51.956  I --- 32:134044 --:------ 32:134044 31D9 004 00000400 < Corrupt payload: Payload doesn't match '^(00|01|21)[0-9A-F]{4}((00|FE)(00|20){12}(00|08))?$': 00000400
```

This also shows that the fan modes are not mapped correctly: 0 is away, 1-3 match fan speed 1 to 3, and 4 should be auto (fan speed determined by air humidity).

The much more complicated 31DA packets are fully recognized though 🙂 The rendering in the CLI is cut off though (maybe it could drop fields that are known to be bogus for this model?):

```
23:13:55.251 || HVC:134044 |            |  I | vent_31da        |      || {'exhaust_fan_speed': 0.15, 'remaining_time': 0.0, 'co2_level': None, 'indoor_humidity': 0.51, 'air_quality': None, 'air_quality_b
```

Finally, I also encountered the following packet with a prompt to file an issue:

```
23:13:59.371 RP --- 32:134044 18:201242 --:------ 10E0 038 000001C87D140D67FEFFFFFFFFFF170C07E3564D442D3135524D533634000000000000000000 < Support the development of ramses_rf by reporting this packet, with the make/model of device: 32:134044 (device type 32 not known to have signature: 0001C87D140D67FEFF)
```

Let me know if you need additional information.

maleadt, 2022-05-10:
Here's a more complete trace, trying all the buttons on my remote:

```
client.py: Starting ramses_rf...
11:08:16.605 It is strongly recommended to provide a known_list, and use it as a whitelist (device_id filter), configure: enforce_known_list = True
11:08:16.608 Not using any device filter: using a known_list (as a whitelist) is strongly recommended)
11:08:16.641 || HGI:201242 | NUL:262142 |  I | puzzle_packet    |      || {'datetime': '2022-05-10T11:08:16.608', 'engine': 'v0.20.0', 'parser': 'v0.20.0'}

# power-on device
11:08:21.617 || HVC:134044 |            |  I | message_042f     |      || {'counter_1': '0x0104', 'counter_3': '0x0004', 'counter_5': '0x0105', 'unknown_7': '0x5E30'}
11:08:21.663 || HGI:201242 | HVC:134044 | RQ | device_info      |      || {}
11:08:21.759  I --- 32:134044 63:262142 --:------ 10E0 038 000001C87D140D67FEFFFFFFFFFF170C07E3564D442D3135524D533634000000000000000000 < Support the development of ramses_rf by reporting this packet, with the make/model of device: 32:134044 (device type 32 not known to have signature: 0001C87D140D67FEFF)
11:08:21.758 || HVC:134044 | NUL:262142 |  I | device_info      |      || {'date_2': '0000-00-00', 'date_1': '2019-12-23', 'product_id': '7D', 'oem_code': '67', 'description': 'VMD-15RMS64'}
11:08:21.861 || HGI:201242 | HVC:134044 | RQ | device_info      |      || {}
11:08:23.723 || HGI:201242 | HVC:134044 | RQ | device_info      |      || {}
11:08:25.705 PktProtocolQos.send_data(sent=10E0|RQ|32:134044): boff=3, want=10E0|RP|32:134044, tout=2022-05-10T11:08:25.701: QoS: IS_EXPIRED (giving up) (2/2)
11:08:42.494  I --- 32:134044 --:------ 32:134044 31D9 004 000001FE < Corrupt payload: Payload doesn't match '^(00|01|21)[0-9A-F]{4}((00|FE)(00|20){12}(00|08))?$': 000001FE
11:08:51.537 || HVC:134044 |            |  I | vent_31da        |      || {'exhaust_fan_speed': 0.15, 'remaining_time': 0.0, 'co2_level': None, 'indoor_humidity': 0.52, 'air_quality': None, 'air_quality_b

# press 'away'
11:09:14.037 || HVC:150155 | HVC:134044 |  I | switch_speed     |      || {'rate': 0.0, '_step_idx': 0, '_step_max': 4, 'fan_mode': 'standby'}
11:09:14.062  I --- 32:134044 --:------ 32:134044 31D9 004 000000FE < Corrupt payload: Payload doesn't match '^(00|01|21)[0-9A-F]{4}((00|FE)(00|20){12}(00|08))?$': 000000FE
11:09:14.100 || HGI:201242 | HVC:150155 | RQ | device_info      |      || {}
11:09:14.595 || HVC:134044 |            |  I | vent_31da        |      || {'exhaust_fan_speed': 0.1, 'remaining_time': 0.0, 'co2_level': None, 'indoor_humidity': 0.52, 'air_quality': None, 'air_quality_ba
11:09:16.105 || HGI:201242 | HVC:150155 | RQ | device_info      |      || {}
11:09:18.111 || HGI:201242 | HVC:150155 | RQ | device_info      |      || {}
11:09:20.090 PktProtocolQos.send_data(sent=10E0|RQ|29:150155): boff=3, want=10E0|RP|29:150155, tout=2022-05-10T11:09:20.089: QoS: IS_EXPIRED (giving up) (2/2)

# press '1'
11:09:24.227 || HVC:150155 | HVC:134044 |  I | switch_speed     |      || {'rate': 0.25, '_step_idx': 1, '_step_max': 4, 'fan_mode': 'auto'}
11:09:24.258  I --- 32:134044 --:------ 32:134044 31D9 004 000001FE < Corrupt payload: Payload doesn't match '^(00|01|21)[0-9A-F]{4}((00|FE)(00|20){12}(00|08))?$': 000001FE
11:09:24.610 || HVC:134044 |            |  I | vent_31da        |      || {'exhaust_fan_speed': 0.15, 'remaining_time': 0.0, 'co2_level': None, 'indoor_humidity': 0.52, 'air_quality': None, 'air_quality_b

# press '2'
11:09:32.185 || HVC:150155 | HVC:134044 |  I | switch_speed     |      || {'rate': 0.5, '_step_idx': 2, '_step_max': 4, 'fan_mode': 'low'}
11:09:32.214  I --- 32:134044 --:------ 32:134044 31D9 004 000002FE < Corrupt payload: Payload doesn't match '^(00|01|21)[0-9A-F]{4}((00|FE)(00|20){12}(00|08))?$': 000002FE

# press '3'
11:09:39.010 || HVC:150155 | HVC:134044 |  I | switch_speed     |      || {'rate': 0.75, '_step_idx': 3, '_step_max': 4, 'fan_mode': 'medium'}
11:09:39.038  I --- 32:134044 --:------ 32:134044 31D9 004 000003FE < Corrupt payload: Payload doesn't match '^(00|01|21)[0-9A-F]{4}((00|FE)(00|20){12}(00|08))?$': 000003FE
11:09:44.656 || HVC:134044 |            |  I | vent_31da        |      || {'exhaust_fan_speed': 0.55, 'remaining_time': 0.0, 'co2_level': None, 'indoor_humidity': 0.52, 'air_quality': None, 'air_quality_b

# press '4'
11:09:49.253 || HVC:150155 | HVC:134044 |  I | switch_speed     |      || {'rate': 1.0, '_step_idx': 4, '_step_max': 4, 'fan_mode': 'high'}
11:09:49.274  I --- 32:134044 --:------ 32:134044 31D9 004 000004FE < Corrupt payload: Payload doesn't match '^(00|01|21)[0-9A-F]{4}((00|FE)(00|20){12}(00|08))?$': 000004FE
11:09:54.667 || HVC:134044 |            |  I | vent_31da        |      || {'exhaust_fan_speed': 0.15, 'remaining_time': 0.0, 'co2_level': None, 'indoor_humidity': 0.52, 'air_quality': None, 'air_quality_b

# press 'timer'
11:09:58.896 || HVC:150155 | HVC:134044 |  I | switch_duration  |      || {'flags': [0, 0, 0, 0, 0, 0, 1, 0], 'minutes': 15, '_new_speed_mode': 'per_vent_speed', '_fallback_speed_mode': None, 'rate': 0.75
11:09:58.919  I --- 32:134044 --:------ 32:134044 31D9 004 000003FE < Corrupt payload: Payload doesn't match '^(00|01|21)[0-9A-F]{4}((00|FE)(00|20){12}(00|08))?$': 000003FE
11:09:59.687 || HVC:134044 |            |  I | vent_31da        |      || {'exhaust_fan_speed': 0.55, 'remaining_time': 15.0, 'co2_level': None, 'indoor_humidity': 0.52, 'air_quality': None, 'air_quality_
```

The replies are now different (000001FE instead of 00000100, etc).

[packet.log](https://github.com/zxdavb/ramses_rf/files/8660115/packet.log)

zxdavb, 2022-05-10:
The issue is that none of this protocol is documented.

I have updated the code, so that it can parse your packet log without error/warning. I will be dropping a refactored build soon.

The **client.py** is for ad-hoc, you can:
```
cat packet.log | python client.py parse > msg.out
```

I will reach out to you shortly, when I have capacity

maleadt, 2022-05-11:
Thanks, confirmed that current master parses the `vent_status` messages now (although it decodes the fan mode as `exhaust_fan_speed`):

```
09:13:39.893 || HVC:134044 |            |  I | vent_status      |      || {'flags': [0, 0, 0, 0, 0, 0, 0, 0], 'exhaust_fan_speed': 0.005, 'passive': False, 'damper_only': False, 'filter_dirty': False, 'frost_cycle': False, 'has_fault': False, '_unknown_3': 'FE'}
```

zxdavb, 2022-06-11:
BTW, the fan modes may be wrong for your orcon-compatible HRU (or the 15RF), but they were OK for the system I had access to before (a Nuaire PIV)...

I think the way forward is to get it working with your system, then use the **`traits`** feature to provide fro any difference.

zxdavb, 2022-06-11:
@maleadt The time has come to get remotes working... Please tell me the make/model of your remote

The latest version of the **client.py** has a new switch, **`-lf`**, which you may like.

zxdavb, 2022-06-11:
@llevering could you tell me the exact make/model of your remote and your vent unit?

llevering, 2022-07-05:
@zxdavb I thought I did reply earlier, but it seems I forget the sent my comment. 

I have an Orcon 15RF simple remote like @maleadt the vent unit is a Orcon MVS 15RHB, with a humidity sensor. I upgraded to 20.9 today.

zxdavb, 2023-05-12:
No response - closing.