ramses-rf/ramses_rf #48: cannot start client.py: TypeError

nl0pvm, 2022-07-20:
python3 client.py monitor /dev/ttyUSB0 -o packet.log
Traceback (most recent call last):
  File "client.py", line 21, in <module>
    from ramses_rf import Gateway, GracefulExit, is_valid_dev_id
  File "/home/pveerman/ramses_rf/ramses_rf/__init__.py", line 16, in <module>
    from .const import __dev_mode__
  File "/home/pveerman/ramses_rf/ramses_rf/const.py", line 9, in <module>
    from .protocol.const import (  # noqa: F401
  File "/home/pveerman/ramses_rf/ramses_rf/protocol/__init__.py", line 9, in <module>
    from .address import Address, is_valid_dev_id
  File "/home/pveerman/ramses_rf/ramses_rf/protocol/address.py", line 24, in <module>
    DEVICE_LOOKUP |= {"NUL": "63", "---": "--"}

zxdavb, 2022-07-20:
Hi, 

You have not completed a full traceback.

You have not said which version of the repo you are using.

The latest version only support Python >3.9.x

nl0pvm, 2022-07-20:
I am sorry for not providing you with these quite essential details. However you even made it to give me (a good) answer as I am running the latest and greatest version on and older version of python. So that's indeed the root cause. Have you got any directions on what version is still python 3.8 compatible?

nl0pvm, 2022-07-27:
Because of a new commit a new error has arisen for which i will create a new issue.

closed 2022-07-27 by nl0pvm