ramses-rf/ramses_rf #56: client.py broken

bheading, 2022-09-24:
Problem seems to have started at commit 981be8346890449a53d462d84d8a5958852ebde5.

I am not sure why client.py is pulling in mock stuff from the test suite. That doesn't seem right. 

```
(venv) bheading@glasnost:~/dev/ramses_rf$ python3 ./client.py
Traceback (most recent call last):
  File "/home/bheading/dev/ramses_rf/./client.py", line 47, in <module>
    from tests_rf.mock import MockGateway
  File "/home/bheading/dev/ramses_rf/tests_rf/mock/__init__.py", line 17, in <module>
    from .device_heat import (  # noqa: F401, F811, pylint: disable=unused-import
  File "/home/bheading/dev/ramses_rf/tests_rf/mock/device_heat.py", line 138, in <module>
    ZoneParamsT = dict[str, None | bool | float | int]
TypeError: unsupported operand type(s) for |: 'NoneType' and 'type'
(venv) bheading@glasnost:~/dev/ramses_rf$
```

zxdavb, 2022-09-27:
**client.py** isn't part of the library, I use it extensively for dev/test work, so it may include dodgy things like pulling in modules from the test suite.

The reason why I am doing that is that parts of the test suite is migrating into a controller (i.e. a replacement for an evohome controller).

The core issue here is that you are probably using an older version of Python, pre 3.10.x. The easiest(?) solution may be to upgrade to Python v3.10.x.

Otherwise, I believe the solution is:
```
from __future__ import annotations
```
Is this fix in the first few lines of your version of **device_heat.py**?

zxdavb, 2022-09-27:
pull the latest master - that may fix it fro you too.

zxdavb, 2022-11-04:
Closing as I believe is fixed.

closed 2022-11-04 by zxdavb