ramses-rf/ramses_rf #57: AssertionError

pbootsma, 2022-10-26:
Logger: ramses_rf.protocol.message
Source: /usr/local/lib/python3.10/site-packages/ramses_rf/protocol/message.py:350 
First occurred: 13:25:17 (222 occurrences) 
Last logged: 14:49:19

W --- 21:041873 02:246902 --:------ 22D0 008 0312001E14030020 < AssertionError(Support the development of ramses_rf by reporting this packet)
I --- 02:246902 21:041873 --:------ 22D0 004 03120000 < AssertionError(Support the development of ramses_rf by reporting this packet)
W --- 21:034832 02:248294 --:------ 22D0 008 0304001E14030020 < AssertionError(Support the development of ramses_rf by reporting this packet)
I --- 02:248294 21:034832 --:------ 22D0 004 03010000 < AssertionError(Support the development of ramses_rf by reporting this packet)
I --- 02:248294 --:------ 02:248294 22D0 004 00010000 < AssertionError(Support the development of ramses_rf by reporting this packet)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/ramses_rf/protocol/message.py", line 331, in _validate
    result = PAYLOAD_PARSERS.get(self.code, parser_unknown)(
  File "/usr/local/lib/python3.10/site-packages/ramses_rf/protocol/parsers.py", line 140, in wrapper
    result = fnc(payload, msg, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/ramses_rf/protocol/parsers.py", line 1425, in parser_22d0
    assert payload in (
AssertionError: Support the development of ramses_rf by reporting this packet

zxdavb, 2022-11-04:
I will tweak the `22D0` parser - but this packet is not well understood at all...

zxdavb, 2023-05-12:
I believe the 22D0 parser has been updated for this - please comment below if I am wrong.

closed 2023-05-12 by zxdavb