ramses-rf/ramses_rf #58: Packet report

pbootsma, 2022-10-26:
HomeAssistant
Loads of Itho devices: Spider, Autotemp, HRU

Logger: ramses_rf.protocol.parsers
Source: runner.py:119 
First occurred: 13:25:03 (32 occurrences) 
Last logged: 14:52:58

I --- 21:034615 --:------ 21:034615 3EF0 006 020000100200 < Support the development of ramses_rf by reporting this packet ([0, 0, 0, 1, 0, 0, 0, 0]), with a description of your system
I --- 21:060034 --:------ 21:060034 3EF0 006 0200001C0200 < Support the development of ramses_rf by reporting this packet ([0, 0, 0, 1, 1, 1, 0, 0]), with a description of your system
I --- 21:041873 --:------ 21:041873 3EF0 006 020000100200 < Support the development of ramses_rf by reporting this packet ([0, 0, 0, 1, 0, 0, 0, 0]), with a description of your system
I --- 21:060034 --:------ 21:060034 3EF0 006 020000100200 < Support the development of ramses_rf by reporting this packet ([0, 0, 0, 1, 0, 0, 0, 0]), with a description of your system
I --- 02:247253 --:------ 02:247253 3120 007 0000B2000000FF < Support the development of ramses_rf by reporting this packet (byte 2: B2)

pbootsma, 2022-11-02:
02: is a Itho Autotemp unit
21: is a Itho Spider thermostat

zxdavb, 2022-11-04:
@pbootsma Can you provide a packet log please?

nl0pvm, 2022-11-04:
If this is of any help to you:

```
18:03:44.965  I --- SpiderLivingroom --:------ SpiderLivingroom 3EF0 006 020000000200 < AssertionError(Invalid payload context: 02)
Traceback (most recent call last):
  File "/home/user/ramsesnewest/ramses_rf/protocol/message.py", line 331, in _validate
    result = PAYLOAD_PARSERS.get(self.code, parser_unknown)(
  File "/home/user/ramsesnewest/ramses_rf/protocol/parsers.py", line 141, in wrapper
    result = fnc(payload, msg, **kwargs)
  File "/home/user/ramsesnewest/ramses_rf/protocol/parsers.py", line 2442, in parser_3ef0
    assert payload[:2] == "00", f"Invalid payload context: {payload[:2]}"
AssertionError: Invalid payload context: 02
```

packetlog:
`2022-11-04T18:03:44.965165 067  I --- 21:04XXXX --:------ 21:04XXXX 3EF0 006 020000000200`

pbootsma, 2022-11-04:
Complete packetlog of yesterday. Loads of opportunities to improve parsing :).
[packet.log.2022-11-03.zip](https://github.com/zxdavb/ramses_rf/files/9940958/packet.log.2022-11-03.zip)

zxdavb, 2023-12-03:
OK, looking at this packet log now - lots in ther with HVAC, but...

For `3EF0`, we _still_ have:
```
08:56:27.107 RP --- 10:104564 34:209036 --:------ 3EF0 006 00C8110000FF < Coding error: ValueError(Invalid result: 2.0 (0xC8) is > 1)
Traceback (most recent call last):
  File "/home/dbonnes/clients/ramses_rf/src/ramses_tx/message.py", line 259, in _validate
    result = parse_payload(self)
             ^^^^^^^^^^^^^^^^^^^
  File "/home/dbonnes/clients/ramses_rf/src/ramses_tx/parsers.py", line 2848, in parse_payload
    result = _PAYLOAD_PARSERS.get(msg.code, parser_unknown)(msg._pkt.payload, msg)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dbonnes/clients/ramses_rf/src/ramses_tx/parsers.py", line 2432, in parser_3ef0
    mod_level = hex_to_percent(payload[2:4], high_res=False)  # 00-64 (or FF)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dbonnes/clients/ramses_rf/src/ramses_tx/helpers.py", line 320, in hex_to_percent
    raise ValueError(f"Invalid result: {result} (0x{value}) is > 1")
ValueError: Invalid result: 2.0 (0xC8) is > 1
```

This is a Schema change: I had previously thought was 0-100, but is clear the above is 0.200.

zxdavb, 2023-12-03:
The Itho packet now decodes OK:
```
2022-11-03T23:58:48.871733 ...  I --- 21:035113 --:------ 21:035113 3EF0 006 0200000A0200  # {'modulation_level': 0.0, '_flags_2': '00', '_flags_3': [0, 0, 0, 0, 1, 0, 1, 0], 'ch_active': True, 'dhw_active': False, 'cool_active': False, 'flame_on': True, '_unknown_4': '02', '_unknown_5': '00'}
```