ramses-rf/ramses_rf #59: development support 3EF0 and error 3150

epicurei, 2022-11-04:
11:48:21.862 I --- 13:103760 --:------ 13:103760 3EF0 003 0000FF < Support the development of ramses_rf by reporting this packet (byte 5: )

11:48:39.094 I --- 13:107116 --:------ 13:107116 3EF0 003 00C8FF < Support the development of ramses_rf by reporting this packet (byte 5: )

15:17:41.881 I --- 04:065911 --:------ 01:095571 3150 002 03CA < Coding error: ValueError(Invalid result: 1.01 (0xCA) is > 1)
Traceback (most recent call last):
File "/home/supervisor/Scrivania/ramses_rf-master v0.21.6 01_11 mod/ramses_rf/protocol/message.py", line 331, in _validate
result = PAYLOAD_PARSERS.get(self.code, parser_unknown)(
File "/home/supervisor/Scrivania/ramses_rf-master v0.21.6 01_11 mod/ramses_rf/protocol/parsers.py", line 141, in wrapper
result = fnc(payload, msg, **kwargs)
File "/home/supervisor/Scrivania/ramses_rf-master v0.21.6 01_11 mod/ramses_rf/protocol/parsers.py", line 2057, in parser_3150
return valve_demand(payload[2:]) # TODO: check UFC/FC is == CTL/FC
File "/home/supervisor/Scrivania/ramses_rf-master v0.21.6 01_11 mod/ramses_rf/protocol/helpers.py", line 27, in wrapper
return fnc(*args, **kwargs)
File "/home/supervisor/Scrivania/ramses_rf-master v0.21.6 01_11 mod/ramses_rf/protocol/helpers.py", line 328, in valve_demand
raise ValueError(f"Invalid result: {result} (0x{value}) is > 1")
ValueError: Invalid result: 1.01 (0xCA) is > 1

[message trace v0.21.6 -.txt](https://github.com/zxdavb/ramses_rf/files/9936385/message.trace.v0.21.6.-.txt)
[packet.log](https://github.com/zxdavb/ramses_rf/files/9936387/packet.log)
[evohome controllers configurations v0.21.06.txt](https://github.com/zxdavb/ramses_rf/files/9936388/evohome.controllers.configurations.v0.21.06.txt)

zxdavb, 2022-11-04:
There are two issues here - this addresses the first issue...

These are bugs, and I have identified them - they will be corrected in the next release:
```
11:48:21.862 I --- 13:103760 --:------ 13:103760 3EF0 003 0000FF < Support the development of ramses_rf by reporting this packet (byte 5: )
11:48:39.094 I --- 13:107116 --:------ 13:107116 3EF0 003 00C8FF < Support the development of ramses_rf by reporting this packet (byte 5: )
```

zxdavb, 2022-11-04:
This is not a bug - more so it is behaviour that is not understood...
```
15:17:41.881 I --- 04:065911 --:------ 01:095571 3150 002 03CA < Coding error: ValueError(Invalid result: 1.01 (0xCA) is > 1)
```

I am looking into it.

zxdavb, 2022-11-06:
@epicurei 

Reagrding these errors:
```text
11:48:21.862 I --- 13:103760 --:------ 13:103760 3EF0 003 0000FF < Support the development of ramses_rf by reporting this packet (byte 5: )
11:48:39.094 I --- 13:107116 --:------ 13:107116 3EF0 003 00C8FF < Support the development of ramses_rf by reporting this packet (byte 5: )
```

It would be great for you to confirm which release of **ramses_rf** (or *ramses_cc**) you were using.

epicurei, 2022-11-07:
the version is the one indicated in the files name ramses_rf v0.21.6 downloaded on the first of november

 

Questa email è stata esaminata alla ricerca di virus dal software antivirus AVG.
www.avg.com

zxdavb, 2022-11-07:
To be exact: you downloaded it via HACS?

zxdavb, 2022-11-07:
The image has not made it to github - I think because you responded via email.

epicurei, 2022-11-08:
I have no idea what you mean by HACS but here attached image of what I use to download new versions as I did today with 0.22.1
[image]

zxdavb, 2022-11-08:
OK, got you now, thanks.

I suggest you only use tagged versions (unless you want), see:
 - https://github.com/zxdavb/ramses_rf/tags
 
The master branch is very likely to have bugs, so reporting issues against that can be counter-productive.

The latest tagged version is **0.22.1**.

Don't worry: this `ValueError` is a genuine bug.

zxdavb, 2023-05-12:
See #71 - continue there

closed 2023-05-12 by zxdavb