ramses-rf/ramses_rf #60: 4E02 Corrupt payload

nl0pvm, 2022-11-04:
```
18:06:43.916  I --- Autotempslave AutotempMaster --:------ 4E02 034 0008667FFF7FFF7FFF7FFF7FFF7FFF7FFF0408CA7FFF7FFF7FFF7FFF7FFF7FFF7FFF < Corrupt payload: Payload doesn't match '^00([0-9A-F]{4}){8}02([0-9A-F]{4}){8}$': 0008667FFF7FFF7FFF7FFF7FFF7FFF7FFF0408CA7FFF7FFF7FFF7FFF7FFF7FFF7FFF
18:06:44.899  I --- Autotempslave AutotempMaster --:------ 4E02 034 0008667FFF7FFF7FFF7FFF7FFF7FFF7FFF0408CA7FFF7FFF7FFF7FFF7FFF7FFF7FFF < Corrupt payload: Payload doesn't match '^00([0-9A-F]{4}){8}02([0-9A-F]{4}){8}$': 0008667FFF7FFF7FFF7FFF7FFF7FFF7FFF0408CA7FFF7FFF7FFF7FFF7FFF7FFF7FFF
18:06:45.881  I --- Autotempslave AutotempMaster --:------ 4E02 034 0008667FFF7FFF7FFF7FFF7FFF7FFF7FFF0408CA7FFF7FFF7FFF7FFF7FFF7FFF7FFF < Corrupt payload: Payload doesn't match '^00([0-9A-F]{4}){8}02([0-9A-F]{4}){8}$': 0008667FFF7FFF7FFF7FFF7FFF7FFF7FFF0408CA7FFF7FFF7FFF7FFF7FFF7FFF7FFF
```

packetlog:
```
2022-11-04T18:06:43.916064 052  I --- 02:24XXXX 02:24YYYY --:------ 4E02 034 0008667FFF7FFF7FFF7FFF7FFF7FFF7FFF0408CA7FFF7FFF7FFF7FFF7FFF7FFF7FFF
2022-11-04T18:06:44.899065 051  I --- 02:24XXXX 02:24YYYY  --:------ 4E02 034 0008667FFF7FFF7FFF7FFF7FFF7FFF7FFF0408CA7FFF7FFF7FFF7FFF7FFF7FFF7FFF
2022-11-04T18:06:45.881991 052  I --- 02:24XXXX 02:24YYYY  --:------ 4E02 034 0008667FFF7FFF7FFF7FFF7FFF7FFF7FFF0408CA7FFF7FFF7FFF7FFF7FFF7FFF7FFF
```

tomkooij, 2022-12-01:
I've fixed this already in a local branch. Will push PR.

zxdavb, 2023-05-12:
Yes, thanks Tom

closed 2023-05-12 by zxdavb