ramses-rf/ramses_rf #61: 4E04 Corrupt payload

nl0pvm, 2022-11-04:
Itho devices: Spider, Autotemp, WPU

```
17:28:46.265  I --- AutotempslaveA AutotempMaster --:------ 4E04 003 000004 < Corrupt payload: Payload doesn't match '^00(00FF|01FE)$': 000004
17:42:40.388  I --- AutotempslaveB AutotempMaster --:------ 4E04 003 00000E < Corrupt payload: Payload doesn't match '^00(00FF|01FE)$': 00000E
17:43:31.603  W --- AutotempMaster AutotempslaveB --:------ 4E04 003 00000E < Corrupt payload: Payload doesn't match '^00(00FF|01FE)$': 00000E
17:43:31.693  I --- AutotempslaveB AutotempMaster --:------ 4E04 003 00000E < Corrupt payload: Payload doesn't match '^00(00FF|01FE)$': 00000E
17:43:32.623  I --- AutotempslaveA AutotempMaster --:------ 4E04 003 000004 < Corrupt payload: Payload doesn't match '^00(00FF|01FE)$': 000004
17:54:36.088  I --- AutotempslaveB AutotempMaster --:------ 4E04 003 00000E < Corrupt payload: Payload doesn't match '^00(00FF|01FE)$': 00000E
17:57:26.302  W --- AutotempMaster AutotempslaveB --:------ 4E04 003 00000E < Corrupt payload: Payload doesn't match '^00(00FF|01FE)$': 00000E
17:57:26.364  I --- AutotempslaveB AutotempMaster --:------ 4E04 003 00000E < Corrupt payload: Payload doesn't match '^00(00FF|01FE)$': 00000E
17:58:17.529  W --- AutotempMaster AutotempslaveB --:------ 4E04 003 00000E < Corrupt payload: Payload doesn't match '^00(00FF|01FE)$': 00000E
17:58:17.603  I --- AutotempslaveB AutotempMaster --:------ 4E04 003 00000E < Corrupt payload: Payload doesn't match '^00(00FF|01FE)$': 00000E
17:58:18.472  W --- AutotempMaster AutotempslaveA --:------ 4E04 003 000005 < Corrupt payload: Payload doesn't match '^00(00FF|01FE)$': 000005
```

packetlog:
```
2022-11-04T17:28:46.265250 078  I --- 02:24AAAA 02:24MMMM --:------ 4E04 003 000004
2022-11-04T17:42:40.388694 052  I --- 02:24BBBB 02:24MMMM --:------ 4E04 003 00000E
2022-11-04T17:43:31.603633 079  W --- 02:24MMMM 02:24BBBB --:------ 4E04 003 00000E
2022-11-04T17:43:31.693901 052  I --- 02:24BBBB 02:24MMMM --:------ 4E04 003 00000E
2022-11-04T17:43:32.623798 079  I --- 02:24AAAA 02:24MMMM --:------ 4E04 003 000004
2022-11-04T17:54:36.088201 052  I --- 02:24BBBB 02:24MMMM --:------ 4E04 003 00000E
2022-11-04T17:57:26.302163 076  W --- 02:24MMMM 02:24BBBB --:------ 4E04 003 00000E
2022-11-04T17:57:26.364058 052  I --- 02:24BBBB 02:24MMMM --:------ 4E04 003 00000E
2022-11-04T17:58:17.529092 076  W --- 02:24MMMM 02:24BBBB --:------ 4E04 003 00000E
2022-11-04T17:58:17.603095 052  I --- 02:24BBBB 02:24MMMM --:------ 4E04 003 00000E
2022-11-04T17:58:18.472183 077  W --- 02:24MMMM 02:24AAAA --:------ 4E04 003 000005
```

Received values since 4 months:
```
000000
000001
000002
000004
000005
000006
000007
000008
000009
00000A
00000B
00000C
00000D
00000E
00000F
000018
00001C
000020
000021
000030
000031
000034
000036
000037
000038
000039
00003C
00003F
```

zxdavb, 2023-05-12:
This is a work in progress - so, far we have:
```python
def parser_4e04(payload, msg) -> dict:
    assert payload[2:4] in ("00", "01", "02")  # off/heat/cool?
    assert int(payload[4:], 16) < 0x3C or payload[4:] in (
        "FB",  # error code?
        "FC",  # error code?
        "FD",  # error code?
        "FE",  # error code?
        "FF",  # N/A?
    )

    return {
        "mode": payload[2:4],
        "_unknown_2": payload[4:],
    }
```

Of course, more information would be useful.

zxdavb, 2023-12-03:
OK< this is what I have:
```
def parser_4e04(payload: str, msg: Message) -> dict:
    
    MODE = {
        "00": "off",
        "01": "heat",
        "02": "cool",
    }

    assert payload[2:4] in MODE, _INFORM_DEV_MSG
    assert int(payload[4:], 16) < 0x40 or payload[4:] in (
        "FB",  # error code?
        "FC",  # error code?
        "FD",  # error code?
        "FE",  # error code?
        "FF",  # N/A?
    )

    return {
        "mode": payload[2:4],
        "_unknown_2": payload[4:],
    }
```

nl0pvm, 2024-02-08:
> Please provide packet logs, annotated with any useful information.

I would if i only could but i can't relate the events so something noticeable.

closed 2026-08-17 by silverailscolo