ramses-rf/ramses_rf #62: Intergas RF (Ramses?)

Ierlandfan, 2022-11-12:
Hi,

according to the Intergas Diagnostic Software (IDS) the Intergas HRE / Xtreme HVAC systems are using some form of Ramses_II.
Definately for connecting a Honeywell thermostat. But when I sniff the connection with ramses_rf when coupled with the Intergas Incomfort gateway (Simple device, houses a PIC18F, some flash, network parts and a CC1101 in it) I don't see any packets. 
There's is an undocumented extra exposed port (rj11 or rj12 or so) that does something but haven't figured out yet what it does. 

So deep diving into the IDS (and some obvious magic later) I extracted attached code for the RF Dongle Intergas  is using and the way it connects to the the CV. Can you perhaps look at it to see if it resembles something that can be integrated into ramses_rf?

zxdavb, 2022-11-12:
I have had a look - it talks to a serial port (presumably RF), but the protocol is not RAMSES II, nor is it obviously a variation of RAMSES II (e.g. same codes, but a different packet format).

I have an "InTouch Round" set here - it also came with a modulating Honeywell Round Thermostat (TR87RF2033) That I never connected - it is the version 1 LAN2RF gateway:
 - https://media.s-bol.com/RLDM8NZ9mgBw/980x1200.jpg
 
Note my v1 gateway is a red/rectangular box with only an RJ45 (ethernet) and a micro-USB (power port). 
The version 2 LAN2RF gateway is square, black and has a third port, looks like RJ11, as you say. 

ANyway, that TR87RF was connected to my Honeywell evohome system - it worked perfectly, as per the other two that I have.

zxdavb, 2022-11-14:
I am closing this issue - please re-open it if required.

closed 2022-11-14 by zxdavb