ramses-rf/ramses_rf #66: 22C9 Corrupt payload

nl0pvm, 2022-11-30:
Itho devices: Spider, Autotemp, WPU

```
21:31:14.530 || Spider | AutotempslaveA |  W | ufh_setpoint     |  05  || {'ufh_idx': '05', 'temp_low': 21.0, 'temp_high': 22.0, '_unknown_0': '01'}
21:31:14.633  I --- AutotempslaveA Spider --:------ 22C9 008 0508347FFF010103 < Corrupt payload: Payload doesn't match '^(0[0-9A-F][0-9A-F]{8}0[12]){1,4}(0203)?$': 0508347FFF010103
21:31:14.776 || Spider | AutotempslaveA |  W | ufh_setpoint     |  05  || {'ufh_idx': '05', 'temp_low': 21.0, 'temp_high': 22.0, '_unknown_0': '01'}
21:31:14.847  I --- AutotempslaveA Spider --:------ 22C9 008 0508347FFF010103 < Corrupt payload: Payload doesn't match '^(0[0-9A-F][0-9A-F]{8}0[12]){1,4}(0203)?$': 0508347FFF010103
21:31:14.977 || Spider | AutotempslaveA |  W | ufh_setpoint     |  05  || {'ufh_idx': '05', 'temp_low': 21.0, 'temp_high': 22.0, '_unknown_0': '01'}
21:31:15.042  I --- AutotempslaveA Spider --:------ 22C9 008 0508347FFF010103 < Corrupt payload: Payload doesn't match '^(0[0-9A-F][0-9A-F]{8}0[12]){1,4}(0203)?$': 0508347FFF010103
```
packetlog
```
2022-11-30T21:31:14.530262 000  W --- 21:044797 02:242981 --:------ 22C9 006 050834089801
2022-11-30T21:31:14.633291 054  I --- 02:242981 21:044797 --:------ 22C9 008 0508347FFF010103
2022-11-30T21:31:14.776232 000  W --- 21:044797 02:242981 --:------ 22C9 006 050834089801
2022-11-30T21:31:14.847527 053  I --- 02:242981 21:044797 --:------ 22C9 008 0508347FFF010103
2022-11-30T21:31:14.977377 000  W --- 21:044797 02:242981 --:------ 22C9 006 050834089801
2022-11-30T21:31:15.042668 054  I --- 02:242981 21:044797 --:------ 22C9 008 0508347FFF010103
```

tomkooij, 2022-12-01:
I've fixed these already on a local branch. Need to submit a PR...

```
(base) cat nl0pvm.log
2022-11-30T21:31:14.530262 000  W --- 21:044797 02:242981 --:------ 22C9 006 050834089801
2022-11-30T21:31:14.633291 054  I --- 02:242981 21:044797 --:------ 22C9 008 0508347FFF010103
2022-11-30T21:31:14.776232 000  W --- 21:044797 02:242981 --:------ 22C9 006 050834089801
2022-11-30T21:31:14.847527 053  I --- 02:242981 21:044797 --:------ 22C9 008 0508347FFF010103
2022-11-30T21:31:14.977377 000  W --- 21:044797 02:242981 --:------ 22C9 006 050834089801
2022-11-30T21:31:15.042668 054  I --- 02:242981 21:044797 --:------ 22C9 008 0508347FFF010103
(base) cat nl0pvm.log | python client.py -lf parse

client.py: Starting ramses_rf...
08:57:51.920 Best practice is exactly one gateway (HGI) in the known_list: []
08:57:51.921 It is strongly recommended to provide a known_list, and use it as a whitelist (device_id filter), configure: enforce_known_list = True
client.py: Starting engine...
01:00:00.000 Datetimes maintained as most recent packet log timestamp
01:00:00.000 Not using any device filter: using a known_list (as a whitelist) is strongly recommended)
2022-11-30T21:31:14.530262 || HVC:044797 |  02:242981 |  W | ufh_setpoint     |  05  || {'ufh_idx': '05', 'temp_low': 21.0, 'temp_high': 22.0, '_unknown_0': '01'}
2022-11-30T21:31:14.633291 || UFC:242981 | HVC:044797 |  I | ufh_setpoint     |  05  || {'ufh_idx': '05', 'temp_low': 21.0, 'temp_high': None, '_unknown_0': '01'}
2022-11-30T21:31:14.776232 || HVC:044797 | UFC:242981 |  W | ufh_setpoint     |  05  || {'ufh_idx': '05', 'temp_low': 21.0, 'temp_high': 22.0, '_unknown_0': '01'}
2022-11-30T21:31:14.847527 || UFC:242981 | HVC:044797 |  I | ufh_setpoint     |  05  || {'ufh_idx': '05', 'temp_low': 21.0, 'temp_high': None, '_unknown_0': '01'}
2022-11-30T21:31:14.977377 || HVC:044797 | UFC:242981 |  W | ufh_setpoint     |  05  || {'ufh_idx': '05', 'temp_low': 21.0, 'temp_high': 22.0, '_unknown_0': '01'}
2022-11-30T21:31:15.042668 || UFC:242981 | HVC:044797 |  I | ufh_setpoint     |  05  || {'ufh_idx': '05', 'temp_low': 21.0, 'temp_high': None, '_unknown_0': '01'}
client.py: Engine stopped: ended without error (e.g. EOF).

```

zxdavb, 2023-05-12:
Has been fixed in an upcoming release - thanks tom

closed 2023-05-12 by zxdavb