ramses-rf/ramses_rf #68: Assertionerror 31DA with faked remote

janvkem, 2022-12-07:
2022-12-06 21:03:19.834 WARNING (MainThread) [ramses_rf.dispatcher] RP --- 32:155617 37:155617 --:------ 31DA 030 00EF007FFF363103AC05DC06CE02C9F80000182A2A0000EFEF091D091D00 < AssertionError(msg from 32:155617 (FAN) inappropriately routed to 37:155617 (REM))

zxdavb, 2022-12-11:
The issue here is an anachronism (battery-operated remotes sending `RQ`s) - the solution will require some refactoring.

janvkem, 2022-12-11:
I also seem to get it with 2411 commands:
2022-12-11 21:05:03.319 WARNING (MainThread) [ramses_rf.dispatcher]  I --- 32:153258 37:153258 --:------ 2411 023 00003F390F0000003800000000000000A000000001A632 < AssertionError(msg from 32:153258 (FAN) inappropriately routed to 37:153258 (REM))

This is what I have sent and got back:
2022-12-11T21:05:03.279121 000  W --- 37:153258 32:153258 --:------ 2411 023 00003F000F0000003800000000000000A0000000010032
2022-12-11T21:05:03.314753 080  I --- 32:153258 37:153258 --:------ 2411 023 00003F390F0000003800000000000000A000000001A632

maleadt, 2023-11-14:
Any change here? I was hoping to periodically send a `RQ 31DA` command to get some better resolution on sensor values, instead of the 5 minute default interval.

zxdavb, 2023-11-15:
I have made the refactoring at the client layer (**ramses_rf**). I have just pushed the corresponding **ramses_cc** and am waiting for it to settle.

Can you pm me the `ramses_cc:` section of your **configuration.yaml**, and a 24h packet log?

zxdavb, 2023-11-15:
Hmmm... This is currently working for me.

```
    32:205847:
      class: REM
      faked: true
      commands:
        refresh:     'RQ --- 32:205847 30:071287 --:------ 31DA 001 00'
```

zxdavb, 2023-11-15:
I miss-read - I'll need to set up a emulated system...

zxdavb, 2023-11-15:
Actually, try this: 
```
refresh:     'RQ --- 18:000730 32:134044 --:------ 31DA 001 00'
```

Obviously, use

maleadt, 2023-11-15:
That's basically what I was using:

```yaml
ramses_cc:
  serial_port: /dev/ttyUSB0

  ramses_rf:
    enforce_known_list: true

  orphans_hvac: [32:134044, 29:150155]

  known_list:
    32:134044 : { class: FAN, _note: Orcon HRC-350 }
    29:150155:
      class: REM
      faked: True
      _note: Orcon 15RF
      commands:
        request31DA:  'RQ --- 29:150155 32:134044 --:------ 31DA 001 00'
        ...
```

With different IDs of course.

zxdavb, 2023-11-15:
I think you really need to try with the 18:000730 address. It's a sentinel value for the gateway

You should also be using 0.30.2: if it doesn't work out for you it's very straightforward to switch back to your current version.

The only issue is that the entity ID of four FAN attributes have changed (unique ID has not changed).

maleadt, 2023-11-15:
I see. How does that work with `enforce_known_list: true`?

Disabling that for now, with `'RQ --- 18:000730 32:134044 --:------ 31DA 001 00'` I get the following packet log:

```
2023-11-15T21:55:07.663238 000 RQ --- 18:066589 32:134044 --:------ 31DA 001 00
2023-11-15T21:55:07.707242 077 RP --- 32:134044 18:066589 --:------ 31DA 029 00EF007FFF3F30041A06C206DD055BF800000248480000EFEF0F810F9D
```

and the following in HA's log:

```
homeassistant_1  | 2023-11-15 21:55:07.707 INFO (MainThread) [ramses_rf.dispatcher] || FAN:134044 | HGI:066589 | RP | hvac_state       |      || {'exhaust_fan_speed': 0.36, 'fan_info': 'speed 2', 'remaining_time': 0.0, 'air_quality': None, 'air_quality_base': 0, 'co2_level': None, 'indoor_humidity': 0.63, 'outdoor_humidity': 0.48, 'exhaust_temperature': 10.5, 'supply_temperature': 17.3, 'indoor_temperature': 17.57, 'outdoor_temperature': 13.71, 'speed_cap': 63488, 'bypass_position': 0.0, 'supply_fan_speed': 0.36, 'post_heat': None, 'pre_heat': None, 'supply_flow': 39.69, 'exhaust_flow': 39.97}
```

So the request now isn't logged anymore, only the reply. However, I still don't get updated sensor values. For example, monitoring `sensor.32_134044_indoor_humidity` in lovelace only shows an update every 5 minutes, regardless of the 31DA requests I sent. Maybe I'm missing something; I would assume that additional 31DA messages should lead to additional sensor updates?

zxdavb, 2023-11-16:
> I see. How does that work with enforce_known_list: true?

It is a sentinel value - it gets changed to `18:066589` by your dongle (and that address should be in your `known_list`).

> So the request now isn't logged anymore, only the reply. 

It is logged in the **packet.log**, it is not logged otherwise, to reduce logspam.

> Maybe I'm missing something; I would assume that additional 31DA messages should lead to additional sensor updates?

What is your `sync_interval`? Set it to:
```
ramses_cc:
  scan_interval: 60
```

You haven't aid what version you're using - I am no longer developing on anything older than **0.30.x** - any bugfixes will go there.

As long as you don't get this:
```
msg from 32:134044 (FAN) inappropriately routed to 29:150155 (REM)
```
... you should be fine.

zxdavb, 2023-11-16:
I am closing this for now - please re-open it if you need to, but only if you're running **0.30.x**.

If you could PM me a packet log to test/dev against, that will be helpful to the HVAC users.

closed 2023-11-16 by zxdavb