ramses-rf/ramses_rf #70: Feature request: Error codes support

remyderuysscher, 2023-01-20:
The real RF15 display (Orcon RF15 display) has readout for error codes. 

Can you implement these? I have the modbus addressed and error messages for the error codes if this helps.

zxdavb, 2023-05-12:
I would need, at the least, a log file taken when such an error appears.

remyderuysscher, 2023-05-12:
These error codes are not RF15 specific but work for Honeywell (OEM) such as Orcon.  I have the error codes from the RF modbus gateway but I don't know how they relate to error codes in ramses_rf?  Would you like to have these error codes?

zxdavb, 2023-05-12:
Please send me whatever you have.

When you say (my quotes):
> These error codes ... work for 'Honeywell (OEM)' such as Orcon

I think you mean HVAC (OEM)?

In my mind, Honeywell is restricted to the Heat domain (products include: evohome, chronotherm, sundial, etc.). The HVAC domain includes products from Itho, Orcon, Nuaire, Ventiline, etc.

The Heat domain uses a `0418` packet, which is well-understood.

I would need to see an equivalent packet for the HVAC domain, in a packet log.
 - sometimes errors are inducible (e.g. taking a battery out of a remote/removing a filter, and then waiting): are you able to think of such a scenario?

remyderuysscher, 2023-05-13:
No Honeywell OEM, I have documentation from Honeywell regarding the modbus gateway. I'll send you the documentation.

wimpie70, 2025-08-05:
Zijn er logs hiervan ?
Ik heb wel deze site gevonden : https://www.wtwstoringscodes.nl/merk/orcon/

silverailscolo, 2025-09-01:
@remyderuysscher, other Orcon users, when you see an error from your HVAC, please attach the packet log of that day (only) here as a zip.

If we collect no logs by november 2025, I intend to close this issue.

closed 2026-10-23 by silverailscolo