ramses-rf/ramses_rf #72: Question

pdbayes, 2023-02-18:
Hi. Amazing work you have done. 
Do you know if the _flags_n have any useful purpose. I only ever see a value of FF?
Thanks

zxdavb, 2023-02-19:
I am sorry - you still haven't provided enough information for me to answer your question.

Please provide a context to your question.

pdbayes, 2023-02-19:
evohome/evogateway/_zone_independent/relays/bdr_hotwater_valve/actuator_state
{"modulation_level": 0.0, "_flags_2": "FF"}

It is the "_flags_2": "FF" i do not understand, I would expect the state of the actuator to be on or off but I only ever see FF

zxdavb, 2023-02-19:
**evogateway** uses an older version of **ramses_rf**.

You have not provided enough information for me to help you. For example, there are two type of actuators - which type do you have?

The modulation level tells you if it is on (1.0, or >0.0, depending) or off (0.0).

Flags are usually state flags - they appear to only be used by OTB, and not BDR.

closed 2023-02-19 by zxdavb

pdbayes, 2023-02-19:
Sorry. We only have 3 BDRs. 2 for heating zones and one for DHW. We use the
touch controller as a thermostat and we have a round Honeywell one for the
other zone and one wireless hot water thermometer