ramses-rf/ramses_rf #74: 3EF1 007 0000AA00A63FF9 < AssertionError(byte 1: 00AA)

epicurei, 2023-03-06:
11:13:48.056 RP --- 13:107116 18:201230 --:------ 3EF1 007 0000AA00A63FF9 < AssertionError(byte 1: 00AA)
Traceback (most recent call last):
  File "/home/supervisor/Scrivania/ramses_rf-0.22.40 mod/ramses_rf/protocol/message.py", line 331, in _validate
    result = PAYLOAD_PARSERS.get(self.code, parser_unknown)(
  File "/home/supervisor/Scrivania/ramses_rf-0.22.40 mod/ramses_rf/protocol/parsers.py", line 141, in wrapper
    result = fnc(payload, msg, **kwargs)
  File "/home/supervisor/Scrivania/ramses_rf-0.22.40 mod/ramses_rf/protocol/parsers.py", line 2582, in parser_3ef1
    assert payload[2:6] == "7FFF", f"byte 1: {payload[2:6]}"
AssertionError: byte 1: 00AA

zxdavb, 2023-03-06:
Thanks for you submission.

Unfortunately, the traceback by itself is not useful.

If you're willing, please submit the corresponding packet log, and a bit of an explanation what your system may have been doing at the time.

Is the relay a **BDR91T** - that would be exciting.

zxdavb, 2023-03-06:
```
RP --- 13:107116 18:201230 --:------ 3EF1 007 0000AA00A63FF9
```

The issue is that the last byte is **`F9`** and not **`FF`**.

I wonder if it is simply a corrupt packet? Or maybe the BDR is throwing a error code?

Is is just a one-off?

zxdavb, 2023-12-03:
I need more packet logs to do anything with this.

closed 2025-07-30 by silverailscolo