ramses-rf/ramses_rf #77: 3EF0 coding error

JeroenTogt, 2023-03-21:
I see multiple times the following error:

Pretty clear whats happening as it expects 00 - 64 at that location, but there are a lot of these errors. Could it be that  64 is not the max value?

Attached logfile as well, hope this gives some info..
[2023.txt](https://github.com/zxdavb/ramses_rf/files/11032796/2023.txt)

Thanks for looking into this and the great work you do on this

2023-03-21 18:19:43.768 ERROR (MainThread) [ramses_rf.protocol.message]  I --- 10:057706 --:------ 10:057706 3EF0 009 008F100A00FF032600 < Coding error: ValueError(Invalid result: 1.43 (0x8F) is > 1)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/ramses_rf/protocol/message.py", line 331, in _validate
    result = PAYLOAD_PARSERS.get(self.code, parser_unknown)(
  File "/usr/local/lib/python3.10/site-packages/ramses_rf/protocol/parsers.py", line 141, in wrapper
    result = fnc(payload, msg, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/ramses_rf/protocol/parsers.py", line 2478, in parser_3ef0
    mod_level = percent_from_hex(payload[2:4], high_res=False)  # 00-64 (or FF)
  File "/usr/local/lib/python3.10/site-packages/ramses_rf/protocol/helpers.py", line 27, in wrapper
    return fnc(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/ramses_rf/protocol/helpers.py", line 252, in percent_from_hex
    raise ValueError(f"Invalid result: {result} (0x{value}) is > 1")
ValueError: Invalid result: 1.43 (0x8F) is > 1

I

zxdavb, 2023-03-22:
Regarding the log you did provide me, these messages are very unusual:
```
No response for task(000C|RP|01:007149|0000): throttling to 1/6h
```

So, could you kindly provide the make/model of your controller?

JeroenTogt, 2023-03-22:
Hey David,

Here my setup
Evotouch controller ATC928G3000 ( firmware 02.00.19.33)
Openterm bridge R8810A1018
  OTGW ( openterm gateway version 6.2) in gateway mode. http://otgw.tclcode.com/ hardware from nodoshop ( this gateway is used to look at the openterm communication between the bridge and the boiler, it is mounted in between the bridge and the boiler)
Boiler Remeha Calenta Ace 40c
In zone 0 6x HR92 and the evotouch controller as sensor
in zone 1 2x Hr92 and T87RF as sensor

Hope this is enough info!
If you need more pls let me know

[home-assistant.log](https://github.com/zxdavb/ramses_rf/files/11043263/home-assistant.log)
[packet.log](https://github.com/zxdavb/ramses_rf/files/11043268/packet.log)
[otgw log.txt](https://github.com/zxdavb/ramses_rf/files/11043289/otgw.log.txt)

zxdavb, 2023-03-22:
Your setup is largely the same as mine:
 - ATC928G3000
 - R8820A (different)
 - EMS Gateway from BBQkees (instead of OTGW)
 - Worcester Greenstar Si (different) with Nefit Opentherm Converter EMS-OT

Give me a few minutes to review your logs and have a think about next steps.

zxdavb, 2023-03-23:
I should start by saying your log files are unhelpfully short - 24h log files (or longer) would be more useful.

In your **OTGW** log, we have:
```text
18:52:41.077956  T900E6400  Write-Data  Maximum relative modulation level: 100.00
18:52:41.227004  B500E6400  Write-Ack   Maximum relative modulation level: 100.00
   ...
18:52:45.074154  T00110000  Read-Data   Relative modulation level: 0.00
18:52:45.176620  BC0110000  Read-Ack    Relative modulation level: 0.00
   ...
18:52:51.074133  T00110000  Read-Data   Relative modulation level: 0.00
18:52:51.175826  B40117C00  Read-Ack    Relative modulation level: 124.00
   ...
18:52:57.063794  T00110000  Read-Data   Relative modulation level: 0.00
18:52:57.176817  BC0111B00  Read-Ack    Relative modulation level: 27.00
```

The above is the _only_ example in that log file where the RelModLevel was > 100%. Note it is above the MaxRelModLevel, 100%.

The equivalent messages in the **ramses_rf** packet log look like this:
```text
2023-03-22T18:41:48.689945 ... RP --- 10:057706 01:007149 --:------ 3220 005 00C0110000
2023-03-22T18:43:47.676770 039 RP --- 10:057706 01:007149 --:------ 3220 005 00C0110000
2023-03-22T18:48:49.517022 040 RP --- 10:057706 01:007149 --:------ 3220 005 00C0110000
2023-03-22T18:53:51.364044 039 RP --- 10:057706 01:007149 --:------ 3220 005 00C0110000
```

Note the above all have a RelModLevel of 0% (0x00). I am not sure if it is always 0x00, or just was so for these 4 instances (again, the logs you sent me need to be > 24h).

zxdavb, 2023-03-23:
The challenge we have is this...

For all other OTB/boiler combinations I've seen, the OT (3220) `RelModLevel` is reported as a percentage from 0x00 to 0x64 (0-100 decimal).

I do know some boilers can modulate above 100%, say 125%, but I've never seen such values.

Most importantly, I have access to the OT specification, and it only permits values from 0 to 100. 

So, on that basis, this is returning an invalid value (the `7C` in the payload):
```text
18:52:51.175826  B40117C00  Read-Ack    Relative modulation level: 124.00
```

zxdavb, 2023-03-23:
So I wonder if - in your specific case - the value is from 0-200 and you need to divide by 200 to get the percentage (i.e. 0.0 to 1.0) rather than 100?

As a first step, maybe you could remove the OTGW and connect the OTB direct to the boiler and see what happens. Is the bad value from the boiler itself, or is it caused by the OTGW?

JeroenTogt, 2023-03-23:
Hi David,
All clear, root cause is that the rel modulation level is outside the 0-100 range as per the OT specification.
When I look at the log, this indeed happens ( this one is this morning and included in the ramses logs, unfortunately not completely in the OTGW log) ill post them later today so they can gather some more...:-)
[image]
This also shows that most of the time the boiler runs at minimum capacity. ( it is just heating the attic).
Also the max capacity of the boiler is limited as well( with parameters in the boiler, you see that as well in this log. ( but it sometimes goes to 101 %...). 

See here an old graph were you see relative mod level even go negative...and well above 100
[image]

Here a sort of a typical run
[image]

My guess is that the boiler is producing the bad numbers.

Ill switch the OTGW to monitor mode, it is not supposed to do anything with the data and see if this still happens
Then ill upgrade the FW ( current one is of oktober 2022) there is a new version, but don't see anything related to this in the changelog...
Then ill hook it up direct...

Ill post the question of relative mod level on the OTGW forum as well, to see if someone else notices this as well...

Ill post the longer logfiles at the end of the day.

Anything on the throttling to 1/6h thing you picked up as unusual?

zxdavb, 2023-03-23:
> Anything on the throttling to 1/6h thing you picked up as unusual?

There are no RPs from the controller to the HGI-compatible gateway in the packet logs you sent me.

Maybe the RF signal strength between the HGI-compatible gateway and the controller is off? Can you tune it (some of the **evofw3** devices can be tuned), or put it in a better location?

I am closing this issue - I don't believe I can provide a fix, and I don't think any possible workaround would be useful.

closed 2023-03-23 by zxdavb

JeroenTogt, 2023-03-23:
Hi David,

Here the longer logs.
Just see if usefull for future reference.
If I find some on the relative modulation level not in between 0 -100 dec then Ill let you know.

[otlog-20230323.txt](https://github.com/zxdavb/ramses_rf/files/11051683/otlog-20230323.txt)
[packet.log](https://github.com/zxdavb/ramses_rf/files/11051701/packet.log)
[home-assistant.log](https://github.com/zxdavb/ramses_rf/files/11051704/home-assistant.log)

JeroenTogt, 2023-03-23:
[packet.log.2023-03-22.log](https://github.com/zxdavb/ramses_rf/files/11051898/packet.log.2023-03-22.log)
first bit of the packet log till midnight

commit 0cbebf1a (2023-11-13): fix #77 hvac_id shoudl be accepted
commit 9d0d787c (2023-11-13): fix #77 idx for 31D9, 31DA