ramses-rf/ramses_rf #81: Problems with Ramses_rf after migrating to Elga Ace

JBSDM, 2023-06-12:
Hello,

I have a HGI80 installed in combination with Home Assistant. This worked like a charm when my Evohome was connected to my Remeha Calenta. But since a month we have a hybrid heat pump and the Evohome is migrated towards this device. After that i do not get good data from the HGI80. Do you have any idea how that is possible?

If you need any information from me i would love to give it to fix my issue... i don't like the Cloud connection to Evohome.

Best Regards,

JBSDM

zxdavb, 2023-09-21:
Hello, as a first step, you would have to supply me with packet logs, please. minimum 24h.

JBSDM, 2023-09-27:
I will make a packet log. What is de best way to share it with you?

JBSDM, 2023-09-27:
I added all again and the problem is fixed now:)

closed 2023-09-27 by JBSDM