ramses-rf/ramses_rf #82: I --- 02:008366 --:------ 02:008366 22D0 004 0010000A < AssertionError(Support the development of ramses_rf by reporting this packet)

mversluijs, 2023-06-25:
`2023-06-25 16:20:20.387 ERROR (MainThread) [ramses_rf.protocol.message] I --- 02:008366 --:------ 02:008366 22D0 004 0010000A < AssertionError(Support the development of ramses_rf by reporting this packet)`

seems to return every 15 minutes

configuration:
2x hce80
evotouch
4x honeywell round
2x relais to heatpump 

no issue just helping development :-)
[packet.log](https://github.com/zxdavb/ramses_rf/files/11860003/packet.log)

zxdavb, 2023-10-02:
Thanks - I will look at this packet log, and make changes to the parsers.

closed 2023-10-02 by zxdavb

zxdavb, 2023-10-02:
BTW, if I parse your packet log:
 - `grep ' RP.* 01:.* 000C ' packet.log | python client.py parse`

I get:
```text
16:26:18.417 ||  01:198098 |  18:262143 | RP | zone_devices     | 000F || {'zone_type': '0F', 'domain_id': 'FC', 'device_role': 'appliance_control', 'devices': []}

16:26:21.583 ||  01:198098 |  18:262143 | RP | zone_devices     | 000E || {'zone_type': '0E', 'domain_id': 'FA', 'device_role': 'hotwater_valve', 'devices': []}
16:26:21.783 ||  01:198098 |  18:262143 | RP | zone_devices     | 010E || {'zone_type': '0E', 'domain_id': 'F9', 'device_role': 'heating_valve', 'devices': []}
16:26:22.185 ||  01:198098 |  18:262143 | RP | zone_devices     | 000D || {'zone_type': '0D', 'domain_id': 'FA', 'device_role': 'dhw_sensor', 'devices': []}

16:27:26.251 ||  01:198098 |  18:262143 | RP | zone_devices     | 0004 || {'zone_type': '04', 'zone_idx': '00', 'device_role': 'zone_sensor', 'devices': []}
16:27:24.393 ||  01:198098 |  18:262143 | RP | zone_devices     | 0104 || {'zone_type': '04', 'zone_idx': '01', 'device_role': 'zone_sensor', 'devices': ['34:229417']}
16:26:23.592 ||  01:198098 |  18:262143 | RP | zone_devices     | 0204 || {'zone_type': '04', 'zone_idx': '02', 'device_role': 'zone_sensor', 'devices': ['34:229409']}
16:26:22.789 ||  01:198098 |  18:262143 | RP | zone_devices     | 0304 || {'zone_type': '04', 'zone_idx': '03', 'device_role': 'zone_sensor', 'devices': ['34:229557']}
16:26:24.478 ||  01:198098 |  18:262143 | RP | zone_devices     | 0404 || {'zone_type': '04', 'zone_idx': '04', 'device_role': 'zone_sensor', 'devices': ['34:229561']}
```

zxdavb, 2023-10-02:
Please implement a known_list - see the wiki for why/how.

The above is consistent with the relays being attached to the master UFH controller (is slightly unusual, but appears to be the case here), but if they are not: then your system is misconfigured?

Also: do you have an OTB for controlling your heatsource?

Don't worry about the missing sensor for zone 0 - it will be the Evo controller, and that's what happens in that case.

Finally, do you have any HVAC kit? Do you have an RFG100?

zxdavb, 2023-10-02:
FYI:

If you have `"enforce_known_list": true`, then you won't need the `block_list`.

Note teh absence of relays in you packet log (`device_id`s starting `13:` for a BDR91A, BDR91T, or even `10:` for an OTB)

```json
{
    "config": {
        "enforce_known_list": false
    },

    "main_tcs": "01:198098",
    "01:198098": {
        "system": {"appliance_control": null},
        "zones": {
            "00": {"sensor": "01:198098"}
        }
    },

    "known_list": {
        "01:111111": {"class": "CTL"},
        "02:008264": {"class": "UFC"},
        "02:008366": {"class": "UFC"},
        "18:262143": {"class": "HGI"},
        "34:229409": {"class": "RND", "_notes": "T87RF2025"},
        "34:229417": {"class": "RND"},
        "34:229557": {"class": "RND"},
        "34:229561": {"class": "RND"}
    },
    "block_list": {
        "21:060649": {"class": "RFS", "_notes": "to 30:"},
        "30:081866": {"class": "FAN", "_notes": "to 21:"},
        "18:109099": {"class": "FAN", "_notes": "to 37:"},
        "37:143568": {"class": "CO2", "_notes": "to 18:"}
    }
}
```

zxdavb, 2023-10-02:
If I use the white list (`known_list`), _with my dev/test version_ of **ramses_rf**, these are the only errors I get:
```
19:09:42.448 RQ --- 18:262143 01:198098 --:------ 2349 002 2000 < Payload doesn't match '^0[0-9A-F](00|[0-9A-F]{12})?$': 2000
19:09:42.449 RP --- 01:198098 18:262143 --:------ 2349 007 207FFF02FFFFFF < Payload doesn't match '^0[0-9A-F]{5}0[0-4][0-9A-F]{6}([0-9A-F]{12})?$': 207FFF02FFFFFF
19:09:42.459 RQ --- 18:262143 01:198098 --:------ 2349 002 0010 < Payload doesn't match '^0[0-9A-F](00|[0-9A-F]{12})?$': 0010
19:09:42.638 RQ --- 18:262143 01:198098 --:------ 2349 002 0040 < Payload doesn't match '^0[0-9A-F](00|[0-9A-F]{12})?$': 0040
19:09:43.034 RQ --- 18:262143 01:198098 --:------ 2E40 001 FF < Unknown code: 2E40
```

They are all obviously corrupt packets.

In addition, the packet you've reported as not parsing, now parses as so:
```
2023-06-25T16:20:20.387000 ||  02:008366 |            |  I | message_22d0     |  00  || {'zone_idx': '00', 'idx': '00', '_flags': [0, 0, 0, 1, 0, 0, 0, 0], 'cool_mode': False, 'heat_mode': False, 'is_active': True, '_unknown': '000A'}
```

zxdavb, 2023-10-02:
... and we're done!

(other than you implementing a `known_list` and working out what happened to your relays.