ramses-rf/ramses_rf #83: ramses_rf won't run

georgeavd, 2023-07-05:
Hi,

I'm trying to get ramses_rf running but still receiving the following error-messages when running
`python client.py monitor /dev/ttyACM1 -o packet.log`:

Output:
`.......k:~/ramses_rf$ python client.py mopnitor /dev/ttyACM1 -o packet.log
Traceback (most recent call last):
  File "/home/george/ramses_rf/client.py", line 49, in <module>
    from ramses_rf import Gateway, GracefulExit, Message
  File "/home/george/ramses_rf/ramses_rf/__init__.py", line 16, in <module>
    from .const import __dev_mode__
  File "/home/george/ramses_rf/ramses_rf/const.py", line 9, in <module>
    from .protocol.const import (  # noqa: F401
  File "/home/george/ramses_rf/ramses_rf/protocol/__init__.py", line 9, in <module>
    from .address import Address, is_valid_dev_id
  File "/home/george/ramses_rf/ramses_rf/protocol/address.py", line 14, in <module>
    from .helpers import typechecked
  File "/home/george/ramses_rf/ramses_rf/protocol/helpers.py", line 130, in <module>
    def timestamp() -> float:
  File "/home/george/.local/lib/python3.9/site-packages/typeguard/_decorators.py", line 213, in typechecked
    retval = instrument(target)
  File "/home/george/.local/lib/python3.9/site-packages/typeguard/_decorators.py", line 76, in instrument
    module_code = compile(module_ast, f.__code__.co_filename, "exec", dont_inherit=True)
ValueError: empty body on ExceptHandler`

I'vo followed the few steps listed on: [https://github.com/zxdavb/ramses_rf](url)
I'm running `Linux iotstack 5.10.0-23-amd64 #1 SMP Debian 5.10.179-1 (2023-05-12) x86_64 GNU/Linux` and `Python 3.9.2`.
With the SSM-D2 dongle running evofw3.

Am I missing something that keeps my setup from running?

zxdavb, 2023-07-05:
Start by upgrading to python 3.10.

zxdavb, 2023-07-06:
Then ensure you're using requirements.txt.

georgeavd, 2023-07-06:
I've tested in on a clean Debian 11 with Python updated to 3.11
Besides running the requirements.txt a few additional packages were needed. E.g. click and others.
After the missing packages were installed there was a complaint about the USB-port.
This simply solved by putting the correct device name: `/dev/ttyACM0`

So the higher Python version did the trick.

closed 2023-09-21 by zxdavb