ramses-rf/ramses_rf #85: Invalid code for 32:161205 (FAN) to Rx/Tx: 22E0

pocketware, 2023-09-26:
In the home-assistant.log:
`2023-09-26 20:36:31.637 INFO (MainThread) [ramses_rf.dispatcher] RQ --- 18:070670 32:161205 --:------ 22F4 001 00 < Invalid code for 32:161205 (FAN) to Rx/Tx: 22F4`

I have a lot of issues with the SSM-D2 from Peter Price. Maybe this is the cause of it...

Retrieving data with a a almost empty configuration:
`ramses_cc:
  serial_port:
    port_name: /dev/serial/by-id/usb-SparkFun_evofw3_atmega32u4-if00
  scan_interval: 30
  packet_log:
    file_name: packet.log
    rotate_backups: 7
  restore_cache:
    restore_state: false
    restore_schema: false
  advanced_features:
    send_packet: true`

is working without any issues. Getting all kind of logs in de packet.log file.

After adding configuration to retrieve (and send) command my system hangs after a while. No log captured anymore and also my other 868MHz devices aren't working anymore. Seems that the SSM-D2 keeps sending packages.

Current config is:
`ramses_cc:
  restore_cache: false
  ramses_rf:
    enforce_known_list: true
  serial_port: /dev/ttyACM0 #/dev/serial/by-id/usb-SparkFun_evofw3_atmega32u4-if00
  packet_log:
    file_name: packet.log
    rotate_backups: 7  
  advanced_features:
    send_packet: false  
#  scan_interval: 180  
  orphans_hvac: [18:070670,32:161205,29:179059]
  known_list:
    32:161205: {class: FAN}
    18:070670: {class: HGI}
    29:179059: {class: REM}`
    
 With this configuration I'm getting data (after resetting HomeAssistant and the SSM serveral times).
 
 Adding configuration like commands:, faked: etc. or enabling send_package is causing issues.
 
 Also enabled extra logging with:
 `logger:
  logs:
    custom_components.ramses_cc: debug  # show ramses_cc/ramses_rf state
    ramses_rf.dispatcher: debug         # show packet payloads
    homeassistant.components.serial: debug`
 
 
 Log after last reboot (only Ramses lines):
 `2023-09-26 20:31:45.498 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration ramses_cc which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-09-26 20:32:16.476 INFO (MainThread) [custom_components.ramses_cc] ramses_cc v0.22.40, is using ramses_rf v0.22.40
2023-09-26 20:32:16.476 DEBUG (MainThread) [custom_components.ramses_cc] 

Config = {'restore_cache': {'restore_schema': False, 'restore_state': False}, 'ramses_rf': {'enforce_known_list': True, 'max_zones': 12, 'use_aliases': False, 'disable_sending': False, 'disable_discovery': False, 'use_native_ot': 'prefer', 'enable_eavesdrop': False, 'reduce_processing': 0}, 'serial_port': {'port_name': '/dev/ttyACM0', 'timeout': 0, 'baudrate': 115200, 'xonxoff': True, 'rtscts': False, 'dsrdtr': False}, 'packet_log': {'file_name': 'packet.log', 'rotate_backups': 7}, 'advanced_features': {'send_packet': False, 'message_events': None}, 'orphans_hvac': ['18:070670', '32:161205', '29:179059'], 'known_list': {'32:161205': {'class': 'FAN', 'faked': None, 'alias': None, '_domain': 'hvac'}, '18:070670': {'class': 'HGI', 'faked': None, 'alias': None, '_domain': 'heat'}, '29:179059': {'class': 'REM', 'faked': None, 'alias': None, '_domain': 'hvac'}}, 'scan_interval': datetime.timedelta(seconds=60), 'block_list': {}}

2023-09-26 20:32:16.485 DEBUG (MainThread) [custom_components.ramses_cc.schemas] A cached schema was not enabled (not recommended)
2023-09-26 20:32:16.485 DEBUG (MainThread) [custom_components.ramses_cc.schemas] Loaded a config schema: {'orphans_hvac': ['18:070670', '32:161205', '29:179059']}
2023-09-26 20:32:16.485 INFO (MainThread) [custom_components.ramses_cc.schemas] Using the config schema (cached schema is not enabled / is invalid), consider using 'restore_cache: restore_schema: true'
2023-09-26 20:32:16.492 INFO (MainThread) [custom_components.ramses_cc.coordinator] Success initialising with the config schema: {'orphans_hvac': ['18:070670', '32:161205', '29:179059']}
2023-09-26 20:32:16.492 INFO (MainThread) [custom_components.ramses_cc.coordinator] Not restoring any cached state (disabled), consider using 'restore_cache: restore_state: true
2023-09-26 20:32:16.492 DEBUG (MainThread) [custom_components.ramses_cc.coordinator] Starting the RF monitor...
2023-09-26 20:32:16.517 DEBUG (MainThread) [custom_components.ramses_cc] 

Store = {'client_state': {'schema': {'main_tcs': None, 'orphans_heat': [], 'orphans_hvac': []}, 'packets': {}}, 'remotes': {'29:179059': {}}}

[... 17 lines cut ...]
2023-09-26 20:32:17.039 INFO (MainThread) [custom_components.ramses_cc.sensor] Found a Sensor for 32:161205: remaining_time
2023-09-26 20:32:17.039 INFO (MainThread) [custom_components.ramses_cc.sensor] Found a Sensor for 32:161205: speed_cap
2023-09-26 20:32:17.040 INFO (MainThread) [custom_components.ramses_cc.sensor] Found a Sensor for 32:161205: supply_fan_speed
2023-09-26 20:32:17.040 INFO (MainThread) [custom_components.ramses_cc.sensor] Found a Sensor for 32:161205: supply_flow
2023-09-26 20:32:17.040 INFO (MainThread) [custom_components.ramses_cc.sensor] Found a Sensor for 32:161205: supply_temperature
2023-09-26 20:32:19.351 INFO (MainThread) [custom_components.ramses_cc.remote] Found a Remote: 29:179059 (REM)
2023-09-26 20:32:20.124 INFO (MainThread) [custom_components.ramses_cc.climate_hvac] Found a HVAC system: 32:161205 (FAN)
2023-09-26 20:32:21.519 INFO (MainThread) [custom_components.ramses_cc.coordinator] Saving the client state cache (packets, schema)...
2023-09-26 20:32:31.613 INFO (MainThread) [ramses_rf.dispatcher] || HGI:070670 | FAN:161205 | RQ | unknown_22e0     |      || {}
2023-09-26 20:32:31.615 INFO (MainThread) [ramses_rf.dispatcher] RQ --- 18:070670 32:161205 --:------ 22E0 001 00 < Invalid code for 32:161205 (FAN) to Rx/Tx: 22E0
2023-09-26 20:32:34.129 INFO (MainThread) [custom_components.ramses_cc.binary_sensor] Found a Binary Sensor for 18:070670: status
2023-09-26 20:32:39.112 INFO (MainThread) [custom_components.ramses_cc.coordinator] Saving the client state cache (packets, schema)...
2023-09-26 20:33:32.283 INFO (MainThread) [ramses_rf.dispatcher] || HGI:070670 | FAN:161205 | RQ | unknown_22e5     |      || {}
2023-09-26 20:33:32.285 INFO (MainThread) [ramses_rf.dispatcher] RQ --- 18:070670 32:161205 --:------ 22E5 001 00 < Invalid code for 32:161205 (FAN) to Rx/Tx: 22E5
2023-09-26 20:34:32.173 INFO (MainThread) [ramses_rf.dispatcher] || HGI:070670 | FAN:161205 | RQ | unknown_22e9     |      || {}
2023-09-26 20:34:32.174 INFO (MainThread) [ramses_rf.dispatcher] RQ --- 18:070670 32:161205 --:------ 22E9 001 00 < Invalid code for 32:161205 (FAN) to Rx/Tx: 22E9
2023-09-26 20:34:42.348 INFO (MainThread) [ramses_rf.dispatcher] || REM:179059 | FAN:161205 |  I | fan_mode         |      || {'fan_mode': 'medium', '_scheme': 'orcon', '_mode_idx': '02', '_mode_max': '04'}
2023-09-26 20:34:42.377 INFO (MainThread) [ramses_rf.dispatcher] || FAN:161205 |            |  I | fan_state        |  00  || {'exhaust_fan_speed': 0.01, 'fan_mode': '02', 'passive': False, 'damper_only': False, 'filter_dirty': False, 'frost_cycle': False, 'has_fault': False, '_flags': [0, 0, 0, 0, 0, 0, 0, 0], '_unknown_3': '00'}
2023-09-26 20:35:31.630 INFO (MainThread) [ramses_rf.dispatcher] || HGI:070670 | FAN:161205 | RQ | unknown_22f2     |      || {}
2023-09-26 20:35:31.630 INFO (MainThread) [ramses_rf.dispatcher] RQ --- 18:070670 32:161205 --:------ 22F2 001 00 < Invalid code for 32:161205 (FAN) to Rx/Tx: 22F2
2023-09-26 20:36:31.636 INFO (MainThread) [ramses_rf.dispatcher] || HGI:070670 | FAN:161205 | RQ | unknown_22f4     |      || {}
2023-09-26 20:36:31.637 INFO (MainThread) [ramses_rf.dispatcher] RQ --- 18:070670 32:161205 --:------ 22F4 001 00 < Invalid code for 32:161205 (FAN) to Rx/Tx: 22F4
2023-09-26 20:37:16.493 INFO (MainThread) [custom_components.ramses_cc.coordinator] Saving the client state cache (packets, schema)...
2023-09-26 20:42:16.493 INFO (MainThread) [custom_components.ramses_cc.coordinator] Saving the client state cache (packets, schema)...
`

pocketware, 2023-09-26:
Update: No packages received after 2023-09-26T20:36:31.636218.
Current time is: 2023-09-26T20:57:35

zxdavb, 2023-09-26:
I see nothing wrong with the configuration / packet logs you have provided.

I would suggest you utilize the cache (rather than disabling it): 
```yaml
restore_cache: true  # recommended
```

zxdavb, 2023-09-26:
No matter which option you choose, keep:
```yaml
ramses_cc:
  serial_port: /dev/serial/by-id/usb-SparkFun_evofw3_atmega32u4-if00
  packet_log: packet.log

  ramses_rf:
    enforce_known_list: true
  
  orphans_hvac: [18:070670, 32:161205, 29:179059]
  known_list:
    32:161205: {class: FAN}
    18:070670: {class: HGI}
    29:179059: {class: REM}
```

I believe the codes you want are `22F1` and `22F3` (rather than says `22F2`, `22F4`), no?

> Adding configuration like commands:, faked: etc. or enabling send_package is causing issues.

You don't make clear what issues you're having. What are you seeing, and what do you expect to see, instead?

For example:
```
Invalid code for 32:161205 (FAN) to Rx/Tx: 22F4
```

I assume you've created an automation, or some such? AFAIK, almost all FANs don't deal with 22F2/22F4 commands.

pocketware, 2023-09-27:
Hello David,

Using a config without the orphans_hvac and known_list items the SSM-D2 is generation log records in de package.log file.
No issues with this config, except I don't get any sensors in HomeAssistant; which is correct, because there is no configuration.

When I add the  orphans_hvac and known_list items the capturing is working for a few minutes. After I get the RQ messages in the log with the invalid code the device stops working.

I don't have any automation configured. Just want to see the sensors from my Orcon HRC-300-EcoMax. That's why I have the
`send_packet: false`
in the config.

I really dont know where the invalid code requests are coming from; I suppose the ramses_rf or ramses_cc code.

zxdavb, 2023-10-02:
Please send me a packet log, minimum 24h, but longer is better.

You can post (say) file hosting links here if you're willing, or PM me that stuff via [community.home-assistant.io](https://community.home-assistant.io/t/honeywell-ch-dhw-via-rf-evohome-sundial-hometronics-chronotherm/151584/3523): I am [zxdavb](https://community.home-assistant.io/u/zxdavb/summary)

pocketware, 2024-09-28:
Closed issue because SSM-D2 device is dead and I'm now using the Ramses_ESP.

closed 2024-09-28 by pocketware