ramses-rf/ramses_rf #86: Support for HCC100

woepies, 2023-10-01:
Honeywell / Resideo introduced the HCC100 underfloor heating controller. I installed it as part of my Evohome setup, replacing my HCE80. Can you please add support for the HCC100 in ramses_rf and ramses_cc? I would be happy to provide packet logs, etc.
I think my HCC100 is device 02:066404. The Home Assistant log includes this line:
2023-10-01 10:03:07.436 WARNING (MainThread) [ramses_rf.protocol.message] I --- 02:066404 --:------ 02:066404 3150 008 2100220023002400 < Corrupt payload: Payload doesn't match '^((0[0-9A-F])[0-9A-F]{2}|FC[0-9A-F]{2})+$': 2100220023002400

zxdavb, 2023-10-02:
UFH is a 2nd-class citizen in **ramses_rf** - a situation I would love to address.

Indeed, I have purchased a HCE80 for that purpose, but I am very time-poor.

Nonetheless, please do send me a packet log, and I can get to work on tweaking the parsers, which will address the above error message, at least.

What would be great would be at least 24h of packet logs (and 3d even better), with the UFH actually turning on/off.

I would also like to know your setup, and specifically if you're using OpenTherm. Which UFH zone are set to which Evohome zones would also be useful

Finally, please let me know what you mean when you want the HCC100 supported. What functionality would you like? I envisage exposing as much info as possible as HA sensors, or is there more?

You can post links here if you're willing, or PM me that stuff via [community.home-assistant.io](https://community.home-assistant.io/t/honeywell-ch-dhw-via-rf-evohome-sundial-hometronics-chronotherm/151584/3523): I am [zxdavb](https://community.home-assistant.io/u/zxdavb/summary)

Thanks

woepies, 2023-10-08:
Outdoor temperatures are still high in the Netherlands and the CH (incl UFH) is switched off. I will send the packet log once the system is on. I was using on/off switched (BDR91) to control the boiler, but recently switched to OpenTherm because the HCC100 is supposed to support it. I still have the BDR91, so I can switch back to experiment.

In the meantime I attempted to study the HCC100 behavior myself. 

- The HCC100 sends a `10E0` Device Info packet when booting. The structure is the same as described on the wiki. However, the payload starts with `FF`, causing the parser to refuse it,
`10:59:43.614  I --- 02:066404 --:------ 02:066404 10E0 038 FF0001C89D6E0600FEFF140107E7010C07E64843453130302D524144494F0000000000000000`

description: "HCE100-RADIO"
firmware: "20-01-2023"
manufactured: "01-12-2022"

zxdavb, 2023-10-09:
I have addressed the above with commit a86bd3f9733a1ea523bc819873e0cb97861fcbc9 
 - Add HCE100's 10E0 to protocol

rvdm, 2025-02-26:
Found this issue looking for more info on the HCC100 in ramses. I'm running evohome and just added a hcc100 to the system; have been running ramses_rf and the home assistant integration for a little over a year now. I run a mix of traditional heating zones combined with ufh (now controlled by the HCC100). The heating system is OpenTherm via a nefit OpenTherm-EMS converter.

If there's anything I can do (packet logs, experiments) I'm happy to help. I'm also interested in 'as much information as possible' in HA sensors.

silverailscolo, 2025-09-01:
@rvdm Can you add a screen shot of the HCC100 device as it appears in the Ramses RF integration, so we get an idea of the sensors etc it currently reports?

I have no evohome.
If you see warnings or interesting additional packets from this device in you logs, add them here as file attachments.

rvdm, 2025-09-18:
> [@rvdm](https://github.com/rvdm) Please see what the (small) change in PR [#270](https://github.com/ramses-rf/ramses_rf/pull/270) means for this issue.
> 
> In tests (that failed otherwise) I no longer see the TypeError mentioned in ramses_cc Issue [295](https://github.com/ramses-rf/ramses_cc/issues/295)

Cool! Will test. Will this be packaged in a release? If so, I'll wait for that if that's OK, otherwise I don't know how to easily build and test it in my HA instance ;)

silverailscolo, 2025-09-21:
> Anything particular to look for?

In issue #295 we see: `TypeError: Invalid device type to be a controller: 02:033535 (UFC): None` (also `0,0` etc).
Do you sometimes see that too? The change intends to fix that, so that the code can create an UFC controller.
Your system uses one of those too, doesn't it?

RUrlus, 2025-12-18:
If this is still something you're willing to work on, I would like to help out where possible. I have two HCC100s, one covering a single zone and one covering two zones. I keep about 14 days of logs (using Ramses CC in HA).

I'm mostly interested in actuator positioning to get a feeling how "stretched" the system is for a given input temperature.
I tend to see regex failures for the HCC100s in the logs like below:
```
Logger: ramses_tx.message
Source: runner.py:289
First occurred: 14 December 2025 at 22:00:40 (3106 occurrences)
Last logged: 22:57:57

I --- 02:079600 --:------ 02:079600 3150 002 2106 < Payload doesn't match '^((0[0-9A-F])[0-9A-F]{2}|FC[0-9A-F]{2})+$': 2106
I --- 02:080124 --:------ 02:080124 3150 004 210C22B2 < Payload doesn't match '^((0[0-9A-F])[0-9A-F]{2}|FC[0-9A-F]{2})+$': 210C22B2
I --- 02:079600 --:------ 02:079600 30C9 018 E17FFFE27FFFE37FFFE47FFFE57FFFE67FFF < Payload doesn't match '^(0[0-9A-F][0-9A-F]{4})+$': E17FFFE27FFFE37FFFE47FFFE57FFFE67FFF
I --- 02:080124 --:------ 02:080124 3150 004 210E22C0 < Payload doesn't match '^((0[0-9A-F])[0-9A-F]{2}|FC[0-9A-F]{2})+$': 210E22C0
I --- 02:080124 --:------ 02:080124 30C9 018 E17FFFE27FFFE37FFFE47FFFE57FFFE67FFF < Payload doesn't match '^(0[0-9A-F][0-9A-F]{4})+$': E17FFFE27FFFE37FFFE47FFFE57FFFE67FFF
```

RUrlus, 2025-12-23:
@silverailscolo for after the holidays, attached is another day off logs with SQL MessageIndex on (ramses-sql-mi.log.2025-12-22) and the same day with it off (ramses.log.2025-12-22) from a secondary

[ramses_logs.zip](https://github.com/user-attachments/files/24317851/ramses_logs.zip)

 HA instance.

silverailscolo, 2026-08-16:
See #933, closing this.

If HCC100 UFC features are missing or off, please create a new issue

closed 2026-08-16 by silverailscolo