ramses-rf/ramses_rf #87: [Itho CVE-S CO2 with AUTO RFT-N] Fan states incorrectly recognised

FrankvdAa, 2023-11-12:
Hi,

Created an issue with evohome-Listener (https://github.com/smar000/evoGateway/issues/61), but problem seems to be with ramses_rf.

I have a Itho Daalderop CVE-S CO2 unit with AUTO RFT-N remote controls, with on the remote controls the following options:
* Low
* High
* Auto
* Timer

Apart from de timer option, all settings seem to be wrongly recognised.
* Low: detected as medium
* High: detected as auto
* Auto: detected as high

In below captures, I pressed Low, High, Auto. I also have an Evohome controller, that's were the setpoint message is coming from.

These captures are from evohome-Listener as the `-o packet.log` option doesn't seem to work on my ramses_rf setup.

```
2023-11-12 19:48:17 |000| HGI evoGateway        -> NUL UNBOUND           | I| puzzle_packet   | datetime: 2023-11-12T19:48:17.729, engine: v0.22.40, parser: v0.22.40
2023-11-12 19:48:24 |044| 37:198669             -> 37:019136             | I| fan_mode        | medium
2023-11-12 19:48:24,593 [301] ----> Unknown zone for src: '37:198669 (HVC) '
2023-11-12 19:48:27 |064| TRV 04:007044         -> CTL Controller        | I| setpoint        | 15.00°C                                                                     [Zone 06 ]
2023-11-12 19:48:27,834 [804] Both 'target_zone_id' and 'src_zone_id' not found in ZONES
2023-11-12 19:48:28 |045| 37:198669             -> 37:019136             | I| fan_mode        | auto
2023-11-12 19:48:28,270 [301] ----> Unknown zone for src: '37:198669 (HVC) '
2023-11-12 19:48:31 |043| 37:198669             -> 37:019136             | I| fan_mode        | high
2023-11-12 19:48:31,789 [301] ----> Unknown zone for src: '37:198669 (HVC) '
```

```
2023-11-12T19:48:17.721425 # ramses_rf 0.22.40
2023-11-12T19:48:17.777540 000  I --- 18:003618 63:262142 --:------ 7FFF 016 0010018BC4DCF84176302E32322E3430
2023-11-12T19:48:24.587738 044  I --- 37:198669 37:019136 --:------ 22F1 003 000204
2023-11-12T19:48:27.828919 064  I --- 04:007044 --:------ 01:204708 2309 003 0605DC
2023-11-12T19:48:28.266746 045  I --- 37:198669 37:019136 --:------ 22F1 003 000404
2023-11-12T19:48:31.786152 043  I --- 37:198669 37:019136 --:------ 22F1 003 000304
```

Below output is from a similar test with ramses_rf:

```
2023-11-12T20:16:24.376134 ||  37:198669 |  37:019136 |  I | fan_mode         |      || {'fan_mode': 'medium', '_scheme': 'orcon', '_mode_idx': '02', '_mode_max': '04'}
2023-11-12T20:16:28.604293 ||  37:198669 |  37:019136 |  I | fan_mode         |      || {'fan_mode': 'auto', '_scheme': 'orcon', '_mode_idx': '04', '_mode_max': '04'}
2023-11-12T20:16:28.644141 ||  37:019136 |  37:198669 |  I | hvac_state       |      || {'fan_info': 'speed 3, high', '_unknown_fan_info_flags': [0, 0, 0], 'air_quality': 1.0, 'air_quality_basis': 'rel_humidity', 'co2_level': 757, 'indoor_humidity': 0.55, 'outdoor_humidity': None, 'exhaust_temp': None, 'supply_temp': None, 'indoor_temp': None, 'outdoor_temp': None, 'speed_capabilities': ['off', 'low_med_high', 'timer', 'boost', 'auto'], 'bypass_position': None, 'exhaust_fan_speed': 0.395, 'supply_fan_speed': 0.0, 'remaining_mins': None, 'post_heat': None, 'pre_heat': None, 'supply_flow': None, 'exhaust_flow': None}
2023-11-12T20:16:33.184460 ||  37:198669 |  37:019136 |  I | fan_mode         |      || {'fan_mode': 'high', '_scheme': 'orcon', '_mode_idx': '03', '_mode_max': '04'}
2023-11-12T20:16:36.879805 ||  37:198669 |  37:019136 |  I | fan_boost        |      || {'minutes': 10, 'flags': [0, 0, 0, 0, 0, 0, 0, 0], '_new_speed_mode': 'fan_boost', '_fallback_speed_mode': None}
```

Let me know if I can help with anything else.

Regards,

Frank

Kars-de-Jong, 2026-08-18:
@silverailscolo Why is this issue closed as "not planned"? Surely we need this - maybe not autodetected but at least the option to specify the manufacturer in the schema so payloads can be interpreted correctly?

silverailscolo, 2026-08-18:
> "We need this"

You are always invited to open a PR. There has been no activity since months.

Kars-de-Jong, 2026-08-18:
@silverailscolo How would activity on this be even possible with you refactoring just about everything? As you commented yourself, there were at least several steps required to even start with this.
Anyway, isn't this particular issue fixed by https://github.com/ramses-rf/ramses_rf/pull/734?

closed 2026-08-18 by silverailscolo