ramses-rf/ramses_rf #90: DTR4 fingerprint

trvrnrth, 2023-12-01:
I have a couple of DTR4 thermostats which report device info as follows:
```
I --- 22:014583 63:262142 --:------ 10E0 038 000002FF1F00FFFFFFFFFFFFFFFFFFFFFFFF4454342030323200C95838F749AFA556B284EBF9
I --- 22:014503 63:262142 --:------ 10E0 038 000002FF1F00FFFFFFFFFFFFFFFFFFFFFFFF4454342030323200C95838A77CF588003884ACE5
```

I'm happy to provide further packet logs if they are helpful, however I am only using them as zone sensors with an evohome controller so probably won't be able to capture all possible actions.

zxdavb, 2023-12-01:
These go in **tests/parsers/code_10e0.log** as well as **fingerprint.py** - thanks.

zxdavb, 2023-12-01:
Once you add them to **tests/parsers/code_10e0.log**, then:

... in **src/ramses_cli/client.py**, change `_DEV_MODE = True`, and: 

```
> cat tests/tests/parsers/code_10e0.log | head | python client.py -lf parse
```

zxdavb, 2023-12-01:
... what follows next is a mess - leave it to me.

zxdavb, 2023-12-01:
This now works:
```
> pytest tests/tests/test_parsers.py::test_parsers_from_log_files[code_10e0.log]
=========================================================================================================== test session starts ============================================================================================================
platform linux -- Python 3.11.6, pytest-7.4.3, pluggy-1.2.0
rootdir: /home/dbonnes/clients/ramses_rf
configfile: pyproject.toml
plugins: anyio-4.0.0, xdist-3.3.1, asyncio-0.21.1
asyncio: mode=Mode.AUTO
collected 1 item

tests/tests/test_parsers.py .                                                                                                                                                                                                        [100%]
```

closed 2023-12-01 by zxdavb

zxdavb, 2023-12-01:
DTS4 supports Ramses-III, whatever that is. If you have a newer OTB (R8820) or a HCC100, they may support Ramses-III as well....

If you could ping me a 24h packet.log, there may be some insights to be had.

Feel free to submit more PRs!