ramses-rf/ramses_rf #93: Packet report for device type 22. DT4R thermostat

BillComben, 2023-12-05:
I don't know if you are still interested in this, but my Ramses_RF script advised that I reported the  following message
```
11:03:56.594 ||  22:006128 |            |  I | device_battery   |      || {'battery_low': False, 'battery_level': 0.0}
11:03:56.716 ||  22:006100 |  01:023621 |  W | ufh_setpoint     |  00  || {'ufh_idx': '00', 'mode': 'heat', 'setpoint_bounds': (16.0, 25.5)}
11:04:02.185  I --- 22:006100 63:262142 --:------ 10E0 038 000002FF1F00FFFFFFFFFFFFFFFFFFFFFFFF4454342030313700C95817D4D7B6CD12902AED17 < Support the development of ramses_rf by reporting this packet, with the make/model of device: DT2:006100 (device type 22 not known to have signature: 0002FF1F00FFFFFFFF)
11:04:02.185  I --- 22:006100 63:262142 --:------ 10E0 038 000002FF1F00FFFFFFFFFFFFFFFFFFFFFFFF4454342030313700C95817D4D7B6CD12902AED17 < AssertionError(C95817D4D7B6CD12902AED17)
Traceback (most recent call last):
  File "C:\Users\WDC\Evohome\ramses_rf-master/src\ramses_tx\message.py", line 259, in _validate
    result = parse_payload(self)
             ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\WDC\Evohome\ramses_rf-master/src\ramses_tx\parsers.py", line 2840, in parse_payload
    result = _PAYLOAD_PARSERS.get(msg.code, parser_unknown)(msg._pkt.payload, msg)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\WDC\Evohome\ramses_rf-master/src\ramses_tx\parsers.py", line 959, in parser_10e0
    assert msg.verb == RP or not unknown, f"{unknown}"
AssertionError: C95817D4D7B6CD12902AED17
11:04:02.442 ||  22:006100 |            |  I | relay_failsafe   |      || {'failsafe_enabled': False, 'unknown_0': 'FF'}
```
It is monitoring a newly installed 12 Zone Evohome system, The type 22 devices on the system are actually Honeywell DT4R wireless thermostats.

Thanks for your efforts in developing this script, it is proving invaluable.

Bill

peternash, 2024-01-22:
ramses_cc 0.31.x also reports this as a warning:
`
2024-01-21 10:27:54.238 WARNING (MainThread) [ramses_rf.dispatcher]  W --- 22:012299 01:216136 --:------ 22C9 006 01076C09F601 < PacketInvalid( W --- 22:012299 01:216136 --:------ 22C9 006 01076C09F601 < Unexpected code for src to Tx)
`
Relevant packet log line?:
`
2024-01-21T10:27:54.236890 039  W --- 22:012299 01:216136 --:------ 22C9 006 01076C09F601
`
See [https://community.home-assistant.io/t/honeywell-ch-dhw-via-rf-evohome-sundial-hometronics-chronotherm/151584/4112?u=peternash](url)

zxdavb, 2024-04-19:
See: https://community.home-assistant.io/t/honeywell-ch-dhw-via-rf-evohome-sundial-hometronics-chronotherm/151584/4372

**ramses_rf** is not able to stop itself from throwing an exception for this message, even though it should not be doing so.

Fixing this issue would require re-writing a portion of the code that is deep within it's core logic - a risky business that I won't take on at the moment.

I may need to address this in the future...