ramses-rf/ramses_rf #101: Reporting packet 01FF: message parsing fails

Onl1ne1373, 2024-01-29:
16:53:49.052 RQ --- 21:041873 18:010547 --:------ 01FF 026 00802A2A2CD0008000243028320000440280800280FF800400A4 < AssertionError(Support the development of ramses_rf by reporting this packet (002430))
Traceback (most recent call last):
  File "/home/online/src/ramses_rf/src/ramses_tx/message.py", line 261, in _validate
    result = parse_payload(self)
             ^^^^^^^^^^^^^^^^^^^
  File "/home/online/src/ramses_rf/src/ramses_tx/parsers.py", line 2873, in parse_payload
    result = _PAYLOAD_PARSERS.get(msg.code, parser_unknown)(msg._pkt.payload, msg)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/online/src/ramses_rf/src/ramses_tx/parsers.py", line 591, in parser_01ff
    assert payload[16:22] in (
           ^^^^^^^^^^^^^^^^^^^

zxdavb, 2024-01-29:
You'll have to provide more information, especially if you're sending contrived payloads.

Onl1ne1373, 2024-02-01:
As I posted in the forum I want to listen to message event in Home Assistant, to reply to the request of the Spider.
This failed, I do not get events. With a little fooling around with the regular expression I found out that all 01FF messages are missing. I think this might be related to the above parsing problem.

```yaml
alias: Thermostaat Onl1ne1373 Setpoint Reply
description: ""
trigger:
  - platform: event
    event_type: ramses_cc_message
condition:
  - condition: template
    value_template: "{{ trigger.event.data.code == \"01FF\" }}"
  - condition: template
    value_template: "{{ trigger.event.data.src == \"21:033620\" }}"
action:
  - service: ramses_cc.send_packet
    data:
      device_id: "21:033620"
      verb: RP
      code: 01FF
      payload: |
        008080{{ ('%x' % ((state_attr('climate.thermostat_onl1ne1373',
         'temperature') | float * 2)) | round) | upper }}28D0000000143C80800000B40080800280FF80040000
mode: single
```

Onl1ne1373, 2024-02-01:
Examples of packets gecaptured. Spider thermostat communicating with Spider Connect Gateway:

```log
072  W --- 18:011649 21:059180 --:------ 01FF 026 0080802727D0000000143C80800000B40080800280FF80040000
066  I --- 21:059180 18:011649 --:------ 01FF 026 0080282729D0008000143C28400000C40280800280FF80040084
073 RQ --- 21:059180 18:011649 --:------ 01FF 026 008028282AD0008000243028400000C40280800280FF80040084
067 RP --- 18:011649 21:059180 --:------ 01FF 026 0080802828D0000000143C80800000B40080800280FF80040000
```

zxdavb, 2024-02-03:
Can you please send me every `01FF` packet you have? Preferably in packet logs, so that I can look at them in context. You can PM them to me.

Onl1ne1373, 2024-02-04:
I did send you the packet logs I have.

Regarding the D0 field, these Spiders should have day/night temperatures according to the manual, but this is disabled because they are connected to a Heat pump WPU5G and Spider Autotemp floor control. You cannot change that on the thermostat.

zxdavb, 2024-02-05:
Closing in favour of #73

closed 2024-02-05 by zxdavb