ramses-rf/ramses_rf #102: Unexpected code for src to Tx for  Itho Daalderop Autotemp zone control

Onl1ne1373, 2024-01-31:
Loads of error messages a day in the Home Assistant log 

02:246902 = Master Itho Daalderop Autotemp
02:246873 = Slave Itho  Daalderop Autotemp
02:248278 = Slave Daalderop Itho Autotemp

Autotemp is a zone control system: https://www.ithodaalderop.nl/nl-NL/professional/product/536-5120

```log
Logger: ramses_rf.dispatcher
Source: runner.py:188
First occurred: 30 januari 2024 om 14:30:44 (13026 occurrences)
Last logged: 12:54:59

I --- 02:248278 --:------ 02:248278 3110 004 00000000 < PacketInvalid( I --- 02:248278 --:------ 02:248278 3110 004 00000000 < Unexpected code for src to Tx)
I --- --:------ --:------ 02:248278 1FD4 003 00223F < PacketInvalid( I --- --:------ --:------ 02:248278 1FD4 003 00223F < Unexpected code for src to Tx)
I --- 02:246873 02:246902 --:------ 4E01 018 007FFF07EB7FFF7FFF7FFF7FFF7FFF7FFF00 < PacketInvalid( I --- 02:246873 02:246902 --:------ 4E01 018 007FFF07EB7FFF7FFF7FFF7FFF7FFF7FFF00 < Unexpected code for src to Tx)
I --- --:------ --:------ 02:246902 1FD4 003 009339 < PacketInvalid( I --- --:------ --:------ 02:246902 1FD4 003 009339 < Unexpected code for src to Tx)
I --- --:------ --:------ 02:246873 1FD4 003 007D9D < PacketInvalid( I --- --:------ --:------ 02:246873 1FD4 003 007D9D < Unexpected code for src to Tx)
```

Onl1ne1373, 2024-02-05:
These are two messages from a slave Autotemp to the master Autotemp.

It sends the room temperature to the main Autotemp of two Spider thermostats that are connected to that Autotemp.
Strangely it uses a message for every room, while it could do it in one message.

```log
17:51:17.955  I --- 02:248278 02:246902 --:------ 4E01 018 007FFF07F57FFF7FFF7FFF7FFF7FFF7FFF00 < PacketInvalid( I --- 02:248278 02:246902 --:------ 4E01 018 007FFF07F57FFF7FFF7FFF7FFF7FFF7FFF00 < Unexpected code for src to Tx)
18:38:22.519 ||  02:248278 |  02:246902 |  I | hvac_4e01        |      || {'temperatures': [None, 20.37, None, None, None, None, None, None]}
17:51:17.956  I --- 02:248278 02:246902 --:------ 4E01 018 0007F17FFF7FFF7FFF7FFF7FFF7FFF7FFF00 < PacketInvalid( I --- 02:248278 02:246902 --:------ 4E01 018 0007F17FFF7FFF7FFF7FFF7FFF7FFF7FFF00 < Unexpected code for src to Tx)
18:39:09.337 ||  02:248278 |  02:246902 |  I | hvac_4e01        |      || {'temperatures': [20.33, None, None, None, None, None, None, None]}
```

To give you some extra info:
The master Autotemp collects all temperatures, setpoints and requested heat demand per room from the slave Autotemp units and calculates the desired total power in percentage. That power percentage is send to the master Spider thermostat and he sends via wired OpenTherm to the Heat Pump (WPU5G).

I have a Addon board connected to the service port of the Autotemp: https://www.nrgwatch.nl

[image]

silverailscolo, 2025-08-13:
Assuming this related to/same setup as issue #102

closed 2025-09-17 by silverailscolo