ramses-rf/ramses_rf #103: Packet reporting for Spider Thermostat

Onl1ne1373, 2024-01-31:
Home Assistant log asked to report this packet.

21:041873 is a Spider Thermostat connected by OpenTherm to my Heat Pump (Itho Daalderop WPU 5G).
It is also wireless connected to a Itho Daalderop Autotemp zone control system (master).

21:033620 is a Spider Room Thermostat wireless connected to a Itho Daalderop Autotemp (slave).

```log
Logger: ramses_tx.parsers
Source: runner.py:188
First occurred: 30 januari 2024 om 14:30:13 (72 occurrences)
Last logged: 12:44:11

I --- 21:041873 --:------ 21:041873 3EF0 006 020000050210 < Support the development of ramses_rf by reporting this packet (byte 5: 10), with a description of your system
I --- 21:041873 --:------ 21:041873 3EF0 006 0200000D0210 < Support the development of ramses_rf by reporting this packet (byte 5: 10), with a description of your system
I --- 21:041873 --:------ 21:041873 3EF0 006 0200000C0210 < Support the development of ramses_rf by reporting this packet (byte 5: 10), with a description of your system
I --- 21:041873 --:------ 21:041873 3120 007 00F0B0060006FF < Support the development of ramses_rf by reporting this packet (byte 1: F0)
I --- 21:033620 --:------ 21:033620 3120 007 00F0B0060006FF < Support the development of ramses_rf by reporting this packet (byte 1: F0)
```

zxdavb, 2024-02-05:
Just remind me - what Spider kit is there?

Onl1ne1373, 2024-02-05:
It is a Itho Daalderop Spider Thermostat type 03-00476
https://www.ithodaalderop.nl/nl-NL/professional/product/03-00476

closed 2025-09-16 by silverailscolo