ramses-rf/ramses_rf #107: Remeha OEM / Status codes sensor

jsimonetti, 2024-02-13:
Hi,
Thank you for your hard work!

I managed to decipher the oem_code for a Remehe Calenta HR unit and wanted to share it with you.
Unfortunately it is in Dutch, as I lost the english version of the manual, but it should be very easy to translate if you know the status codes. Unfortuantely, binary arithmetic is hard in jinja, so I had to make do with some substitutions, but it works fine afaict.

You should substitute the `sensor.10_041226_oem_code` with the correct `oem_code` sensor for this to work on your system.

```yaml
sensor:
  - platform: template
    sensors:
      remeha_status:
        value_template: >-
          {{ state_attr("sensor.remeha_status", "status") }}
        attribute_templates:
          oem_code: >-
            {% if states('sensor.10_041226_oem_code') == "unavailable" %}
            {{ state_attr("sensor.remeha_status", "oem_code")|default(0) }}
            {% else %}
            {{ states('sensor.10_041226_oem_code')|int }}
            {% endif %}
          status_code: >-
            {% set oemcode = state_attr("sensor.remeha_status", "oem_code") %}
            {% set hex = '%0x' % oemcode | int %}
            {{ hex[-3]|default(0) }}
          substatus_code: >-
            {% set oemcode = state_attr("sensor.remeha_status", "oem_code") %}
            {% set hex = '%0x' % oemcode | int %}
            {% set lowstr = "0x" + hex[-2] + hex[-1] %}
            {{ lowstr | int(base=16,default=0) }}
          status: >-
            {% set status = state_attr("sensor.remeha_status", "status_code") %}
            {% if status == 0 %}Rust
            {% elif status == 1 %}Ketel start
            {% elif status == 2 %}Brander start
            {% elif status == 3 %}Branden op CV-bedrijf
            {% elif status == 4 %}SWW-bedrijf actief
            {% elif status == 5 %}Branderstop
            {% elif status == 6 %}Ketelstop (einde warmtevraag)
            {% elif status == 8 %}Regelstop
            {% elif status == 9 %}Blokkering
            {% elif status == 7 %}Ontluchten
            {% elif status == 17 %}Ontluchten
            {% else %}Unknown
            {% endif %}
          substatus: >-
            {% set status = state_attr("sensor.remeha_status", "status_code") %}
            {% set substatus = state_attr("sensor.remeha_status", "substatus_code") %}
            {% if status == 0 %}Rust
            {% elif status == 1 %}
              {% if substatus == 1 %}Anti-pendel
              {% elif substatus == 2 %}Driewegklep aansturen
              {% elif substatus == 3 %}Start pomp
              {% elif substatus == 4 %}Wacht op juiste temperatuur voor branderstart
              {% else %}{{ substatus }}
              {% endif %}
            {% elif status == 2 %}
              {% if substatus == 10 %}Open rookgasklep
              {% elif substatus == 11 %}Verhogen toerental ventilator
              {% elif substatus == 13 %}Voorventileren
              {% elif substatus == 14 %}Wacht op releasesignaal
              {% elif substatus == 15 %}Brander aan
              {% elif substatus == 17 %}Voorontsteking
              {% elif substatus == 18 %}Hoofdontsteking
              {% elif substatus == 19 %}Vlamdetectie
              {% elif substatus == 20 %}Tussenventileren
              {% else %}{{ substatus }}
              {% endif %}
            {% elif status == 3 %}
              {% if substatus == 30 %}Temperatuurregeling
              {% elif substatus == 31 %}Begrende temperatuurregeling (ΔT bveveiliging)
              {% elif substatus == 32 %}Vermogensregeling
              {% elif substatus == 33 %}Bescherming temperatuurgradient niveau 1 (Terugmoduleren)
              {% elif substatus == 34 %}Bescherming temperatuurgradient niveau 2 (Laaglast)
              {% elif substatus == 35 %}Bescherming temperatuurgradient niveau 3 (Blokkering)
              {% elif substatus == 36 %}Opmodeleren voor vlambeveiliging
              {% elif substatus == 37 %}Temperatuurstabilisatietijd
              {% elif substatus == 38 %}Koude start
              {% else %}{{ substatus }}
              {% endif %}
            {% elif status == 4 %}
              {% if substatus == 30 %}Temperatuurregeling
              {% elif substatus == 31 %}Begrensde temperatuurregeling (ΔT bveveiliging)
              {% elif substatus == 32 %}Vermogensregeling
              {% elif substatus == 33 %}Bescherming temperatuurgradient niveau 1 (Terugmoduleren)
              {% elif substatus == 34 %}Bescherming temperatuurgradient niveau 2 (Laaglast)
              {% elif substatus == 35 %}Bescherming temperatuurgradient niveau 3 (Blokkering)
              {% elif substatus == 36 %}Opmodeleren voor vlambeveiliging
              {% elif substatus == 37 %}Temperatuurstabilisatietijd
              {% elif substatus == 38 %}Koude start
              {% else %}{{ substatus }}
              {% endif %}
            {% elif status == 5 %}
              {% if substatus == 40 %}Brander uit
              {% elif substatus == 41 %}Naventileren
              {% elif substatus == 42 %}Sluit rookgasklep
              {% elif substatus == 43 %}Recirculatiebeveiliging
              {% elif substatus == 44 %}Stop ventilator
              {% else %}{{ substatus }}
              {% endif %}
            {% elif status == 6 %}
              {% if substatus == 60 %}Pomp nadraaien
              {% elif substatus == 61 %}Pomp uit
              {% elif substatus == 62 %}Driewegklep aansturen
              {% elif substatus == 63 %}Start anti-pendel
              {% else %}{{ substatus }}
              {% endif %}
            {% elif status == 7 %}
              {% if substatus == 0 %}Rust
              {% elif substatus == 2 %}Driewegklep aansturen
              {% elif substatus == 3 %}Start pomp
              {% elif substatus == 61 %}Pomp uit
              {% elif substatus == 62 %}Driewegklep aansturen
              {% else %}{{ substatus }}
              {% endif %}
            {% elif status == 8 %}
              {% if substatus == 0 %}Wacht op branderstart
              {% elif substatus == 1 %}Anti-pendel
              {% else %}{{ substatus }}
              {% endif %}
            {% elif status == 9 %}
              {% if substatus == 0 %}Parameter fout
              {% elif substatus == 1 %}Maximale aanvoertemperatuur overschreden
              {% elif substatus == 2 %}Maximale stijging van de aanvoertemperatuur overschreden
              {% elif substatus == 7 %}Maximaal verschil tussen aanvoer- en retourtemperatuur overschreden
              {% elif substatus == 10 %}Blokkerende ingang is actief
              {% elif substatus == 11 %}Blokkerende ingang of vorstbeveiliging is actief
              {% elif substatus == 12 %}Communicatiefout met de HMI-print
              {% elif substatus == 13 %}Communicatiefout met de SCU-print
              {% elif substatus == 14 %}Waterdruk te laag
              {% elif substatus == 15 %}Gasdruk te laag
              {% elif substatus == 16 %}Configuratiefout
              {% elif substatus == 17 %}Configuratiefout of default parametertabel niet in orde
              {% else %}{{ substatus }}
              {% endif %}
            {% else %}{{ substatus }}
            {% endif %}
```

zxdavb, 2024-03-08:
I think this should go in the wiki.

silverailscolo, 2025-04-14:
Thanks for the template, @jsimonetti 
Added it to the [wiki](https://github.com/zxdavb/ramses_cc/wiki/8.-How-to:-Sensor-templates) with update template yaml style.
@zxdavb Can you close this issue?

closed 2025-04-20 by jsimonetti