ramses-rf/ramses_rf #123: After HA update, exception in 0.41.23 and other one in beta 0.42

rfcdejong, 2024-08-19:
I updated my HA to the latest version
- Core 2024.8.2
- Supervisor 2024.08.0
- Operating System 13.0

And also updated ramses to the latest version.
Tried latest non-beta 0.41.23

Logger: homeassistant.config_entries
Source: config_entries.py:604
First occurred: 7:58:30 AM (1 occurrences)
Last logged: 7:58:30 AM

Error setting up entry RAMSES RF for ramses_cc
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 604, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ramses_cc/__init__.py", line 89, in async_setup_entry
    await broker.async_setup()
  File "/config/custom_components/ramses_cc/broker.py", line 146, in async_setup
    await self.client.start(cached_packets=cached_packets())
  File "/usr/local/lib/python3.12/site-packages/ramses_rf/gateway.py", line 184, in start
    load_schema(self, known_list=self._include, **self._schema)  # create faked too
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/ramses_rf/schemas.py", line 378, in load_schema
    dev = _get_device(gwy, device_id)  # , **traits)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/ramses_rf/schemas.py", line 343, in _get_device
    return gwy.get_device(dev_id, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/ramses_rf/gateway.py", line 396, in get_device
    traits: dict[str, Any] = SCH_TRAITS(self._include.get(device_id, {}))
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/validators.py", line 259, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/validators.py", line 315, in _exec
    raise error if self.msg is None else AnyInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.12/site-packages/voluptuous/validators.py", line 307, in _exec
    return func(v)
           ^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 205, in __call__
    return self._compiled([], data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/validators.py", line 256, in _run
    return self._exec(self._compiled, value, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/validators.py", line 315, in _exec
    raise error if self.msg is None else AnyInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.12/site-packages/voluptuous/validators.py", line 309, in _exec
    return func(path, v)
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 779, in validate_callable
    return schema(data)
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 205, in __call__
    return self._compiled([], data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 549, in validate_dict
    return base_validate(path, data.items(), out)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 382, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: not a valid value for dictionary value @ data['class']

And beta 0.42

Logger: ramses_tx.message
Source: /usr/local/lib/python3.12/site-packages/ramses_tx/message.py:270
First occurred: 8:06:56 AM (1 occurrences)
Last logged: 8:06:56 AM

RP --- 32:160269 18:034927 --:------ 2210 042 00FF00FFFFFF0000000000FFFFFFFFFF00FFFFFF0000000000FFFFFFFFFFFFFFFF000000000000020800 < AssertionError(Support the development of ramses_rf by reporting this packet)
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/ramses_tx/message.py", line 255, in _validate
    result = parse_payload(self)
             ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/ramses_tx/parsers.py", line 2938, in parse_payload
    result = _PAYLOAD_PARSERS.get(msg.code, parser_unknown)(msg._pkt.payload, msg)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/ramses_tx/parsers.py", line 1385, in parser_2210
    assert payload in (
           ^^^^^^^^^^^^
AssertionError: Support the development of ramses_rf by reporting this packet

and warnings

Logger: ramses_rf.dispatcher
Source: runner.py:190
First occurred: 8:06:56 AM (7 occurrences)
Last logged: 8:07:13 AM

RP --- 32:160269 18:034927 --:------ 22E5 004 0050C800 < PacketInvalid(RP --- 32:160269 18:034927 --:------ 22E5 004 0050C800 < Unexpected code for src (FAN) to Tx)
RP --- 32:160269 18:034927 --:------ 22E9 004 006EC800 < PacketInvalid(RP --- 32:160269 18:034927 --:------ 22E9 004 006EC800 < Unexpected code for src (FAN) to Tx)
RP --- 32:160269 18:034927 --:------ 22F2 006 000597010641 < PacketInvalid(RP --- 32:160269 18:034927 --:------ 22F2 006 000597010641 < Unexpected code for src (FAN) to Tx)
RP --- 32:160269 18:034927 --:------ 22F4 013 0040B000000000000000200000 < PacketInvalid(RP --- 32:160269 18:034927 --:------ 22F4 013 0040B000000000000000200000 < Unexpected code for src (FAN) to Tx)
RP --- 32:160269 18:034927 --:------ 3222 003 00EF00 < PacketInvalid(RP --- 32:160269 18:034927 --:------ 3222 003 00EF00 < Unexpected code for src (FAN) to Tx)

Weird warnings

Detected blocking call to open with args ('/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/bNumInterfaces',) inside the event loop by custom integration 'ramses_cc' at custom_components/ramses_cc/broker.py, line 146: await self.client.start(cached_packets=cached_packets()) (offender: /usr/local/lib/python3.12/site-packages/serial/tools/list_ports_linux.py, line 84: with open(os.path.join(*args)) as f:), please create a bug report at https://github.com/zxdavb/ramses_cc/issues For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#open Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 674, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 641, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1990, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/setup.py", line 165, in async_setup_component result = await _async_setup_component(hass, domain, config) File "/usr/src/homeassistant/homeassistant/setup.py", line 461, in _async_setup_component await asyncio.gather( File "/usr/src/homeassistant/homeassistant/setup.py", line 463, in <genexpr> create_eager_task( File "/usr/src/homeassistant/homeassistant/util/async_.py", line 37, in create_eager_task return Task(coro, loop=loop, name=name, eager_start=True) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 752, in async_setup_locked await self.async_setup(hass, integration=integration) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 604, in async_setup result = await component.async_setup_entry(hass, self) File "/config/custom_components/ramses_cc/__init__.py", line 94, in async_setup_entry await broker.async_setup() File "/config/custom_components/ramses_cc/broker.py", line 146, in async_setup await self.client.start(cached_packets=cached_packets())

zxdavb, 2024-08-19:
Thanks for submitting, this...

It should really be multiple issues.

zxdavb, 2024-08-19:
```text
voluptuous.error.MultipleInvalid: 
not a valid value for dictionary value @ data['class']
```

The above has been fixed on **0.42.0**.

zxdavb, 2024-08-19:
```text
AssertionError: Support the development of ramses_rf by reporting this packet
```

This is 'nomral' behaviour for RAMSES RF - you should submit an issue here: https://github.com/zxdavb/ramses_rf/issues, _unless_ there is already one for that packet code (`2210`).

zxdavb, 2024-08-19:
Because the bug in RAMSES CC has been fixed, and the other issues are in RAMSES RF, I am closing this issue.

zxdavb, 2024-08-19:
To progress this issue, please provide the additional information, as described above.

rfcdejong, 2024-08-19:
> Thanks for submitting, this...
> 
> It should really be multiple issues.

sorry, for me it was one issue, with all the info I noticed.. and after I posted it, I noticed it was working again from the Beta

rfcdejong, 2024-08-19:
> ```
> AssertionError: Support the development of ramses_rf by reporting this packet
> ```
> 
> This is 'normal' behaviour for RAMSES RF - you will get this in any recent version of the integration, including the latest, **0.42.0**.
> 
> You should submit an issue here: https://github.com/zxdavb/ramses_rf/issues, _unless_ there is already one for that packet code (in this case, `2210`).
> 
> ```
> Unexpected code for src (FAN) to Tx)
> ```
> 
> The same applies to the above. I would provide a 24h packet log, including a description of teh make/model of you FAN.

There is LOTS of noice from my neighbors, every neighbor has the same FAN and CO2 sensors. I haven't configured my config to only use known devices though, call me lazy :)

See last packet logs added as zip

and my configuration

"18:034927":
  class: HGI
"32:160269":
  class: FAN
  alias: Orchon 500 ExoMax
"37:097626":
  class: REM
  faked: true
  alias: Orchon 15RF remote
  commands:
    request31DA: "RQ --- 37:097626 32:160269 --:------ 31DA 001 00 "
    request10D0: "RQ --- 37:097626 32:160269 --:------ 10D0 001 00 "
    away: " I --- 37:097626 32:160269 --:------ 22F1 003 000007"
    low: " I --- 37:097626 32:160269 --:------ 22F1 003 000107"
    medium: " I --- 37:097626 32:160269 --:------ 22F1 003 000207"
    high: " I --- 37:097626 32:160269 --:------ 22F1 003 000307"
    auto: " I --- 37:097626 32:160269 --:------ 22F1 003 000407"
    auto2: " I --- 37:097626 32:160269 --:------ 22F1 003 000507"
    boost: " I --- 37:097626 32:160269 --:------ 22F1 003 000607"
    disable: " I --- 37:097626 32:160269 --:------ 22F1 003 000707"
    bypass_open: " W --- 37:097626 32:160269 --:------ 22F7 003 00C8EF"
    bypass_close: " W --- 37:097626 32:160269 --:------ 22F7 003 0000EF"
    bypass_auto: " W --- 37:097626 32:160269 --:------ 22F7 003 00FFEF"
    high_60: " I --- 37:097626 32:160269 --:------ 22F3 007 00123C03040404"
    med_60: " I --- 37:097626 32:160269 --:------ 22F3 007 00123C02040404"
    high_30: " I --- 37:097626 32:160269 --:------ 22F3 007 00121E03040404"
    high_15: " I --- 37:097626 32:160269 --:------ 22F3 007 00120F03040404"
    reset_filter: " W --- 37:097626 32:160269 --:------ 10D0 002 00FF"
"32:100470":
  class: CO2
  alias: Orchon CO2 woonkamer
"37:218902":
  class: CO2
  alias: Orchon CO2 slaapkamer

[packet.log.zip](https://github.com/user-attachments/files/16659684/packet.log.zip)

rfcdejong, 2024-08-19:
Oops, I moved the slider to the right and get an error
```

File "/usr/local/lib/python3.12/site-packages/ramses_rf/schemas.py", line 337, in check_filter_lists
    raise LookupError(
LookupError: Can't create 01:000730: it is in the schema, but not in the known_list (check the lists and the schema)
```

zxdavb, 2024-08-19:
```text
LookupError: Can't create 01:000730: it is in the schema, but not in the known_list (check the lists and the schema)
```

This is simply a corrupt packet.

I strongly recommend you implement a known list, which exists to handle such.

rfcdejong, 2024-08-19:
> ```
> LookupError: Can't create 01:000730: it is in the schema, but not in the known_list (check the lists and the schema)
> ```
> 
> This is simply a corrupt packet.
> 
> > I haven't configured my config to only use known devices though, call me lazy :)
> 
> I strongly recommend you implement a known list, which exists to handle such. There is no other way of handling corrupt (but valid) device ids.

The documentation doesn't say where to add the known list. I only use the "config_flow" and added all Known device IDs and wanted to slide the slider "Accept packets from known devices IDs only" to the right. Don't want to go to file browser and mix config.

zxdavb, 2024-08-24:
> I only use the "config_flow" and added all Known device IDs and wanted to slide the slider "Accept packets from known devices IDs only" to the right.

That's all you need to do.

rfcdejong, 2024-08-24:
> > I only use the "config_flow" and added all Known device IDs and wanted to slide the slider "Accept packets from known devices IDs only" to the right.
> 
> That's all you need to do.

created a new issue
https://github.com/zxdavb/ramses_cc/issues/202

zxdavb, 2024-12-04:
Can we close this issue? I think it will be fixed in 0.50.0 - if not, raise an issue in ramses_cc.

silverailscolo, 2025-07-30:
`2210` parser was fixed in #151. Closing this issue

closed 2025-07-30 by silverailscolo