ramses-rf/ramses_rf #128: Not working after 2024.9.1 update

sciorty, 2024-09-09:
This scheme always worked:

```
#RAMSES RF
ramses_cc:
  scan_interval: 3
  serial_port: /dev/ttyACM0
  #packet_log: packets.log
  advanced_features:
    send_packet: true
  ramses_rf:
    enforce_known_list: true
  orphans_hvac: [32:142858, 32:126818]
  known_list:
    18:136974:
        class: HGI
    32:142858:
        class: FAN
    32:126818:
      class: REM
      faked: true
      commands:
        away:         ' I --- 32:126818 32:142858 --:------ 22F1 003 000007'
        low:          ' I --- 32:126818 32:142858 --:------ 22F1 003 000207'
        medium:       ' I --- 32:126818 32:142858 --:------ 22F1 003 000307'
        high:         ' I --- 32:126818 32:142858 --:------ 22F1 003 000707'
        bypass_open:  ' W --- 32:126818 32:142858 --:------ 22F7 003 00C8EF'
        bypass_close: ' W --- 32:126818 32:142858 --:------ 22F7 003 0000EF'
        bypass_auto:  ' W --- 32:126818 32:142858 --:------ 22F7 003 00FFEF'
        reset_filter: ' W --- 32:126818 32:142858 --:------ 10D0 002 00FF'
```

After the update I get:
```
024-09-09 17:39:03.379 DEBUG (MainThread) [custom_components.ramses_cc] Setting up entry c6509887066098f382c00f2221a2dc12...
2024-09-09 17:39:03.380 DEBUG (MainThread) [custom_components.ramses_cc.broker] Config = {'ramses_rf': {'enforce_known_list': True}, 'serial_port': {'port_name': '/dev/serial/by-id/usb-SparkFun_evofw3_atmega32u4-if00'}, 'scan_interval': 3, 'advanced_features': {'send_packet': True}, 'known_list': {'18:136974': {'class': 'HGI'}, '32:142858': {'class': 'FAN'}, '32:126818': {'class': 'REM', 'faked': True, 'commands': {'away': ' I --- 32:126818 32:142858 --:------ 22F1 003 000007', 'low': ' I --- 32:126818 32:142858 --:------ 22F1 003 000207', 'medium': ' I --- 32:126818 32:142858 --:------ 22F1 003 000307', 'high': ' I --- 32:126818 32:142858 --:------ 22F1 003 000707', 'bypass_open': ' W --- 32:126818 32:142858 --:------ 22F7 003 00C8EF', 'bypass_close': ' W --- 32:126818 32:142858 --:------ 22F7 003 0000EF', 'bypass_auto': ' W --- 32:126818 32:142858 --:------ 22F7 003 00FFEF', 'reset_filter': ' W --- 32:126818 32:142858 --:------ 10D0 002 00FF'}}}, 'schema': {'orphans_hvac': ['32:142858', '32:126818']}}
2024-09-09 17:39:03.383 DEBUG (MainThread) [custom_components.ramses_cc.broker] Storage = {'client_state': {'schema': {'main_tcs': None, 'orphans_heat': [], 'orphans_hvac': ['32:142858']}, 'packets': {'2024-09-08T16:01:40.788000': '...  I --- 32:142858 63:262142 --:------ 10E0 038 000001C88E0C0A6AFEFFFFFFFFFF1D0307E5564D442D30325250533636000000000000000000 # 10E0| I|32:142858', '2024-09-08T19:32:58.114000': '...  I --- 32:142858 --:------ 32:142858 313F 009 007C041246170207D0 # 313F| I|32:142858', '2024-09-08T19:47:09.290000': '...  I 005 32:142858 --:------ 32:142858 31D9 017 003A020020202020202020202020202008 # 31D9| I|32:142858|00 (00)', '2024-09-08T19:52:09.702000': '...  I --- 32:142858 --:------ 32:142858 31DA 030 00EF007FFFEFEF096B7FFF0A0708B1F000C84150500000EFEF7FFF7FFF00 # 31DA| I|32:142858|00 (00)'}}, 'remotes': {'32:126818': {'away': ' I --- 32:126818 32:142858 --:------ 22F1 003 000007', 'low': ' I --- 32:126818 32:142858 --:------ 22F1 003 000207', 'medium': ' I --- 32:126818 32:142858 --:------ 22F1 003 000307', 'high': ' I --- 32:126818 32:142858 --:------ 22F1 003 000707', 'bypass_open': ' W --- 32:126818 32:142858 --:------ 22F7 003 00C8EF', 'bypass_close': ' W --- 32:126818 32:142858 --:------ 22F7 003 0000EF', 'bypass_auto': ' W --- 32:126818 32:142858 --:------ 22F7 003 00FFEF', 'reset_filter': ' W --- 32:126818 32:142858 --:------ 10D0 002 00FF'}, '32:126819': {'low': ' I --- 32:126819 32:142858 --:------ 22F1 003 000207', 'medium': ' I --- 32:126819 32:142858 --:------ 22F1 003 000307', 'high': ' I --- 32:126819 32:142858 --:------ 22F1 003 000707', 'bypass_open': ' W --- 32:126819 32:142858 --:------ 22F7 003 00C8EF', 'bypass_close': ' W --- 32:126819 32:142858 --:------ 22F7 003 0000EF', 'bypass_auto': ' W --- 32:126819 32:142858 --:------ 22F7 003 00FFEF', 'reset_filter': ' W --- 32:126819 32:142858 --:------ 10D0 002 00FF'}}}
2024-09-09 17:39:03.383 INFO (MainThread) [custom_components.ramses_cc.schemas] Using a merged schema
2024-09-09 17:39:03.388 DEBUG (MainThread) [ramses_rf.device] Using an explicitly-defined class for: 32:142858 (FAN)
2024-09-09 17:39:03.388 DEBUG (MainThread) [ramses_rf.device.base] Creating a Device: 32:142858 (<class 'ramses_rf.device.hvac.HvacVentilator'>)
2024-09-09 17:39:03.393 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry RAMSES RF for ramses_cc
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 604, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ramses_cc/__init__.py", line 89, in async_setup_entry
    await broker.async_setup()
  File "/config/custom_components/ramses_cc/broker.py", line 146, in async_setup
    await self.client.start(cached_packets=cached_packets())
  File "/usr/local/lib/python3.12/site-packages/ramses_rf/gateway.py", line 184, in start
    load_schema(self, known_list=self._include, **self._schema)  # create faked too
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/ramses_rf/schemas.py", line 370, in load_schema
    _get_device(gwy, device_id)  # domain=key[-4:])
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/ramses_rf/schemas.py", line 343, in _get_device
    return gwy.get_device(dev_id, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/ramses_rf/gateway.py", line 396, in get_device
    traits: dict[str, Any] = SCH_TRAITS(self._include.get(device_id, {}))
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/validators.py", line 259, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/validators.py", line 315, in _exec
    raise error if self.msg is None else AnyInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.12/site-packages/voluptuous/validators.py", line 307, in _exec
    return func(v)
           ^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 205, in __call__
    return self._compiled([], data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/validators.py", line 256, in _run
    return self._exec(self._compiled, value, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/validators.py", line 315, in _exec
    raise error if self.msg is None else AnyInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.12/site-packages/voluptuous/validators.py", line 309, in _exec
    return func(path, v)
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 779, in validate_callable
    return schema(data)
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 205, in __call__
    return self._compiled([], data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 549, in validate_dict
    return base_validate(path, data.items(), out)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/voluptuous/schema_builder.py", line 382, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: not a valid value for dictionary value @ data['class']
```

sciorty, 2024-09-09:
Sorry, wrong repository

zxdavb, 2024-12-04:
closing as wrong repo (was for ramses_cc)

closed 2024-12-04 by zxdavb