ramses-rf/ramses_rf #162: 3EF0 - byte 5: 08

lloyda, 2024-12-07:
I've just started seeing lots of these messages in my HA log.  

```
2024-12-07 16:48:11.192 WARNING (MainThread) [ramses_tx.parsers] RP --- 10:048456 01:185426 --:------ 3EF0 009 000010020008032F64 < Support the development of ramses_rf by reporting this packet (byte 5: 08), with a description of your system
2024-12-07 16:49:11.235 WARNING (MainThread) [ramses_tx.parsers] RP --- 10:048456 18:136212 --:------ 3EF0 009 000010020008032F64 < Support the development of ramses_rf by reporting this packet (byte 5: 08), with a description of your system
2024-12-07 16:49:11.360 WARNING (MainThread) [ramses_tx.parsers] RP --- 10:048456 01:185426 --:------ 3EF0 009 000010020008032F64 < Support the development of ramses_rf by reporting this packet (byte 5: 08), with a description of your system
2024-12-07 16:49:18.711 WARNING (MainThread) [ramses_tx.parsers]  I --- 10:048456 --:------ 10:048456 3EF0 009 000010000008032F64 < Support the development of ramses_rf by reporting this packet (byte 5: 08), with a description of your system
2024-12-07 16:50:11.552 WARNING (MainThread) [ramses_tx.parsers] RP --- 10:048456 01:185426 --:------ 3EF0 009 000010000008032F64 < Support the development of ramses_rf by reporting this packet (byte 5: 08), with a description of your system
```

They seem to have started when the boiler was serviced, and have continued since then.  May just be coincidence, but....

ramses rf 0.42.0
Intergas boiler
Opentherm bridge with OTGW

zxdavb, 2024-12-08:
i'm curious - what did code `08` mean?

lloyda, 2024-12-09:
At the moment I have no idea. The boiler itself was showing no error codes. The first appearance did coincide with when the boiler was being serviced, so maybe something during the service induced a temporary fault.

The OEM fault code is part of OT MsgID 115, but I did not see any of these whilst monitoring with OTGW.  Post boiler (and OTGW) power cycle, some messages with id 115 did appear, but set to zero.   So still not sure exactly what was going on.

closed 2025-07-17 by silverailscolo