ramses-rf/ramses_rf #164: RP --- 32:133288 18:101036 --:------ 2210 042

hermanbanken, 2025-01-13:
My Home Assistant logs contain 171 entries of the following.
I have a Orcon HVAC and a controller/sensor for it. Don't know if this is related to that, or some random noise in my radio space.
```
Logger: ramses_tx.message
Source: /usr/local/lib/python3.12/site-packages/ramses_tx/message.py:270
First occurred: 12 December 2024 at 08:52:38 (171 occurrences)
Last logged: 16:23:27

RP --- 32:133288 18:101036 --:------ 2210 042 00FF00FFFF580000000003FFFFFFFFFF00FFFF580000000003FFFFFFFFFFFFFFFF000000000000000800 < AssertionError(Support the development of ramses_rf by reporting this packet)
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/ramses_tx/message.py", line 255, in _validate
    result = parse_payload(self)
             ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/ramses_tx/parsers.py", line 2938, in parse_payload
    result = _PAYLOAD_PARSERS.get(msg.code, parser_unknown)(msg._pkt.payload, msg)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/ramses_tx/parsers.py", line 1385, in parser_2210
    assert payload in (
           ^^^^^^^^^^^^
AssertionError: Support the development of ramses_rf by reporting this packet
```

hermanbanken, 2025-01-13:
In v0.50 I get
```
Logger: ramses_rf.dispatcher
Source: runner.py:154
First occurred: 19:11:10 (45 occurrences)
Last logged: 21:41:34

RP --- 32:133288 18:101036 --:------ 22E9 004 007EC800 < PacketInvalid(RP --- 32:133288 18:101036 --:------ 22E9 004 007EC800 < Unexpected code for src (FAN) to Tx)
RP --- 32:133288 18:101036 --:------ 22F2 006 0005CB01056C < PacketInvalid(RP --- 32:133288 18:101036 --:------ 22F2 006 0005CB01056C < Unexpected code for src (FAN) to Tx)
RP --- 32:133288 18:101036 --:------ 22F4 013 0060E400000000000000200000 < PacketInvalid(RP --- 32:133288 18:101036 --:------ 22F4 013 0060E400000000000000200000 < Unexpected verb/code for src (FAN) to Tx)
RP --- 32:133288 18:101036 --:------ 313E 011 00000038FA0B003C800000 < PacketInvalid(RP --- 32:133288 18:101036 --:------ 313E 011 00000038FA0B003C800000 < Unexpected code for src (FAN) to Tx)
RP --- 32:133288 18:101036 --:------ 3222 003 000000 < PacketInvalid(RP --- 32:133288 18:101036 --:------ 3222 003 000000 < Unexpected code for src (FAN) to Tx)
```

silverailscolo, 2025-04-15:
@hermanbanken, can you please provide the model of your Orcon HVAC?

WillemAnchor, 2025-05-28:
orcon: 15RF, 15RF display, CO2 inbouw 15RF, HRC-300 EcoMax
Duco: 325 energy premium, RF switch, rf co2 switch, humidity sensor

this error seems to be related:

Logger: ramses_tx.message
Source: /usr/local/lib/python3.13/site-packages/ramses_tx/message.py:270
First occurred: 10:27:47 AM (1 occurrence)
Last logged: 10:27:47 AM

RP --- 32:153289 18:149488 --:------ 2210 042 00FF00FFFFFF0000000000FFFFFFFFFF00FFFFFF0000000000FFFFFFFFFFFFFFFF000000000000020800 < AssertionError(Support the development of ramses_rf by reporting this packet)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/ramses_tx/message.py", line 255, in _validate
    result = parse_payload(self)
  File "/usr/local/lib/python3.13/site-packages/ramses_tx/parsers.py", line 2908, in parse_payload
    result = _PAYLOAD_PARSERS.get(msg.code, parser_unknown)(msg._pkt.payload, msg)
  File "/usr/local/lib/python3.13/site-packages/ramses_tx/parsers.py", line 1370, in parser_2210
    assert payload in (
           ^^^^^^^^^^^^
        "00FF" + "00FFFFFF0000000000FFFFFFFFFF" * 2 + "FFFFFF000000000000000800",
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ), _INFORM_DEV_MSG
    ^
AssertionError: Support the development of ramses_rf by reporting this packet

silverailscolo, 2025-06-02:
> RP --- 32:153289 18:149488 --:------ 2210 042 00FF00FFFFFF0000000000FFFFFFFFFF00FFFFFF0000000000FFFFFFFFFFFFFFFF000000000000020800 <

@WillemAnchor your `2210` Orcon message will be fixed in PR #151