ramses-rf/ramses_rf #170: ruff vs pre-commit

silverailscolo, 2025-03-16:
I see contradicting behavior between running ``ruff format .`` (V0.11.0 on macOS)
```
(venv) mac$ ruff format .
5 files reformatted, 83 files left unchanged

(venv) mac$ ruff format --check .
88 files already formatted
```
and then run pre-commit:

```
(venv) mac$ pre-commit run -a
ruff.....................................................................Passed
ruff-format..............................................................Failed
- hook id: ruff-format
- files were modified by this hook

5 files reformatted, 83 files left unchanged
```
which I expect to run ``ruff format --check .`` looking at the .github/workflow configuration.

Any ideas?

silverailscolo, 2025-03-17:
Somehow Python 3.12 and Python 3.13 have a subtly different handling of ``ruff``. 

Fixed the GitHub workflow lint test by running ``ruff format .`` in a Python 3.13 virtual environment Terminal session outside of my IDE, and once ready to commit from IntelliJ IDE, unchecked the Commit Options checkbox 'run pre-commit hooks' to be able to push the 5 reformatted files.

The f-string assert message was moved to its own line without parentheses, with the comma separating the assert conditions and message at the end of the previous line.

closed 2025-03-17 by silverailscolo

wimpie70, 2025-09-10:
I commented out 'ruff format .' in .pre-commit-config.yaml, and run it now myself just before pushing, so I have more control over it.
Before changing this file, I did add it to .gitignore. (and had to untrack .gitignore as well)

silverailscolo, 2025-09-18:
ruff 0.13.0 was released some time ago. ramses_cc requirements_dev is still at >=0.9.1.
OK to make the jump?

silverailscolo, 2025-09-22:
HomeAssistant core is now at [ruff 0.13.0](https://github.com/home-assistant/core/blob/dev/requirements_test_pre_commit.txt) so will stay with that. Merged via PR #283 

Every dev should update by running `pip install -r requirements_dev.txt`