ramses-rf/ramses_rf #229: 10D0 request gives packet invalid

janvkem, 2025-08-12:
**Describe the bug**
10D0 request (RQ --- 37:123456 32:123456 --:------ 10D0 001 00) gives packet invalid. Value for filter remaining stays at 'unavailable' since 0.51.0. Worked fine before.

**To Reproduce**
Send 10D0 request as described above (with correct fan and remote id's) in HA and check entity change

**Expected behaviour**
Parse of the 10D0 return values

**Please include the following information:**
 - USB connected atmega32u4 on Orcon HRC ecomax

I can provide packets logs if required

janvkem, 2025-08-12:
This is the log from Home Assistant (yes, I have two HVAC systems):

Logger: ramses_rf.dispatcher
Source: runner.py:156
First occurred: 3:00:33 PM (15 occurrences)
Last logged: 9:00:39 PM

RQ --- 37:153258 32:153258 --:------ 10D0 001 00 < PacketInvalid(RQ --- 37:153258 32:153258 --:------ 10D0 001 00 < Unexpected verb/code for src (REM) to Tx)
RP --- 32:153258 37:153258 --:------ 10D0 006 003DB4440000 < PacketInvalid(RP --- 32:153258 37:153258 --:------ 10D0 006 003DB4440000 < Unexpected verb/code for dst (REM) to Rx)
RQ --- 37:155617 32:155617 --:------ 10D0 001 00 < PacketInvalid(RQ --- 37:155617 32:155617 --:------ 10D0 001 00 < Unexpected verb/code for src (REM) to Tx)
RP --- 32:155617 37:155617 --:------ 10D0 006 003DB4440000 < PacketInvalid(RP --- 32:155617 37:155617 --:------ 10D0 006 003DB4440000 < Unexpected verb/code for dst (REM) to Rx)

This is what I see in the packet log:
2025-08-12T21:00:39.156048 000 RQ --- 37:155617 32:155617 --:------ 10D0 001 00
2025-08-12T21:00:39.178738 071 RP --- 32:155617 37:155617 --:------ 10D0 006 003DB4440000

silverailscolo, 2025-08-13:
Applied a fix in PR #231

closed 2025-08-22 by silverailscolo