ramses-rf/ramses_rf #232: Indoor temperature data processing for Orcon system appears to stop at random point in time.

janvkem, 2025-08-13:
**Describe the bug**
Indoor temperature on my Orcon systems stops being processed at random points in time. A reload of the integration temporarily fixes this effect. 31DA messages are sent and other parameters do not seem to suffer from this effect.

**To Reproduce**
Steps to reproduce the behaviour (for example):
1. Have Orcon running continuously and just wait, look at values in Home Assistant

Please provide enough details. For example, if you are having an issue with a device, provide details of the model and specs

**Expected behaviour**
Indoor temperature being processed continuously

**Please include the following information:**
 - USB connected atmega32u4 on Orcon HRC ecomax

First fan:
[image]

Second fan:
[image]

Other parameters seem fine. Outdoor temperature as example:
[image]

Values from packet log at timestamps of no processing to show '31DA' messages are sent:

2025-08-13T08:23:33.665646 070  I --- 32:155617 --:------ 32:155617 31DA 030 00EF007FFF4148093808F2094A08DFF800C802727E0000EFEF1B5818BD00 
2025-08-13T08:24:19.213428 081  I --- 29:183967 32:153258 --:------ 31E0 008 0000000001001E00
2025-08-13T08:24:19.251296 077  I 051 32:153258 --:------ 32:153258 31D9 017 001A020020202020202020202020202008
2025-08-13T08:24:19.347261 078  I --- 32:153258 --:------ 32:153258 31DA 030 00EF007FFF3E4A09B008A209A008FBF800C80296740000EFEF12DE186700
2025-08-13T08:25:46.687900 078  I --- 32:153258 --:------ 32:153258 31DA 030 00EF007FFF3E4A09B008AC09A008FDF800C80296740000EFEF12C2186700
2025-08-13T08:26:53.805848 069  I 239 32:155617 --:------ 32:155617 31D9 017 001A020020202020202020202020202008
2025-08-13T08:26:53.911156 068  I --- 32:155617 --:------ 32:155617 31DA 030 00EF007FFF4148093808FC094D08E9F800C802727E0000EFEF1B5818BD00
2025-08-13T08:26:55.438354 067  I --- 29:186497 32:155617 --:------ 31E0 008 0000000001001E00
2025-08-13T08:26:55.476344 068  I 240 32:155617 --:------ 32:155617 31D9 017 001A020020202020202020202020202008
2025-08-13T08:27:40.301207 067  I --- 29:186497 --:------ 29:186497 1298 003 000225
2025-08-13T08:29:42.724557 075  I --- 29:183967 32:153258 --:------ 31E0 008 0000000001001E00
2025-08-13T08:29:42.752466 077  I 052 32:153258 --:------ 32:153258 31D9 017 001A020020202020202020202020202008
2025-08-13T08:29:42.848467 077  I --- 32:153258 --:------ 32:153258 31DA 030 00EF007FFF3E4A09B008AC09A10900F800C80296740000EFEF12DE186700
2025-08-13T08:31:39.600741 065  I --- 29:186497 32:155617 --:------ 31E0 008 0000000001001E00
2025-08-13T08:31:39.633746 069  I 241 32:155617 --:------ 32:155617 31D9 017 001A020020202020202020202020202008
2025-08-13T08:32:03.310224 070  I 242 32:155617 --:------ 32:155617 31D9 017 001A020020202020202020202020202008
2025-08-13T08:32:03.411143 070  I --- 32:155617 --:------ 32:155617 31DA 030 00EF007FFF4148094209100950091BF800C802727E0000EFEF1B5818BD00
2025-08-13T08:32:13.669506 069  I --- 32:155617 --:------ 32:155617 31DA 030 00EF007FFF414709420910094E0917F800C802727E0000EFEF1B5818BD00

janvkem, 2025-08-14:
I had a Quick look on my phone and indeed I do see a '12A0' around the time it goes flat

silverailscolo, 2025-08-16:
@janvkem Could you help me test if recent changes made this any better, by changing (only) the "ramses-rf==0.51.0" version number to 0.51.1 in `config/custom-components/ramses_cc_/manifest.json` using the File editor add-on, and rebooting?
If it doesn't reload, I often roll back by choosing Download again in HACS.

janvkem, 2025-08-16:
I have changed the version number and performed a restart. I will keep an eye on any flatlining and let you know.

silverailscolo, 2025-08-17:
Still OK when you load up `ramses_rf==0.51.3`, just out?
I split out more temperatures, so please also check other sensors available for your device (my Ventura never has anything in `31DA` exhaust_temp so I use that for an extra `12A0` temperature).

closed 2025-08-22 by silverailscolo