ramses-rf/ramses_rf #276: Add support for Orcon specific content of `22F3` code

Kars-de-Jong, 2025-09-18:
The `22F3` code is not fully understood yet, but I think I understand it for Orcon fans.
I have two different devices that send these codes. One is a battery powered remote (EAN 08718403862997), the other one a CO2 sensor with button controls for fan speeds (EAN 08718403863376).

When using the remote, `22F3` is sent with a specified mode and duration, and the "return mode" is specified as "use previous". So the fan returns to the previously set mode after the timer has expired.

When using the CO2 sensor, `22F3` is sent with a specified mode and duration, and the "return mode" is specified as "use specified", where the mode is specified in the packet (it always returns to `auto`).

Using this info we can determine the flag bits for Orcon.

The packet from the remote to request speed 3 for 15 minutes:
```
2025-09-06T09:11:31.977068 ...  I --- 29:123456 32:123456 --:------ 22F3 007 00020F03040000  # {'minutes': 15, 'flags': [0, 0, 0, 0, 0, 0, 1, 0], '_new_speed_mode': 'per_vent_speed', '_fallback_speed_mode': None, 'rate': None, '_unknown_5': '0000'}
```

So for Orcon this should actually be: `'_new_speed_mode': 'per_request', '_fallback_speed_mode: 'per_vent_speed', 'rate': 'high'``.

I think it would be better to use `new_rate` and `fallback_rate` instead, because there can be 2 rates in the packet.

The packet from the CO2 sensor to request speed 3 for 60 minutes:
```
2025-09-06T22:31:19.792895 ...  I --- 37:123456 32:123456 --:------ 22F3 007 00123C03040404  # {'minutes': 60, 'flags': [0, 0, 0, 1, 0, 0, 1, 0], '_new_speed_mode': 'per_vent_speed', '_fallback_speed_mode': 'per_request', 'rate': None, '_unknown_5': '0404'}
```

For Orcon this should be (using the suggest names for the rates): `'_new_speed_mode': 'per_request', '_fallback_speed_mode: 'per_request', 'new_rate': 'high', 'fallback_rate': 'auto'`.

This also shows that `_unknown_5` is actually the requested fallback rate, where `'0000'` is sent when the mode is set to `'per_vent_speed'`. It is set to `'0404'` in the other packet, which indeed means mode 'auto' for Orcon fans.

By the way,, this logging also shows that the current attempt to decode the mode by calling the `22f1` parser doesn't work (it always shows `None` in my case).

Additional info:
`10E0` packet from the CO2 sensor:
```
2025-09-05T05:04:03.781873 ...  I --- 37:123456 63:262142 --:------ 10E0 038 000001C88D030167FEFFFFFFFFFF1C0707E5564D492D31354D43303100000000000000000000  # {'oem_code': '67', 'manufacturer_sub_id': 'C8', 'product_id': '8D', 'date_1': '2021-07-28', 'date_2': '0000-00-00', 'description': 'VMI-15MC01'}
```
`10E0` packet from the remote (obtained by entering binding mode):
```
2025-09-18T10:52:59.922182 ...  I --- 29:123456 63:262142 --:------ 10E0 029 000001C827070167FFFFFFFFFFFFFFFFFFFF564D4E2D31354C46303100  # {'oem_code': '67', 'manufacturer_sub_id': 'C8', 'product_id': '27', 'date_1': '0000-00-00', 'date_2': '0000-00-00', 'description': 'VMN-15LF01'}
```

silverailscolo, 2025-09-18:
Is there something in the payload we can use to confirm this is an Orcon `22F3`? (length, some unique bytes)
Otherwise we might spoil other fan remote info.
Please try by adding these messages to the tests/tests/parsers/_22F3 log and compare the proposed parsing to other payloads.

silverailscolo, 2025-09-18:
Perfect plan. In the long run we may develop model-specific parsing but for now you can monitor your remote & CO2