ramses-rf/ramses_rf #290: `2411` assert fails

silverailscolo, 2025-09-30:
**Describe the bug**
Out of the blue, I see this in the log after reboot:

```
Logger: ramses_tx.message
Bron: /usr/local/lib/python3.13/site-packages/ramses_tx/message.py:270
Eerst voorgekomen: 00:13:28 (6 gebeurtenissen)
Laatst gelogd: 00:13:29

RQ --- 18:154951 37:153226 --:------ 2411 003 0000DA < AssertionError(param DA is unknown)
RP --- 37:153226 18:154951 --:------ 2411 023 0000DA7F00000000000000000000000003000000018A00 < AssertionError(param DA is unknown)
RQ --- 18:154951 37:153226 --:------ 2411 003 000007 < AssertionError(param 07 is unknown)
RQ --- 18:154951 37:153226 --:------ 2411 003 000088 < AssertionError(param 88 is unknown)
RP --- 37:153226 18:154951 --:------ 2411 023 0000881510000002BC000001900000076C000000018A33 < AssertionError(param 88 is unknown)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/ramses_tx/message.py", line 255, in _validate
    result = parse_payload(self)
  File "/usr/local/lib/python3.13/site-packages/ramses_tx/parsers.py", line 2973, in parse_payload
    result = _PAYLOAD_PARSERS.get(msg.code, parser_unknown)(msg._pkt.payload, msg)
  File "/usr/local/lib/python3.13/site-packages/ramses_tx/parsers.py", line 1912, in parser_2411
    assert payload[4:6] in _2411_TABLE, (
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
AssertionError: param 87 is unknown
```

**Please include the following information:**
 - your **RF Gateway** ramses_esp over (MQTT)
 - output from the **Serial Console**, if applicable
 - contemporaneous portion of the **packet.log**, preferably for several hours

**Screenshots**
Only if applicable, add screenshots to help explain your problem. Do not provide a screenshot as an alternative to providing the above information.

**Additional context**
Add any other context about the problem here, such as the make/model of the device etc.

wimpie70, 2025-10-01:
If its not listed in ramses.py it will generate this error. I understood there are a lot of settings missing there but we don't have the docs yet. Do I understand correctly that you can use both Ramses RF and Modbus ? That would be great to see what's happening.
I used AI to decode some of these messages, it can be pretty good at that. If you can gather some of that info...
I wouldn't do all of them, just a few to see if we can decode them. If you need help with that, send me the info.

silverailscolo, 2025-10-01:
This is Modbus setting the `base_vent_enabled` register to Off for the night:
```
2025-09-30T22:00:01.804915 056 RQ --- 18:154951 37:153226 --:------ 2411 003 000007
2025-09-30T22:00:02.006368 056  W --- 18:154951 37:153226 --:------ 2411 022 00000700000000000000000000000000000000000000
2025-09-30T22:00:02.140166 075  I --- 37:153226 --:------ 37:153226 12A0 021 004007587FFF0001EF7FFF7FFF000245071A04E000
2025-09-30T22:00:06.633053 055 RQ --- 18:154951 37:153226 --:------ 2411 003 000087
2025-09-30T22:00:06.658298 075 RP --- 37:153226 18:154951 --:------ 2411 023 0000871400000000000000000000000002000000018A00
```
So there is also a `12A0` being sent by the HRU, expect it to contain the change:
```
2025-09-30T22:00:02.140166 075  I --- 37:153226 --:------ 37:153226 12A0 021 004007587FFF0001EF7FFF7FFF000245071A04E000
```

wimpie70, 2025-10-01:
I didn't get a lot out of that. Maybe we can get a better result if we also have the 'On' sequence. I checked 12a0 in parsers.py, looks you decoded this before as Indoor_humidity ? Anyways, this is what I got back:
Request status:
RQ --- 18:154951 → 37:153226 --:------ 2411 003 000007

Follow up, sometimes required to trigger a reply:
W --- 18:154951 → 37:153226 --:------ 2411 022 00000700000000000000000000000000000000000000

Reply with status info:
I --- 37:153226 --:------ 37:153226 12A0 021 004007587FFF0001EF7FFF7FFF000245071A04E000
Example decoding:
0040 → some parameter / version
0758 → likely a reading (could be 18.00°C if divided by 40, depends on context)
7FFF → “not available / invalid”
0001 → flag set

4.5 seconds later:
RQ --- 18:154951 → 37:153226 --:------ 2411 003 000087

Reply:
RP --- 37:153226 → 18:154951 --:------ 2411 023 0000871400000000000000000000000002000000018A00
000087 → index requested
14 → value at index 87 (could be a counter, type ID, or parameter code)
trailing values: flags, counters, possibly error log entries.
8A00 (at the end) could be a checksum or a value in little-endian (0x008A = 138).

I tried sending RQ 2411 - 07 to my Orcon FAN, but got no response, but 87 gave me:
2025-10-01T15:57:38.669062 000 RQ --- 18:130236 32:153289 --:------ 2411 003 000087
2025-10-01T15:57:38.711529 035 RP --- 32:153289 18:130236 --:------ 2411 023 000087651100000BB8000000000000271000000001CB3C

closed 2025-10-16 by silverailscolo